Koenkk/zigbee2mqtt #16947: Sonoff ZBBridge cannot connect

DavidArthur2, 2023-03-07:
### What happened?

Yesterday afternoon the Z2M stopped, and could not restart it. Socket error. I have not changed anything, wasn't even in the area.

### What did you expect to happen?

This may be a version problem, so fixing it.

### How to reproduce it (minimal and precise)

Upgrade to 1.30.2

### Zigbee2MQTT version

1.30.2

### Adapter firmware version

12.3.1

### Adapter

EZSP

### Debug log

[08:31:10] INFO: Preparing to start...
[08:31:10] INFO: Socat not enabled
[08:31:10] INFO: Zigbee Herdsman debug logging enabled
[08:31:10] INFO: Starting Zigbee2MQTT...
Zigbee2MQTT:debug 2023-03-07 08:31:11: Loaded state from file /config/zigbee2mqtt/state.json
Zigbee2MQTT:info  2023-03-07 08:31:11: Logging to console and directory: '/config/zigbee2mqtt/log/2023-03-07.08-31-11' filename: log.txt
Zigbee2MQTT:debug 2023-03-07 08:31:11: Removing old log directory '/config/zigbee2mqtt/log/2023-03-03.11-05-27'
Zigbee2MQTT:info  2023-03-07 08:31:11: Starting Zigbee2MQTT version 1.30.2 (commit #unknown)
Zigbee2MQTT:info  2023-03-07 08:31:12: Starting zigbee-herdsman (0.14.96)
Zigbee2MQTT:debug 2023-03-07 08:31:12: Using zigbee-herdsman with settings: '{"adapter":{"concurrent":null,"delay":null,"disableLED":false},"backupPath":"/config/zigbee2mqtt/coordinator_backup.json","databaseBackupPath":"/config/zigbee2mqtt/database.db.backup","databasePath":"/config/zigbee2mqtt/database.db","network":{"channelList":[11],"extendedPanID":[221,221,221,221,221,221,221,221],"networkKey":"HIDDEN","panID":6754},"serialPort":{"adapter":"ezsp","path":"tcp://192.168.67.99:8888"}}'
2023-03-07T06:31:12.074Z zigbee-herdsman:controller:log Starting with options '{"network":{"networkKeyDistribute":false,"networkKey":[1,3,5,7,9,11,13,15,0,2,4,6,8,10,12,13],"panID":6754,"extendedPanID":[221,221,221,221,221,221,221,221],"channelList":[11]},"serialPort":{"path":"tcp://192.168.67.99:8888","adapter":"ezsp"},"databasePath":"/config/zigbee2mqtt/database.db","databaseBackupPath":"/config/zigbee2mqtt/database.db.backup","backupPath":"/config/zigbee2mqtt/coordinator_backup.json","adapter":{"disableLED":false,"concurrent":null,"delay":null}}'
2023-03-07T06:31:12.076Z zigbee-herdsman:adapter:ezsp:uart Opening TCP socket with 192.168.67.99:8888
2023-03-07T06:31:12.096Z zigbee-herdsman:adapter:ezsp:uart Socket error
2023-03-07T06:31:12.113Z zigbee-herdsman:adapter:ezsp:erro Connection attempt 1 error: Error: Error while opening socket
    at Socket.<anonymous> (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/uart.ts:147:24)
    at Socket.emit (node:events:525:35)
    at emitErrorNT (node:internal/streams/destroy:157:8)
    at emitErrorCloseNT (node:internal/streams/destroy:122:3)
    at processTicksAndRejections (node:internal/process/task_queues:83:21)
2023-03-07T06:31:12.113Z zigbee-herdsman:adapter:ezsp:uart Port closed
2023-03-07T06:31:12.113Z zigbee-herdsman:adapter:ezsp:ezsp Close ezsp
2023-03-07T06:31:12.113Z zigbee-herdsman:adapter:ezsp:driv Close driver
2023-03-07T06:31:17.119Z zigbee-herdsman:adapter:ezsp:ezsp Next attempt 2
2023-03-07T06:31:17.119Z zigbee-herdsman:adapter:ezsp:uart Opening TCP socket with 192.168.67.99:8888
2023-03-07T06:31:17.126Z zigbee-herdsman:adapter:ezsp:uart Socket error
2023-03-07T06:31:17.126Z zigbee-herdsman:adapter:ezsp:erro Connection attempt 2 error: Error: Error while opening socket
    at Socket.<anonymous> (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/uart.ts:147:24)
    at Socket.emit (node:events:525:35)
    at emitErrorNT (node:internal/streams/destroy:157:8)
    at emitErrorCloseNT (node:internal/streams/destroy:122:3)
    at processTicksAndRejections (node:internal/process/task_queues:83:21)
2023-03-07T06:31:17.126Z zigbee-herdsman:adapter:ezsp:uart Port closed
2023-03-07T06:31:17.126Z zigbee-herdsman:adapter:ezsp:ezsp Close ezsp
2023-03-07T06:31:17.127Z zigbee-herdsman:adapter:ezsp:driv Close driver
2023-03-07T06:31:22.130Z zigbee-herdsman:adapter:ezsp:ezsp Next attempt 3
2023-03-07T06:31:22.130Z zigbee-herdsman:adapter:ezsp:uart Opening TCP socket with 192.168.67.99:8888
2023-03-07T06:31:22.135Z zigbee-herdsman:adapter:ezsp:uart Socket error
2023-03-07T06:31:22.136Z zigbee-herdsman:adapter:ezsp:erro Connection attempt 3 error: Error: Error while opening socket
    at Socket.<anonymous> (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/uart.ts:147:24)
    at Socket.emit (node:events:525:35)
    at emitErrorNT (node:internal/streams/destroy:157:8)
    at emitErrorCloseNT (node:internal/streams/destroy:122:3)
    at processTicksAndRejections (node:internal/process/task_queues:83:21)
2023-03-07T06:31:22.136Z zigbee-herdsman:adapter:ezsp:uart Port closed
2023-03-07T06:31:22.136Z zigbee-herdsman:adapter:ezsp:ezsp Close ezsp
2023-03-07T06:31:22.136Z zigbee-herdsman:adapter:ezsp:driv Close driver
2023-03-07T06:31:27.141Z zigbee-herdsman:adapter:ezsp:ezsp Next attempt 4
2023-03-07T06:31:27.141Z zigbee-herdsman:adapter:ezsp:uart Opening TCP socket with 192.168.67.99:8888
2023-03-07T06:31:27.147Z zigbee-herdsman:adapter:ezsp:uart Socket error
2023-03-07T06:31:27.148Z zigbee-herdsman:adapter:ezsp:erro Connection attempt 4 error: Error: Error while opening socket
    at Socket.<anonymous> (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/uart.ts:147:24)
    at Socket.emit (node:events:525:35)
    at emitErrorNT (node:internal/streams/destroy:157:8)
    at emitErrorCloseNT (node:internal/streams/destroy:122:3)
    at processTicksAndRejections (node:internal/process/task_queues:83:21)
2023-03-07T06:31:27.148Z zigbee-herdsman:adapter:ezsp:uart Port closed
2023-03-07T06:31:27.149Z zigbee-herdsman:adapter:ezsp:ezsp Close ezsp
2023-03-07T06:31:27.149Z zigbee-herdsman:adapter:ezsp:driv Close driver
2023-03-07T06:31:32.153Z zigbee-herdsman:adapter:ezsp:ezsp Next attempt 5
2023-03-07T06:31:32.154Z zigbee-herdsman:adapter:ezsp:ezsp ==> version: {"desiredProtocolVersion":4}
Zigbee2MQTT:error 2023-03-07 08:31:32: Error while starting zigbee-herdsman
Zigbee2MQTT:error 2023-03-07 08:31:32: Failed to start zigbee
Zigbee2MQTT:error 2023-03-07 08:31:32: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
Zigbee2MQTT:error 2023-03-07 08:31:32: Exiting...
Zigbee2MQTT:error 2023-03-07 08:31:32: Error: Connection not initialized
    at Ezsp.execCommand (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/ezsp.ts:540:19)
    at Ezsp.version (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/ezsp.ts:338:35)
    at Driver.startup (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/driver.ts:141:25)
    at EZSPAdapter.start (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/adapter/ezspAdapter.ts:165:16)
    at Controller.start (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:132:29)
    at Zigbee.start (/app/lib/zigbee.ts:58:27)
    at Controller.start (/app/lib/controller.ts:101:27)
    at start (/app/index.js:107:5)

DavidArthur2, 2023-03-07:
It doesn't work either with ZHA, but another bridge does work with both.
May something in the add-on change on the bridge something? I can open the web on it's ip, so it 's on network.

ryny24, 2023-03-20:
This does appear to be the same issue I've had for a year now.   I have resolved it by replacing my gateway with a [Moes ZHUB Ethernet Gateway](https://www.aliexpress.com/item/2255800279107159.html?gatewayAdapt=4itemAdapt). It does require some flashing, but at least my devices stay online.

gerritdrost, 2023-03-22:
On 2022-03-17 I downgraded zigbee2mqtt to version 1.29.2. Since then I noticed two things:
- I no longer have to reconfigure the TCP-port in Tasmota, ever
- zigbee2mqtt will crash irregularly with "Error: ZdoRequest error" (0-5 times a day)

Since systemd will restart zigbee2mqtt after crashing, and it will immediately work again, I now have a sort of working situation. This does convince me that the cause of the problem seems to be zigbee2mqtt, not tasmota. I know wifi-based devices are not recommended, but I have a decent wifi network, and this sonoff zbbridge had been rock stable for months using ZHA.

Since I want to be able to run the latest greatest and I don't want my service to restart multiple times a day, I'm switching to a Smartlight SLZB-06 next weekend.

Here's an example of the ZdoRequest error in my logs:
```
Mar 17 13:07:09 zigbee2mqtt npm[10103]: Error: ZdoRequest error
Mar 17 13:07:09 zigbee2mqtt npm[10103]:     at Driver.zdoRequest (/home/zigbee2mqtt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/driver.ts:571:19)
Mar 17 13:07:09 zigbee2mqtt npm[10103]:     at EZSPAdapter.nodeDescriptorInternal (/home/zigbee2mqtt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/ezsp/adapter/ezspAdapter.ts:373:28)
Mar 17 13:07:09 zigbee2mqtt npm[10103]:     at Object.func (/home/zigbee2mqtt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/ezsp/adapter/ezspAdapter.ts:363:32)
Mar 17 13:07:09 zigbee2mqtt npm[10103]:     at Queue.executeNext (/home/zigbee2mqtt/zigbee2mqtt/node_modules/zigbee-herdsman/src/utils/queue.ts:32:32)
Mar 17 13:07:09 zigbee2mqtt systemd[118]: zigbee2mqtt.service: Main process exited, code=exited, status=1/FAILURE
Mar 17 13:07:09 zigbee2mqtt systemd[118]: zigbee2mqtt.service: Failed with result 'exit-code'.
```

ryny24, 2023-03-24:
I was not aware of this Zigbee adapter. Does it work without needing flashing? Please let me know how it works, I would like to try it.

mklan, 2023-04-14:
I am running the following script using crontab every minute. This will restart tcp if it stops:

```bash
#!/bin/bash

HOST=192.168.1.102
WEB_PORT=80
TCP_PORT=8888

if (nc -z -w 2 $HOST $WEB_PORT 2>&1 >/dev/null)
then
    if !(nc -z -w 2 $HOST $TCP_PORT 2>&1 >/dev/null)
    then
        echo "[$(date)] tcp not running. Will start!" >> zbbridge.log
        curl http://$HOST/cm?cmnd=TCPStart%20$TCP_PORT
    fi
else
        echo "[$(date)] zbbridge not reachable!" >> zbbridge.log
fi
```

closed 2023-05-22 by github-actions[bot]

1moere1, 2023-07-09:
I get the same error with the sonoff zigbee bridge pro. This
`backlog rule1 on system#boot do TCPStart 8888 endon ; rule1 1 ; template {"NAME":"Sonoff ZHABridge","GPIO":[56,208,0,209,59,58,0,0,0,0,0,0,17],"FLAG":0,"BASE":18} ; module 0`
seemed not to help (even more, it destroyed the device and i have to reflash it).