Koenkk/zigbee2mqtt #16949: Aqara roller shade companion E1 not updating automatically

WorlockM, 2023-03-07:
### What happened?

The position status of my Aqara roller shade companion E1 do not update unless I manually press the refresh button. I have 3 of them and they all have this problem.

I also have a Aqara curtain driver E1 that **does not** have this problem.

### What did you expect to happen?

Automatically refresh the position status when the curtain is being controlled

### How to reproduce it (minimal and precise)

Controlling the curtain from Zigbee2MQTT or manually by pressing the buttons on the device.

### Zigbee2MQTT version

1.30.2

### Adapter firmware version

0x26580700

### Adapter

ConBee2

mamrai1, 2023-03-07:
you need newer firmware for conbee ii (at least 0x26720700]) for full functionality of aqara curtain driver e1...
But i suggest to wait because you are on the most stable firmware of conbee for z2m

closed 2023-04-14 by github-actions[bot]

victor7325, 2023-06-14:
I have the same problem, my z2m adapter is Xiaomi Multimode Gateway 2 (CN)	(DMWG03LM) with AlexxIT/XiaomiGateway3 plug-in.