Koenkk/zigbee2mqtt #16974: Tuya TS0601 BRT-100-TRV TRV connects to Coordinator instead of closest Router resulting in very low LQI

mdzhigarov, 2023-03-09:
### What happened?

I have 5 Tuya TS0601 BRT-100-TRV devices on the second floor of my home. 
I have a SONOFF ZBMINI router on the same floor and a ZBDongle-E coordinator on the first floor.

4 of the Tuya devices are connected to the SONOFF ZBMINI router with excellent lqi - 148 - 160.
The last Tuya device initially connects to the SONOFF ZBMINI upon successful pairing and also has excellent LQI.
After a couple of hours - it decides to connect directly to the Coordinator ZBDongle-E on the first floor and LQI drops to 80 and the Tuya device stops responding to commands.

When pairing the problematic Tuya device I'm explicitly selecting my SONOFF ZBMINI router when I click the "Permit join" button. Initially, the pairing is done with the SONOFF ZBMINI and the connection is stable - 160~, then couple of hours later the Tuya device connects directly to the coordinator and breaks.

### What did you expect to happen?

The BRT-100-TRV device should have connected to the SONOFF ZBMINI router which is located in the same room.

### Zigbee2MQTT version

1.30.2

### Adapter firmware version

6.10.3.0 build 297

### Adapter

ZBDongle-E

### Debug log

[mqtt.log](https://github.com/Koenkk/zigbee2mqtt/files/10931974/mqtt.log)

mdzhigarov, 2023-03-09:
Probably also related to https://github.com/Koenkk/zigbee2mqtt/issues/15507

mdzhigarov, 2023-03-09:
[home-assistant_2023-03-09T13-53-17.641Z.log](https://github.com/Koenkk/zigbee2mqtt/files/10932281/home-assistant_2023-03-09T13-53-17.641Z.log)

Koenkk, 2023-03-09:
Zigbee2MQTT cannot influence to which parent a device chooses, this is decided by the device itself. Therefore this can not be fixed by Zigbee2MQTT.

closed 2023-03-10 by mdzhigarov