Koenkk/zigbee2mqtt #16979: Router has no link to parten on map

MikOsle, 2023-03-09:
### What happened?

When loading a map some of the routers in my network have no parent link to the coordinator or another router. Some routers seem to exist as small clusters with links only to each other but not the coordinator.
Though, when checking off 'IsASibling' all routers have a link to another router and to the coordinator, except for one which have no solid line to the coordinator. What could be the reason for these missing links?
[image]
[image]

### What did you expect to happen?

All routers to have a link parent

### Zigbee2MQTT version

1.30.2 commit: unknown

### Adapter firmware version

6.10.3.0 build 297

### Adapter

Sonoff Zigbee 3.0 USB Dongle-E

Koenkk, 2023-03-11:
See https://www.zigbee2mqtt.io/guide/faq/#why-are-some-links-missing-from-my-networkmap

MikOsle, 2023-03-11:
> See https://www.zigbee2mqtt.io/guide/faq/#why-are-some-links-missing-from-my-networkmap

Thanks. Yes, I had already read it, but I understand it as only between end device to router or coordinator. My issue is the link between router to router which are not battery powered

closed 2023-04-19 by github-actions[bot]

h00g, 2024-12-20:
> > See https://www.zigbee2mqtt.io/guide/faq/#why-are-some-links-missing-from-my-networkmap
> 
> Thanks. Yes, I had already read it, but I understand it as only between end device to router or coordinator. My issue is the link between router to router which are not battery powered

I have the same situation with
- two IKEA Trådfri bulbs forming a cluster of their own without any parent link to a router or coordinator
- two cases of a IKEA Trådfri bulb and plug respectively and a remote forming such separate clusters with no parent

[image]