Koenkk/zigbee2mqtt #16981: [New device support]: Update ZM16EL-03/33 _TZE200_68nvbio9

jracoma, 2023-03-10:
### Link

https://www.zemismart.com/products/zm16el-zigbee

### Database entry

{"id":20,"type":"EndDevice","ieeeAddr":"0x540f57fffe7bdf79","nwkAddr":15998,"manufId":4098,"manufName":"_TZE200_68nvbio9","powerSource":"Battery","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[0,4,5,61184],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"r�+\u0019��+\u0019","65506":31,"65508":0,"65534":0,"modelId":"TS0601","manufacturerName":"_TZE200_68nvbio9","stackVersion":0,"dateCode":"","zclVersion":3,"appVersion":72,"powerSource":3}},"genPowerCfg":{"attributes":{}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":72,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":-708457359},"lastSeen":1678422064626,"defaultSendRequestWhen":"immediate"}

### Comments

Adding missing features from existing _TZE200_68nvbio9. I tried using tuyaDatapoints but a few were missing and ended up having to write converter.

### External converter

```shell
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const dataPointsLocal = {
	battery: 13,
	border: 16,
	clickControl: 20,
};

const fzLocal = {
	ZM16EL: {
		cluster: 'manuSpecificTuya',
		type: ['commandDataReport'],
		options: [exposes.options.invert_cover()],
		convert: (model, msg, publish, options, meta) => {
			const dpValue = tuya.firstDpValue(msg, meta, 'ZM16EL');
			const dp = dpValue.dp;
			const value = tuya.getDataValue(dpValue);

			switch (dp) {
				case tuya.dataPoints.coverPosition: // Started moving to position (triggered from Zigbee)
				case tuya.dataPoints.coverArrived: { // Arrived at position
					const invert = tuya.isCoverInverted(meta.device.manufacturerName) ? !options.invert_cover : options.invert_cover;
					const position = invert ? 100 - (value & 0xff) : value & 0xff;

					if (position > 0 && position <= 100) {
						return { 'position': position, 'state': 'OPEN' };
					} else if (position == 0) {
						return { 'position': position, 'state': 'CLOSE' };
					} else {
						// result.running = running; // Not calibrated yet, no position is available
					}
				}
					break;
				case tuya.dataPoints.state:
					return { 'state': { 'open': 0, 'stop': 1, 'close': 2 }[value] };
					break;
				case tuya.dataPoints.motorDirection:
					return { 'motorDirection': value };
					break;
				case dataPointsLocal.battery:
					return { 'battery': value };
					break;
				case dataPointsLocal.clickControl:
					return { 'clickControl': value };
					break;
				default:
					meta.logger.warn(`ZM16EL03(0x540f57fffe7bdf79) fz : Unhandled DP #${dp} for ${meta.device.manufacturerName}:
                    ${JSON.stringify(value)}`);
			}
		}
	},
};

const tzLocal = {
	ZM16EL: {
		key: ['state', 'position', 'border', 'options', 'click_control'],
		options: [exposes.options.invert_cover()],
		convertSet: async (entity, key, value, meta) => {
			switch (key) {
				case 'state':
					await tuya.sendDataPointEnum(entity, tuya.dataPoints.state, { 'open': 0, 'stop': 1, 'close': 2 }[value.toLowerCase()]);
					break;
				case 'position':
					const invert = tuya.isCoverInverted(meta.device.manufacturerName) ? !meta.options.invert_cover : meta.options.invert_cover;
					const position = invert ? 100 - (value & 0xff) : value & 0xff;

					await tuya.sendDataPointValue(entity, tuya.dataPoints.coverPosition, position);
					break;
				case 'border':
					await tuya.sendDataPointEnum(entity, dataPointsLocal.border, { 'up': 0, 'down': 1, 'up_delete': 2, 'down_delete': 3, 'remove_top_bottom': 4 }[value]);
					break;
				case 'options':
					if (value.reverse_direction != undefined) {
						if (value.reverse_direction) {
							meta.logger.info('Motor direction reverse');
							await tuya.sendDataPointEnum(entity, tuya.dataPoints.motorDirection, 1);
						} else {
							meta.logger.info('Motor direction forward');
							await tuya.sendDataPointEnum(entity, tuya.dataPoints.motorDirection, 0);
						}
					}
					break;
				case 'click_control':
					await tuya.sendDataPointEnum(entity, dataPointsLocal.clickControl, { 'up': 0, 'down': 1 }[value]);
					break;
				default:
					meta.logger.warn(`ZM16EL03(0x540f57fffe7bdf79) tz: Unhandled key #${key} for ${meta.device.manufacturerName}:
                    ${JSON.stringify(value)}`);
			}
		}
	},
};

const definition = {
	fingerprint: [
		{
			modelID: 'TS0601',
			manufacturerName: '_TZE200_68nvbio9',
		},
	],
	model: 'ZM16EL-03/33',
	vendor: 'Zemismart',
	description: 'Smart tubular motor',
	fromZigbee: [fzLocal.ZM16EL, fz.ignore_basic_report],
	toZigbee: [tzLocal.ZM16EL],
	exposes: [
		e.battery(),
		e.cover_position().setAccess('position', ea.STATE_SET),
		exposes.composite('options', 'options', ea.STATE_SET)
			.withFeature(exposes.binary('reverse_direction', ea.STATE_SET, true, false)
				.withDescription('Reverse the motor direction')),
		exposes.enum('border', ea.STATE_SET, ['up', 'down', 'up_delete', 'down_delete', 'remove_top_bottom']),
		exposes.enum('click_control', ea.STATE_SET, ['up', 'down'])
			.withDescription('Single motor steps'),
	],
};

module.exports = definition;
```

Koenkk, 2023-03-11:
What was missing in order to support it via `tuyaDataPoints`?

closed 2023-03-11 by Koenkk

jracoma, 2023-03-11:
These are the datapoints I pulled from the Tuya IOT platform.
Control/State: 1 - open/stop/close/continuous - Enum
(Set) Curtain Position Setting: 2 - Integer
Current Curtain Position: 3 - Integer
Mode: 4 - night/morning - Enum
Motor Direction (Control Back Mode): 5 - forward/back - Enum
Work State: 7 - opening/closing - Enum
Situation Set: 11 - fully_open/fully_close - Enum (Doesn't work)
Fault: 12 - motor_fault - Label
Battery Percentage: 13 - Integer
Border: 16 - up/down/up_delete/down_delete/remove_top_bottom (all) - Enum
Best Position: 19 - Integer - (Doesn't work) 
Click Control: 20 - up/down - Enum

I tried adding datapoint 2 using below:
`[1, 'state', tuya.valueConverterBasic.lookup({ 'OPEN': tuya.enum(0), 'STOP': tuya.enum(1), 'CLOSE': tuya.enum(2) })],
[2, 'position', tuya.valueConverter.raw],[13, 'battery', tuya.valueConverter.raw]` using `e.cover_position().setAccess('position', ea.STATE_SET),`

State and battery work but position kept giving this error:
`TypeError: Cannot read properties of undefined (reading '1')
    at Object.options (/app/data/tuya.js:1788:40)
    at new ExternalConverters (/app/lib/extension/externalConverters.ts:15:17)
    at Object.addDefinition [as addDeviceDefinition] (/app/node_modules/zigbee-herdsman-converters/index.js:92:81)
    at new Controller (/app/lib/controller.ts:84:58)
    at start (/app/index.js:106:18)
Using '/app/data' as data directory
/app/data/tuya.js:1788
                const dpKey = datapoint[1];`