Koenkk/zigbee2mqtt #16989: Tuya TS0601 model BRT-100-TRV OTA update fails with "No image available for imageType '5634'"

mdzhigarov, 2023-03-10:
### What happened?

Trying to update Tuya TS0601 model BRT-100-TRV via OTA and getting error: "No image available for imageType '5634'"
[image]
[image]

### What did you expect to happen?

Either a successful update or zigbee2mqqt should not suggest that the Tuya device supports OTA updates at all.

### Zigbee2MQTT version

1.30.2

### Adapter firmware version

6.10.3.0 build 297

### Adapter

ZBDongle-E

### Debug log

[z2m.log](https://github.com/Koenkk/zigbee2mqtt/files/10945160/z2m.log)

Koenkk, 2023-03-11:
See https://github.com/Koenkk/zigbee2mqtt/issues/16584#issuecomment-1422885637

closed 2023-03-11 by Koenkk

mdzhigarov, 2023-03-11:
Is there a chance to hide or disable the OTA if it's not supported?

Koenkk, 2023-03-11:
Currently not, as we don't know wether OTA is supported upfront. It would be better if someone grabs the OTA.