Koenkk/zigbee2mqtt #16990: [Feature request]: Option to store/restore previous configuration when a previously paired device is re-paired

ankushg, 2023-03-10:
### Is your feature request related to a problem? Please describe

Sometimes it's necessary to remove or re-pair a device to a zigbee network.

When you do that, it usually involves a factory reset which is disappointing because it wipes all of the configuration data from it

### Describe the solution you'd like

In the SmartThings Zigbee driver, there is an option toward the bottom that says “Re-send all configuration”. 

After you factory reset a device and then re-add it back to ST, you can have ST send all of your configuration options and recreate any bindings, groups, etc.

It would be great if Z2M asked if we want to restore configuration when it detects a that a previously-configured device

### Describe alternatives you've considered

Manually restoring things?

### Additional context

Potential situations
- OTAs are only supported on the official hub, so you temporarily re-pair it to the main hub
- You want to change your zigbee network channel
- You want to try out a device on different platform to either test out or implement support 
- Multiple devices are on the same switch and you need to pair a new device, but that involves unpairing all of the preexisting ones...

ankushg, 2023-04-10:
Please don’t close this 🙏🏽

ankushg, 2023-05-11:
Still curious about this. Is there an easy way to extract all configuration state for a given entity? Even if I can just download a json file and restore it later, that would be super helpful

closed 2023-06-26 by github-actions[bot]

mcwitt, 2023-11-02:
I realize this is closed, but wanted to +1 having a feature like this. I have a large network of Inovelli Blue switches with customized delay, brightness calibration, bindings, etc. that are a pain to manually restore when I need to perform a factory reset on one of the switches. I'd be very interested in a way to declaratively manage these parameters, or restore the last state for known devices after a reset.