Koenkk/zigbee2mqtt #17002: EssentielB EB-E14-P45-RGBW

scottjim, 2023-03-11:
### What happened?

It is a color bulb (16 million colors), we have:
Expose: light (state, brightness, color_temp, color_temp_startup), effect, power_on_behavior, linkquality
but missing light(color_xy)

### What did you expect to happen?

I would like to be able to choose the color of the bulb

### Zigbee2MQTT version

1.30.2

### Adapter firmware version

20220219

### Adapter

zStack3x0

scottjim, 2023-04-02:
> No idea ? @Koenkk

Koenkk, 2023-04-02:
Could you check if the issue is fixed with the following external converter: 

```js
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const utils = require('zigbee-herdsman-converters/lib/utils');
const globalStore = require('zigbee-herdsman-converters/lib/store');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['EB-E14-P45-RGBW'],
    model: 'EB-E14-P45-RGBW',
    vendor: 'EssentielB',
    description: 'Smart LED bulb',
    extend: extend.light_onoff_brightness_colortemp({colorTempRange: [153, 370]}),
};

module.exports = definition;
```

- save this as file next to `configuration.yaml` as `ext_converter.js`
- add it to `configuration.yaml`:
```yaml
external_converters:
  - ext_converter.js
```
- start z2m, check if issue is fixed

scottjim, 2023-04-03:
It doesn't work but if I change  : 

- extend: extend.light_onoff_brightness_colortemp({colorTempRange: [153, 370]}),          

- extend: extend.light_onoff_brightness_colortemp_color({colorTempRange: [153, 370]}),

it works fine now, thank you

 ```js
 const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
 const tz = require('zigbee-herdsman-converters/converters/toZigbee');
 const exposes = require('zigbee-herdsman-converters/lib/exposes');
 const reporting = require('zigbee-herdsman-converters/lib/reporting');
 const extend = require('zigbee-herdsman-converters/lib/extend');
 const ota = require('zigbee-herdsman-converters/lib/ota');
 const tuya = require('zigbee-herdsman-converters/lib/tuya');
 const utils = require('zigbee-herdsman-converters/lib/utils');
 const globalStore = require('zigbee-herdsman-converters/lib/store');
 const e = exposes.presets;
 const ea = exposes.access;
 
 const definition = {
     zigbeeModel: ['EB-E14-P45-RGBW'],
     model: 'EB-E14-P45-RGBW',
     vendor: 'EssentielB',
     description: 'Smart LED bulb',
     extend: extend.light_onoff_brightness_colortemp_color({colorTempRange: [153, 370]}),
 };
 
 module.exports = definition;
 ```

Koenkk, 2023-04-03:
Integrated the fix, thanks!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2023-04-03 by Koenkk