Koenkk/zigbee2mqtt #17009: [Feature request]: Filter out some attribute's value(s)

Chuckame, 2023-03-12:
### Is your feature request related to a problem? Please describe

I have a Philips Hue Smart button, that sends `action` attribute with hold, press, release, shutdown, brightness_step_down and empty string, while I just care about hold, press and release, that makes home assistant automations hard to make because of noise made by unwanted attribute values

### Describe the solution you'd like

Having a way to receive through mqtt only wanted attribute values (regex would be perfect) 

### Describe alternatives you've considered

Nothing, since today we can just fully remove the attribute itself, and not only some values.
Maybe filtering with custom configuration in home assistant, but it will still send useless events through mqtt

### Additional context

Here you can see 4 updates, while only 1 of them are interesting. 
[image]

Chuckame, 2023-03-23:
As said, I don't want to filter out the attribute, but only some values of a specific attribute

danieledwardgeorgehitchcock, 2023-03-23:
Have a look at the `optomistic_filter`

Chuckame, 2023-03-23:
The documentation is not saying that we can filter specific attribute
values :
Same as the filtered_attributes option but only applies to the optimistic
published attributes. Has no effect when optimistic: false is set.

Chuckame, 2023-04-06:
So if filtered_optomistic is not the good candidate, we agree that this feature does not exist?

Chuckame, 2023-04-13:
Any insight for this feature?

closed 2023-06-21 by github-actions[bot]