Koenkk/zigbee2mqtt #17011: RTX ZVG1 ignores the timer value

jvcsw, 2023-03-12:
### What happened?

When I change the timer value (for example, setting 20 minutes) the valve execute auto off after 10  minutes.

This problem has always existed, since the beginning and it still does not work well.

This is the device:

https://www.zigbee2mqtt.io/devices/ZVG1.html

### What did you expect to happen?

Auto off action close the valve accordind to the timer value.

### How to reproduce it (minimal and precise)

Change the timer value with a value different to 10.
Open valve
Wait 10 minutes.

### Zigbee2MQTT version

1.30.2

### Adapter firmware version

-

### Adapter

-

jvcsw, 2023-04-12:
@Koenkk Do you know how to solve this?

Thanks

closed 2023-05-22 by github-actions[bot]

fliespl, 2023-05-29:
@jvcsw did you every sorted it out?

fliespl, 2023-05-30:
Is your model recognized as well as: _TZE200_81isopgh?

Im my case my only issue is timer being unreliable. Set to 15, but still finishes after 10 minutes.

https://ss.codeone.pl/ss-2023-05-30-11-20-33-1685438433-lIdS4RdL.png
https://ss.codeone.pl/ss-2023-05-30-11-21-11-1685438471-7UumjkDy.png

Yet, water consumption works when I wait for timer to kill (not if I switch it manually).
https://ss.codeone.pl/ss-2023-05-30-11-21-54-1685438514-gnV0CwWa.png

markhendrix79, 2023-06-07:
I'm having the same issues as well.  I can't disable the timer and any changes do not take affect - it just continues to turn off after 10 minutes, regardless.  Very frustrating.  I'm also not seeing water consumed.  My MQTT device info looks just like the screenshot above.  I'm sad to see this issue is listed as closed, but wanted to add a comment and follow any updates that may appear.

cptbucky, 2023-06-24:
I'm currently having the same issues, I've never seen a water usage figure come through, as for the timer I appear to be able to set it to 60 and have the timer last for 60 minutes, I think this must follow @ytzelf 's workaround. 

If I switch on the device again the timer remains at 60 but the "time left on the timer" counts down from 10 and shuts off after 10. This feels like a bit of an odd issue, does the device's timer reset on every use? Anyway I'll perhaps follow the workaround for now. Does anyone use the Tuya app to know what the behaviour is? Or know where the device integration lives within z2m that I might look through?

zivillian, 2023-06-26:
I'm using the following json to enable the valve via MQTT:
```json
{
  "state":"ON",
  "timer":30
}
```

this result in the timer being respected and the water consumed to be reported (even if it is totally wrong):

[image]

jvcsw, 2023-06-27:
> I'm currently having the same issues, I've never seen a water usage figure come through, as for the timer I appear to be able to set it to 60 and have the timer last for 60 minutes, I think this [...]

Here there is the code:

https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/src/devices/rtx.ts

I don't understand why using MQTT it works fine and with HA it doesn't.

jvcsw, 2023-06-27:
Last week @kamaradclimber changed the unit from "l" to "L", maybe that solves the water consumed problem, or not...

https://github.com/Koenkk/zigbee-herdsman-converters/pull/5897/commits/d5ba786745cd8dc9fc6ca080a9db0b96d7faf4d2

markhendrix79, 2023-10-08:
I just wanted to add here in case anyone finds this thread later - this issue was fixed in release 1.32.2.  I had been using the device through the tuya integration, which still didn't pass in a working timer to home assistant.  I switched back over to z2m today and everything is working - no extra scripts or json mods required.

jvcsw, 2023-10-09:
It doesn't tell me the water consumed, nor does it respect the watering time.

kidekoop2016, 2024-10-16:
I an using z2m 1.40.2-1. HA Supervised updated (supervisor-2024.10.2).
Water consumed is always at 0. **It doesn't work.** 
Have you got any solution, please?
thanks

PacmanForever, 2024-10-16:
I haven't found any solution, everything remains the same.