Koenkk/zigbee2mqtt #17012: Zigbee2MQTT will not start in Boot up

ALPF5555, 2023-03-12:
### What happened?

When I tried to load the Running as a daemon with systemctl so that Zigbee2MQTT will run on boot up I do as in the steps 
When I go to show Status I get the listing below

simon@pi:~ $ sudo systemctl start zigbee2mqtt
simon@pi:~ $ systemctl status zigbee2mqtt.service
● zigbee2mqtt.service - zigbee2mqtt
     Loaded: loaded (/etc/systemd/system/zigbee2mqtt.service; enabled; vendor preset: enabled)
     Active: activating (auto-restart) (Result: exit-code) since Thu 2023-03-09 12:10:07 GMT; 4s ago
    Process: 1720 ExecStart=/usr/bin/npm start (code=exited, status=217/USER)
   Main PID: 1720 (code=exited, status=217/USER)
        CPU: 11ms

Mar 09 12:10:17 pi systemd[1]: zigbee2mqtt.service: Scheduled restart job, restart counter is at 12.
Mar 09 12:10:17 pi systemd[1]: Stopped zigbee2mqtt.
Mar 09 12:10:17 pi systemd[1]: Started zigbee2mqtt.
Mar 09 12:10:18 pi systemd[1733]: zigbee2mqtt.service: Failed to determine user credentials: No such process
Mar 09 12:10:18 pi systemd[1733]: zigbee2mqtt.service: Failed at step USER spawning /usr/bin/npm: No such process
Mar 09 12:10:18 pi systemd[1]: zigbee2mqtt.service: Main process exited, code=exited, status=217/USER
Mar 09 12:10:18 pi systemd[1]: zigbee2mqtt.service: Failed with result 'exit-code'.
Mar 09 12:10:28 pi systemd[1]: zigbee2mqtt.service: Scheduled restart job, restart counter is at 13.
Mar 09 12:10:28 pi systemd[1]: Stopped zigbee2mqtt.
Mar 09 12:10:28 pi systemd[1]: Started zigbee2mqtt.
Mar 09 12:10:28 pi systemd[1739]: zigbee2mqtt.service: Failed to determine user credentials: No such process
Mar 09 12:10:28 pi systemd[1739]: zigbee2mqtt.service: Failed at step USER spawning /usr/bin/npm: No such process
Mar 09 12:10:28 pi systemd[1]: zigbee2mqtt.service: Main process exited, code=exited, status=217/USER
Mar 09 12:10:28 pi systemd[1]: zigbee2mqtt.service: Failed with result 'exit-code'.
simon@pi:~ $ 

simon@pi:~ $ sudo mkdir
mkdir: missing operand
Try 'mkdir --help' for more information.
simon@pi:~ $ sudo chown
chown: missing operand
Try 'chown --help' for more information.
simon@pi:~ $

### What did you expect to happen?

I can start Zigbee2MQTT in the manual method and all works well I can see the Frontend and have installed a few devices can see reading on Temperature sensors etc. 

But when I try to load the Start up on Boot I get the errors above 

### How to reproduce it (minimal and precise)

I try to load the Autostart on Boot from step Running as a daemon with systemctl

I paste sudo nano /etc/systemd/system/zigbee2mqtt.service

Then see the following file 
[Unit]
Description=zigbee2mqtt
After=network.target

[Service]
Environment=NODE_ENV=production
ExecStart=/usr/bin/npm start
WorkingDirectory=/opt/zigbee2mqtt
StandardOutput=inherit
# Or use StandardOutput=null if you don't want Zigbee2MQTT messages filling syslog, for more options see systemd.exec(5)
StandardError=inherit
Restart=always
RestartSec=10s
User=pi

[Install]
WantedBy=multi-user.target

Then save the file as I should and after that I go to 
sudo systemctl start zigbee2mqtt

when I ask to show the status I get the error messages listed below.

simon@pi:~ $ sudo systemctl start zigbee2mqtt
simon@pi:~ $ systemctl status zigbee2mqtt.service
● zigbee2mqtt.service - zigbee2mqtt
     Loaded: loaded (/etc/systemd/system/zigbee2mqtt.service; enabled; vendor preset: enabled)
     Active: activating (auto-restart) (Result: exit-code) since Thu 2023-03-09 12:10:07 GMT; 4s ago
    Process: 1720 ExecStart=/usr/bin/npm start (code=exited, status=217/USER)
   Main PID: 1720 (code=exited, status=217/USER)
        CPU: 11ms

Mar 09 12:10:17 pi systemd[1]: zigbee2mqtt.service: Scheduled restart job, restart counter is at 12.
Mar 09 12:10:17 pi systemd[1]: Stopped zigbee2mqtt.
Mar 09 12:10:17 pi systemd[1]: Started zigbee2mqtt.
Mar 09 12:10:18 pi systemd[1733]: zigbee2mqtt.service: Failed to determine user credentials: No such process
Mar 09 12:10:18 pi systemd[1733]: zigbee2mqtt.service: Failed at step USER spawning /usr/bin/npm: No such process
Mar 09 12:10:18 pi systemd[1]: zigbee2mqtt.service: Main process exited, code=exited, status=217/USER
Mar 09 12:10:18 pi systemd[1]: zigbee2mqtt.service: Failed with result 'exit-code'.
Mar 09 12:10:28 pi systemd[1]: zigbee2mqtt.service: Scheduled restart job, restart counter is at 13.
Mar 09 12:10:28 pi systemd[1]: Stopped zigbee2mqtt.
Mar 09 12:10:28 pi systemd[1]: Started zigbee2mqtt.
Mar 09 12:10:28 pi systemd[1739]: zigbee2mqtt.service: Failed to determine user credentials: No such process
Mar 09 12:10:28 pi systemd[1739]: zigbee2mqtt.service: Failed at step USER spawning /usr/bin/npm: No such process
Mar 09 12:10:28 pi systemd[1]: zigbee2mqtt.service: Main process exited, code=exited, status=217/USER
Mar 09 12:10:28 pi systemd[1]: zigbee2mqtt.service: Failed with result 'exit-code'.
simon@pi:~ $ 

imon@pi:~ $ sudo mkdir
mkdir: missing operand
Try 'mkdir --help' for more information.
simon@pi:~ $ sudo chown
chown: missing operand
Try 'chown --help' for more information.
simon@pi:~ $

### Zigbee2MQTT version

1.29.0 commit: dddce571

### Adapter firmware version

20220219

### Adapter

SonOff version 3 Plus

EvertDekker, 2023-03-16:
https://www.shellhacks.com/systemd-service-not-starting-status217-user/

From your logs I see you are login with the user **Simon**.
In the zigbee2mqtt.service you are using the user pi. Does the user **pi** exist on your system?

ALPF5555, 2023-03-17:
Thank you for your comments I have been able to get the system to work and autoboot so everything is OK at this time

closed 2023-03-17 by ALPF5555

ScotsDon, 2024-02-02:
What did you do to solve this issue?