Koenkk/zigbee2mqtt #17019: [New device support]:

gid204, 2023-03-13:
### Link

http://raexmotor.com/english.php

### Database entry

{"id":43,"type":"Router","ieeeAddr":"0x84ba20fffe941229","nwkAddr":43358,"manufId":4098,"manufName":"_TZE200_tvltdfcj","powerSource":"Mains (single phase)","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[0,4,5,61184],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"Q��+\u0013���+\u0012���+\u0012","65506":31,"65508":0,"modelId":"TS0601","manufacturerName":"_TZE200_tvltdfcj","stackVersion":0,"dateCode":"","appVersion":68}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":68,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1678698452773,"defaultSendRequestWhen":"immediate"}

### Comments

I'm adding support for a new RAEX branded motor, this motor uses a TuYa Zigbee module.
I've used a TuYa Zigbee bridge to find all the datapoints of the device.
It has very similar datapoints to the following Zemismart device: [https://www.zigbee2mqtt.io/devices/ZM-AM02_cover.html](url)

I have forked and begun working on the device here: https://github.com/gid204/zigbee-herdsman-converters/tree/master

I have based my code off of the AM02 (I am not a programmer but have hobbyist experience).
The difference is the the AM02 uses an enum for the motor direction whereas my motor uses a boolean. I've confirmed this by connecting both devices up to the TuYa IOT platform.

In its current implementation from my fork, motor direction isn't working:
[image]

I'm hoping to get some guidance/support for implementing this functionality which I believe will be quite trivial for someone with experience.

Many thanks!

### External converter

```shell
I've added my converter directly within the /devices/tuya.js file.
```

gid204, 2023-03-23:
@Koenkk hoping you may be able to support me on this. Cheers!

Koenkk, 2023-03-23:
I suggest to switch to the new tuya data point api. This should make integration simpler. I've created a first external converter:

```js
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const utils = require('zigbee-herdsman-converters/lib/utils');
const globalStore = require('zigbee-herdsman-converters/lib/store');
const e = exposes.presets;
const ea = exposes.access;

const tzLocal = {
    datapoints: {
        key: ['position', 'state'],
        ...tuya.tzDataPoints,
    }
}

const definition = {
    fingerprint: tuya.fingerprint('TS0601', ['_TZE200_tvltdfcj']),
    model: 'TS0601_cover_3',
    vendor: 'TuYa',
    description: 'Curtain motor',
    whiteLabel: [
        {vendor: 'RAEX', model: 'TP30-2/26EC'},
    ],
    fromZigbee: [tuya.fzDataPoints],
    toZigbee: [tzLocal.datapoints],
    exposes: [
        e.cover_position().setAccess('position', ea.STATE_SET),
        exposes.binary('motor_direction', ea.STATE_SET, true, false),
        exposes.enum('limit_point', ea.STATE_SET, ['set_limit_up', 'set_limit_down', 'move_up_step', 'move_down_step']),
    ],
    meta: {
        tuyaDatapoints: [
            [1, 'state', tuya.valueConverterBasic.lookup({'CLOSE': tuya.enum(2), 'OPEN': tuya.enum(0), 'STOP': tuya.enum(1)})],
            [2, 'position', tuya.valueConverter.raw],
            [5, 'motor_direction', tuya.valueConverter.raw],
            [101, 'limit_point', tuya.valueConverterBasic.lookup({'set_limit_up': tuya.enum(0), 'set_limit_down': tuya.enum(1), 'move_up_step': tuya.enum(2), 'move_down_step': tuya.enum(3)})],
        ],
    },
};

module.exports = definition;
```
- save this as file next to `configuration.yaml` as `ext_converter.js`
- add it to `configuration.yaml`:
```yaml
external_converters:
  - ext_converter.js
```
- start z2m, check what works

gid204, 2023-03-24:
So using this external converter I can control the cover using the 'state' (open/close/stop).
However all other datapoints do not work, when I try to control them I get a no converter available error.

Koenkk, 2023-03-24:
I've updated https://github.com/Koenkk/zigbee2mqtt/issues/17019#issuecomment-1481459021, please check again.

gid204, 2023-03-24:
I think this was the line you changed:
```key: ['position', 'state', 'motor_direction', 'limit_point'],```

I actually had just tried that (still copied the whole updated code block and still not working).

gid204, 2023-03-24:
Just reading over the docs and wrapping my head around things...is this (the fact some datapoints aren't working) indicating that the value converters for DPs 2, 5 and 101 are likely incorrect?
In here:
```
    meta: {
        tuyaDatapoints: [
            [1, 'state', tuya.valueConverterBasic.lookup({'CLOSE': tuya.enum(2), 'OPEN': tuya.enum(0), 'STOP': tuya.enum(1)})],
            [2, 'position', tuya.valueConverter.raw],
            [5, 'motor_direction', tuya.valueConverter.raw],
            [101, 'limit_point', tuya.valueConverterBasic.lookup({'set_limit_up': tuya.enum(0), 'set_limit_down': tuya.enum(1), 'move_up_step': tuya.enum(2), 'move_down_step': tuya.enum(3)})],
        ],
    },
```

gid204, 2023-03-24:
Ok, needs a bit of a clean up but just wanted to re-post as I've gotten most of it working, but once again the motor direction is what I'm stuck on...

All functionality works currently except for the motor direction buttons. 
Similarly to the AM02 I don't believe this datapoint can be read therefore I think I'm correct in saying nothing needs to be added to fzlocal for this datapoint...please let me know if I've gotten that wrong.

```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const utils = require('zigbee-herdsman-converters/lib/utils');
const globalStore = require('zigbee-herdsman-converters/lib/store');
const e = exposes.presets;
const ea = exposes.access;

const tzLocal = {
    datapoints: {
        key: ['position', 'state', 'motor_direction', 'limit_point'],
        options: [exposes.options.invert_cover()],
        convertSet: async (entity, key, value, meta) => {
            if (key === 'position') {
                if (value >= 0 && value <= 100) {
                    const invert = tuya.isCoverInverted(meta.device.manufacturerName) ?
                        !meta.options.invert_cover : meta.options.invert_cover;

                    value = invert ? 100 - value : value;
                    await tuya.sendDataPointValue(entity, 2, value);
                } else {
                    throw new Error('TuYa_cover_control: Curtain motor position is out of range');
                }
            } else if (key === 'state') {
                const stateEnums = tuya.getCoverStateEnums(meta.device.manufacturerName);
                meta.logger.debug(`ZMAM02: Using state enums for ${meta.device.manufacturerName}:
                ${JSON.stringify(stateEnums)}`);
                value = value.toLowerCase();
                switch (value) {
                case 'close':
                    await tuya.sendDataPointEnum(entity, 1, stateEnums.close);
                    break;
                case 'open':
                    await tuya.sendDataPointEnum(entity, 1, stateEnums.open);
                    break;
                case 'stop':
                    await tuya.sendDataPointEnum(entity, 1, stateEnums.stop);
                    break;
                default:
                    throw new Error('ZMAM02: Invalid command received');
                }
            } else if (key == 'limit_point') {
                switch (value) {
                    case 'set_limit_up':
                        await tuya.sendDataPointEnum(entity, 101, 0);
                        break;
                    case 'set_limit_down':
                        await tuya.sendDataPointEnum(entity, 101, 1);
                        break;
                    case 'move_up_step':
                        await tuya.sendDataPointEnum(entity, 101, 2);
                        break;
                    case 'move_down_step':
                        await tuya.sendDataPointEnum(entity, 101, 3);
                        break;
                }
            }
            else if (key == 'motor_direction') {
                switch (value) {
                    case 'forward':
                        await tuya.sendDataPointBool(entity, 5, 0);
                        return {state: {motor_direction: value}};
                        break;
                    case 'backward':
                        await tuya.sendDataPointBool(entity, 5, 1);
                        return {state: {motor_direction: value}};
                        break; 
                }
            }
        }
    }
}

const fzLocal = {
    datapoints: {
        cluster: 'manuSpecificTuya',
        type: ['commandDataReport', 'commandDataResponse'],
        options: [exposes.options.invert_cover()],
        convert: (model, msg, publish, options, meta) => {
            const dpValue = tuya.firstDpValue(msg, meta, 'ZMAM02_cover');
            const dp = dpValue.dp;
            const value = tuya.getDataValue(dpValue);
            let result = null;
            switch (dp) {
            case 2: // Started moving to position (triggered from Zigbee)
            case 3: { // Arrived at position
                const running = dp === 3 ? false : true;
                const invert = tuya.isCoverInverted(meta.device.manufacturerName) ? !options.invert_cover : options.invert_cover;
                const position = invert ? 100 - (value & 0xFF) : (value & 0xFF);
                if (position > 0 && position <= 100) {
                    return {running, position, state: 'OPEN'};
                } else if (position == 0) { // Report fully closed
                    return {running, position, state: 'CLOSE'};
                } else {
                    return {running}; // Not calibrated yet, no position is available
                }
            }
        }
    }
}
}
const definition = {
    fingerprint: tuya.fingerprint('TS0601', ['_TZE200_tvltdfcj']),
    model: 'TS0601_cover_3',
    vendor: 'TuYa',
    description: 'Curtain motor',
    whiteLabel: [
        {vendor: 'RAEX', model: 'TP30-2/26EC'},
    ],
    fromZigbee: [fzLocal.datapoints],
    toZigbee: [tzLocal.datapoints],
    exposes: [
        e.cover_position().setAccess('position', ea.STATE_SET),
        exposes.binary('motor_direction', ea.STATE_SET, true, false),
        exposes.enum('limit_point', ea.STATE_SET, ['set_limit_up', 'set_limit_down', 'move_up_step', 'move_down_step']),
    ],
};

module.exports = definition;
```

Koenkk, 2023-03-24:
What if you use true/false instead of 0/1 (e.g. await tuya.sendDataPointBool(entity, 5, false);)

gid204, 2023-03-24:
> What if you use true/false instead of 0/1 (e.g. await tuya.sendDataPointBool(entity, 5, false);)

Unfortunately this also doesn't work

Koenkk, 2023-03-26:
If both boolean and enums are not working, then you probably need to sniff the traffic with the TuYa gateway to figure out what is really send. https://www.zigbee2mqtt.io/advanced/zigbee/04_sniff_zigbee_traffic.html

gid204, 2023-03-26:
Ack, thanks for the guidance, will try this.

closed 2023-05-04 by github-actions[bot]