Koenkk/zigbee2mqtt #17020: [New device support]: Tuya S10 dimmer not supported, manu and device model "unknown"

c0dr1ver, 2023-03-13:
### Link

https://www.aliexpress.com/item/10000015907761.html?spm=a2g0o.productlist.main.1.7b5065afynndsT&algo_pvid=f24f2cb8-ad06-4c90-98bd-d81c23b0ff72&algo_exp_id=f24f2cb8-ad06-4c90-98bd-d81c23b0ff72-0&pdp_ext_f=%7B%22sku_id%22%3A%2212000024694180744%22%7D&pdp_npi=3%40dis%21KHR%2187181.42%2152308.85%21%21%21%21%21%40211bd93f16787056673538424d06db%2112000024694180744%21sea%21KH%211743991859&curPageLogUid=8rB13jLbYxRk

### Database entry

{"id":25,"type":"Unknown","ieeeAddr":"0xa4c13841c276881e","nwkAddr":47790,"epList":[],"endpoints":{},"interviewCompleted":false,"meta":{},"lastSeen":1678526864718,"defaultSendRequestWhen":"immediate"}

### Comments

I tried to add TuYa zigbee dimmer to my network but interview process failed. No Manufacturer, Model name, Entry points to create own converter

[image]

[image]

Z2M version 1.30.2
Coordinator type zStack3x0
Coordinator revision 20211217

rzhivotov, 2023-03-17:
I have a same device and same problem, the interview process failed, the model and vendor are missing in the logs. Please, add support this device.

rzhivotov, 2023-03-18:
I connected the device to another software that works with the connection, Manufacturer: eWeLight, Model: ZB-DL01

Gambietz, 2023-04-07:
> I connected the device to another software that works with the connection, Manufacturer: eWeLight, Model: ZB-DL01

Hello, I have the same problem how can I interpret this with the other software

rzhivotov, 2023-04-08:
This is my external converter eWeLight-ZB-DL01.js
`const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const utils = require('zigbee-herdsman-converters/lib/utils');
const globalStore = require('zigbee-herdsman-converters/lib/store');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    fingerprint: [
        {
            modelID: 'ZB-DL01',
            manufacturerName: 'eWeLight',
            type: 'Router',
            //ieee_address: '0xa4c1387a2d710427',
        },
    ],
    zigbeeModel: ['ZB-DL01'],
    model: 'ZB-DL01',
    vendor: 'eWeLight',
    description: 'eWeLight ZB-DL01',
    extend: extend.light_onoff_brightness(),
    icon: 'data:image/png;base64,iVBORw0KGgoAAAANSUhEUgAAAQMAAAGTCAYAAAAst3eIAAAABmJLR0QAYABgAGBlFGZKAAAACXBIWXMAAC4jAAAuIwF4pT92AAAAB3RJTUUH5wQIFQUbGl5ThwAAIABJREFUeNrsvddvped1PbxO773w8JDDNsNpGk21JEuyHCtODEQ2jDQkN0ZuchPkIkAC5DKA/4cAufOF04A4yEUcJEYcufxkKbKKJU2RZjgkZ9iG5Gk8vbfvwt/aefjqPYV1yNF5AGKG5bT3fZ5d1l57bUO32+1itE7E4q3Q3pJutytfXAaDYdffqN+r/zcajaMLO1pDLfPoEpzs1el0dI0GDQMPfrfb/ZyB0DMs+3l9g8EgX3t9br33NFonc43cxjMcYRzl844CylFkMFon3AhoI4XDOLTaKGS0RsZgtE5wFKCXLhyWIVCNgDY6GBmIUZowWic4NRilCaM1igxGqcGhRgV60cfICIyMwWidQoNwWEahVyowShNGacJonfA0YbRGaxQZnPJFL6vn2Rmiq8Qk1TP3Ih2phCX+vdFohNFohMFgQKPR0H3tvVQQtIZIj5MwWiNjMFpHHdoZjZ878OqhJGlJSxziz00mk3zfi8w0WiNjMFqnLE3odDq7Dj8PuxoNqH/b7XZhMpl2sRy10YY2qtAzRqM1MgajdUKMgGoIeqUO6u/V/9NYaMN7vcigX2+EXsVhlCqMjMFoHeNqt9u7PLf2q9VqodPpoN1uo91ufy4dsFqtMJlMMJvNMJvNu6IHlcWol0LoGYlRhWFkDEbrKRoD9YCqHl2NArTGQDUW6vfaw66NELTRgfbnWvByZBhGxmC0jmnpdRWqaQCXyWTSze+JFzDC6HQ68rdao6A1BP34B6M0YWQMRuspYAZab9zpdCQ96HQ6Ah6qB5yRg9ls3hVB0IhYLJZdKcOg9zCKBkbGYLQO+WDrHS69sF9bDeh0OrsOdrfb/dzh11Ya1MPLEqP6WipOQMPAVINGRv29Gi1oMYZeeggjw3Fy1qgu9AVZ/Q6d1uD0MkyDooLRGhmD0TpFhqBXJUAbDWgJTHqPH/S3ozUyBqN1CgxBr9C93yEfFF2MDMIIMxitU24UBoX9xB/0IgPt84xwgJExGK1TghX0MwTaqKHX4VZ7HrSRQb9y5GiNjMFoneDoQOUe9Gpq0jMSepjDaI2MwWg9YynDQcP8UaowMgb73jhfpDB9r3/bq2GIX+12W2r82rbkdrst/++VLqgt0CpXgBEDf99ut3f9vNvtotVqfa4Tko/RNj/1+2yjNTIGo3UIS0sb1rYck67cq5FJLyo4jFx/REkeGYPROsJISa98ZzKZ5GdqRyIPtbYxSfulZQnuNdzvVZrspdY0WiNjMDrcQxymXtoBekagH0mI4Ty7EvUOu17TkWogVLrzXj6r3vCWQQ1No3U4aebIGDyDRmPQfAL1Z9ruRBoCVbSkV3ShipscRLewH1ehV0vzyBCMIoNRZLAPo6DXO9ArjNfqEtCz9+o/UEuL6s/3axj6GS89LGJkFEaRwWjtITro973aOaj+Tu1c7Le5tIKq+92Iegakn3biyBCMIoPROqBR0P5MqzmgtjoDgMVi0Y0atN9ro4L9eCQtXjACDE9ZFNI9RPOsx0bTC0N7/e2gMElvk1L2a79h+n7Q8r08/0F/r1Uu1n4NUiceBABqacV6n18rrKpKqKl/Q4UkVUjlaRuEg97/ow7j1chOr7Kjh78c1TU1H/cN2evwTnWz6l2EYTbzceb8h/38g/J3Vgv2e9iH3ezaNEKVXtfbwCoB6STn1Ee9f4a5//vBlY7ifZmPwxAcxBhokfPT5lkOQxxEb5bBcV0TbalRW63QYhZ71Tc46ZHDcZ+PYSXojyIVMx/XxT7Mcd56clzPAj4wTJqkhpbUKjzqw6AN99U0QBvm7pW9eNSR20HTwIP+fj/nYxAOM6zGxIlMEw6aU2mNgHpBBhmDk45Y72WzqbRiXotBmMBhyZZp1Zf1iE772ZgHNeYHDfOftjEYphW81z067KjKfBybvdfI8GFyH9XzHAVh5ahv9mEchl4zEQ7j/Q9jTLXsRFUUtRfAO+y9etqYwmFcn4Pef5VOrk0NeulDHIVWxLEYg4Pm3P2INYcFkB0VJnBQY6A3Lm0vGgKHudlVtiOrB/3EU47jsB80MnramIPePVWNvdp1epSG4NjShH5RwTCHbVAzzUkOAw8jZ9YaA226cNSeTW+oq9rO/DRTsqddLTosZ6BXiVHTwOPQiDg0noGW5qpatkHgyFFu5qe9WQZFLtrZhlqjt5fDflCQVu/xTxvtV8fHadOQYYyh3iRp1ZDpzZIYJl/fr+fvdc1V/oZatdE6Aa1WxGFGXkceGehdbL0c6KgAoqN+/EGNwaAIaC+bcT/GYNAQ1aMOo/dKedYesEGkMzojvT4PvQhrryH4QXpPej1e+96Oax2LMdAb+X1Yh+2gh3mQZT0OY9CrL2AvG3O/h7hXo9JhO4P9Xn+9KHIvn1Xb1andl1ojPCySf1Bj0auPQ69ZrBdudmqNgVaR57AAusOk+z6NnFKLIO/1tfVKUwdNEw6zjn3YpKi9ek01cuiFXfXCZg47MtBr5+41xGaYVO3UlRZ73bTDGrpxXGH8UV6fvXiQ/T7PMJ//KIzBYV3/Xqy7YZ1Br1TgaRiDfimgXnSg/dxH1fNxrKQjbZ18GOt+1LP+Tkpvw164A8OkByclMjiqPbQXY6IFH3sZTj1jcFiYkl40oxcZaI1AvzNyKklHvbCCkxAZPG1jMwgz2GtksB9vddig4V6u33E4g15lUD18YK9g4rCNZoMc3xcCM+iVL6n11JN8mJ91Y6DVQzhsg3BQAHEQljKIdETMQHvAe13zfmDiYe6/flyRvbJ0T40xIHCoR2NVG120N4F/d9A66iAE+mnzEAaNONtrHX2v3mPQpOWDfn49Us1eMKNBoPOwrz/M6/XjWehxHIDBLeSqQrXa4MWfqWdB+/qdTmcX6Uj9Uh93KjGDQYd1mMEb+3n+fr0Rh53DHrYxOMjrH4e4yLAMzl7GQC8yeZoYxHE8/qQ2z50YY6BHtT3oYTgOAOwwNpNevXvY59U+XuvRjqPRZq/XqNe4915Vp6d5/w7z9U+6GOyJMAb90N3D9N5HkYsdRu+DXhVBHYIy6DD2SnuGMSrHZSx71fUHqT2fVL2KYSs/WqB8P5oPXxhjoAeiPUtCmvuNbg46o3GvOfVRYSb9hr/wsOh9hqPW+9urd+41/2Evr3PSZ0aYT5Ih0OOPH/Rm9iudDSMoehqMzaCRbCfhMPW6/gc5YE/TmO9njw6akvWFNwYHRUiH8UyHScw57MM2aCbhfjzvXjzqUVVDeh1yPeOlBxwflxc97q7Xkzyu/kQZg/2UuYY9LEdJrDmKDbmXTaN3/Q4y/+Cw06B+f9ergsT/D+pKPIn3bhij+Ewbg145X7PZ3KWhrx0OCkD4B1TO6TUN6CA571H3q+936TXSqO9NbcHV86j9DsthpFmD8vj9dFWqh31Q6HzSq0HaiVS9IgG1UW8vZLte1/woWpyfWmQwGq01Ws/CGkRq61UpOonnwHich/4o+uWftU21l5RmdB1Pxn3rl44NSoO/0JjBadvAT2s8lx6XXs/THPX7O+rrc9rLyFqS3EGaxJ55Y9AL3Bq2hXm0ekdZo2t38pxFLyVvvUrJFzoyOI1h7XFFBsNSVXtFDM9qZHDS98wgDQiC4ifdEByrMTgNJb6TEG72Mgyja3VyjUE/tSS9wTInde8f2xTmUUg7ShOe5Xui13TVKxo4qUC6+TAvSq+5CVyc3ttsNtHpdGA2m3Vrrr0mDh9VatKvzfawGpH2W0MfNJJ+0HPo8Rj2a0T6XctepKFBNXW9z3eajJxeRKftPtUzCnoSgNp5Cc80ZvAsAEQjPGXwZzgolXgU9TydZRxdAvQUpTzqQ36cOMpx56v7wTlGeNIXADN4mh71IMNJ9yqVNShNGOQ9j+IwDMNjOKz7oE3r9iIoelRKVKM1igwO5MWOCr3vd9ifJa+4H8bdKCoYRQZPPTIYFD4fdh18kAz3YVcMBvEYDjM33w9mMEifcoQdjCKDp4IX9PvZceTPR+kdjxMz2GtkMIoKRpHBqVhH4TkHef0v4oEYVRCeIWPQ60YajcZdopDapg51MKtWkGOYvm+TySQzFrQhKp+fevXqDAe+J5UHoccDOKzxa72oqP14FodhrAY9/0E1Eg8zTduPMRzEU1DVs/TIP4ehZDSMOI9WvYn776A8lFFkcMDweJSPPjuY0Enba6d5b5lP02EeNj9VIw/tkA5t9HHStexHa39G/zgP4zCA6WnARZ7paoK2x1zPYBz14NbT7jlP+ud72riLHrfiaYX5I2PQY5LQaI3WSUgTep [... cut ...] zYYtMGgPR6AZRDdpOHv45s9bkVETfdGzUzqWR3R97Ysq/Y/YaiwnpsRJw2jfR3b4zPijoo25fupBoBGJcJx3YBGmzQqf1YvrBhPI46PqNpR1DWIuiPbFXCN/y3qXrRH2zJoj+2idgNhkLgLEf95p6XO27mORvLtrUqe2zJobTBoj12CQCP2P/59vZ/rlTnvZEPutEdg3O1oRU62RxsM2mOXp3F0s8cthUYgcS8tvuulDjfrVtyqV8In0VC0Pdpg8Ll1E+KbsFn3pTiBVy/j8F7eP7rhG/VLuBfwaY82GLTHLjdlnERsdSJvt8y41WZuFMloWwBtMGiPj2Hz7+T/64mO3K9raMQntMGgDQbt8QnyCY04g1bEXasMxXqhw2YWRjsD8TO8ltp5Bp/t8aCnr1XSUKs8gTYgtC2D9vgzdVPa47M7/n/GOq5Zqj8+BwAAAABJRU5ErkJggg=='
};

module.exports = definition; 
`

Gambietz, 2023-04-10:
Danke erstmal für die Angabe, aber was mache ich damit, könntest du mir das etwas genauer erläuter.
Es ist mühsam, wenn alles Neuland ist :-)

closed 2023-05-19 by github-actions[bot]

Rootbeat, 2023-12-09:
I would like to see support for the Tuya S10 as well. I would like to contribute myself if I would know how.

c0dr1ver, 2023-12-09:
> I would like to see support for the Tuya S10 as well. I would like to contribute myself if I would know how.

This device is already supported. Just update your zigbee coordinator firmware to latest version. 
[image]

BenJamesAndo, 2024-01-19:
I have this device and have added it in. Does anyone else experience that once the light value gets to around 20% that the device it's controlling (a light in my case) turns off? It would be good if we could set the minimal level of dimming, similar to what was mentioned [here](https://github.com/zigpy/zha-device-handlers/issues/1149).
[image]