Koenkk/zigbee2mqtt #17023: IKEA E1745 motion sensor not detecting movement after occupancy_timeout < 90s

jeanpascal395, 2023-03-13:
### What happened?

I paired my IKEA E1745 with Zigbee2MQTT and after I installed it I tested the function of the motion sensor. 
The detection is working as expected but changing occupancy_timeout in device settings (specific) only clears the occupancy of the device after the time in seconds written in the textfield. What's missing in my opinion is that the value also changes the time the sensor detecting again. For now the sensor always wait 90s and after that time its starts to detect again no matter what time was entered in occupancy_timeout. I can now move around in front of the sensor but nothing happens. Only after the 90 seconds have elapsed, a new movement is detected.

I'm also using a Xiaomi RTCGQ11LM and here the occupancy_timeout is working as expected.
If the problem is restricted to the hardware of the sensor I have to change to another manufacturer.

### What did you expect to happen?

Changing occupancy_timeout in device settings (specific) also changes the time the sensor detecting again.
For example "occupancy_timeout: 5s":
-  occupancy is cleared after 5s
- detecting of motions starts again after clearing process

### How to reproduce it (minimal and precise)

Pair a IKEA E1745 motion sensor to a SONOFF Zigbee 3.0 USB Dongle Plus coordiantor and change the occupancy_timeout to a value smaller 90s.

### Zigbee2MQTT version

1.30.2

### Adapter firmware version

20221226

### Adapter

SONOFF Zigbee 3.0 USB Dongle Plus

yunoh87, 2023-03-15:
Pretty sure that is a hardware limitation of the IKEA motion sensor with a 90 second cooldown period.

closed 2023-04-22 by github-actions[bot]

sammyke007, 2024-01-29:
I can confirm this...