Koenkk/zigbee2mqtt #17025: [New device support]: EGLO 900057

edalongeville, 2023-03-13:
### Link

https://www.eglo.com/en/ceiling-light-turcona-z-900057.html

### Database entry

{"id":25,"type":"Router","ieeeAddr":"0xa4c138f095bb1c2d","nwkAddr":54883,"manufId":4417,"manufName":"AwoX","powerSource":"Mains (single phase)","modelId":"TLSR82xx","epList":[1,242,3],"endpoints":{"1":{"profId":260,"epId":1,"devId":269,"inClusterList":[0,3,4,5,6,8,768,4096,64599,10],"outClusterList":[6],"clusters":{"genBasic":{"attributes":{"appVersion":0,"hwVersion":0}},"lightingColorCtrl":{"attributes":{"colorCapabilities":25,"colorTempPhysicalMin":153,"colorTempPhysicalMax":370}},"genOnOff":{"attributes":{"onOff":1}}},"binds":[],"configuredReportings":[],"meta":{}},"3":{"profId":4751,"epId":3,"devId":269,"inClusterList":[65360,65361],"outClusterList":[65360,65361],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":0,"stackVersion":2,"hwVersion":0,"swBuildId":"0122052017","zclVersion":3,"interviewCompleted":true,"meta":{"configured":88764544},"lastSeen":1678723460597,"defaultSendRequestWhen":"immediate"}

### Comments

This device is wrongly identified as an [AwoX 33951/33948](https://www.zigbee2mqtt.io/devices/33951_33948.html#awox-33951%252F33948) (see my answer in [this issue](https://github.com/Koenkk/zigbee2mqtt/issues/16691)).

I'm trying to create an external converter following [this documentation](https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html), but because I do not get the error `Device '0xa4c138f095bb1c2d' with Zigbee model 'some_model' and manufacturer name 'some_name' is NOT supported`, I do not know which value to set for the field `zigbeeModel`.

### Supported color modes

color temperature, RGB

### Color temperature range

{"colorCapabilities":25,"colorTempPhysicalMin":153,"colorTempPhysicalMax":370}

Koenkk, 2023-03-13:
This is actually a white label AwoX product, are all expected features supported?

edalongeville, 2023-03-13:
> This is actually a white label AwoX product, are all expected features supported?

No, it's missing RGB and Colour Temperature.
I believe this device should rather be identified as the [AwoX 33955](https://www.zigbee2mqtt.io/devices/33955.html), which seems to have all the expected features (and physically looks like the device I have).

edalongeville, 2023-03-14:
The situation has evolved. After updating Zigbee2MQTT, my device is now recognised as the [AwoX 33943/33944/33946](https://www.zigbee2mqtt.io/devices/33943_33944_33946.html).

It's better, but still not perfect.

Features that work:
 - on/off
 - brightness
 - colour temp: coolest (153), cool (250), warmest (370)
 - colour temp startup: coolest (153), cool (250), warmest (370), previous
 - color_xy: red, yellow, green, blue, purple
 - color_hs: red, yellow, green, blue, purple
 - effect: blink, okay, finish_effect, stop_effect
 - power_on_behavior
 - link_quality
 
 Features that don't work:
  - colour temp neutral: both neutral and warmest are 370
  - colour temp startup: same issue as with colour temp (neutral and warmest both 370)
  - color_xy orange: no change to the colour value when I click orange, the previous colour remains active
  - color_hs orange: same as color_xy
  - effect: breathe (its blinks repeatedly, but doesn't breathe), channel_change (turns off the light, no idea what the expected behaviour is)
  
I'm quite new to Zigbee2MQTT, but it looks to me like my device actually has different colour ranges (both white temp and RGB) from the device it's recognised as, causing this erratic behaviour when it comes to setting a specific colour.

Out of curiosity, I exposed the device to my Alexa integration in Home Assistant, and observed similar behaviours: The neutral white is the same as the warmest, and the colour displayed is different from the expected one (orange displays red, pink displays red, purple displays blue).

While the red and blue colours seem off, the green ones are surprisingly accurate (green, turquoise and cyan are all properly displayed).

Any advice on how I could troubleshoot this further?

Koenkk, 2023-03-14:
> Out of curiosity, I exposed the device to my Alexa integration in Home Assistant, and observed similar behaviours

I think the issues you are seeing are mainly due to inaccuracies of the bulb. Not something we ca fix from the z2m side.

edalongeville, 2023-03-15:
> > Out of curiosity, I exposed the device to my Alexa integration in Home Assistant, and observed similar behaviours
> 
> I think the issues you are seeing are mainly due to inaccuracies of the bulb. Not something we ca fix from the z2m side.

Seems likely.
The values between the database entry and the device it's recognised as are the same (`{"colorCapabilities":25,"colorTempPhysicalMin":153,"colorTempPhysicalMax":370}`).

Is the database entry something read directly from the device, unaffected by what the device is identified as ? If so, I'll proceed and close this issue.

Koenkk, 2023-03-15:
> Is the database entry something read directly from the device, unaffected by what the device is identified as ?

Yes, this is pulled from the device

closed 2023-03-15 by Koenkk