Koenkk/zigbee2mqtt #17027: Uncontrollable Ikea T2035

knudsvik, 2023-03-13:
### What happened?

The light is being paired and configured, however not possible to operate the light, not from z2m dashboard, nor from Home Assistant, see logs.  I have tried to force remove it and pair it a few times, also in other locations (e.g. next to controller).

### What did you expect to happen?

I expected that light is switched on and off. When I do this for other lights I see an `INFO` log detail, but not for this light.

### How to reproduce it (minimal and precise)

Pair the light, try to switch it on or off.

### Zigbee2MQTT version

1.30.2-dev commit: 461519c

### Adapter firmware version

0x26720700

### Adapter

ConBee2

### Debug log

[z2m.log](https://github.com/Koenkk/zigbee2mqtt/files/10959649/z2m.log)

knudsvik, 2023-04-13:
This is still a problem, should not be closed

knudsvik, 2023-05-15:
not stale, still issue

csuno, 2023-05-30:
same here
DresdenElectronic Conbee2 USB Stick

knudsvik, 2023-07-03:
Not stale

knudsvik, 2023-08-10:
not stale

knudsvik, 2023-09-10:
Still an issue

MattWestb, 2023-10-11:
ZHA with latest RCP have allays working OK and the light is very Zigbee 3 so shall not being any problem is the coordinator and host system is doing it right.

knudsvik, 2024-04-13:
Still an issue

unluckyquote, 2024-04-26:
> Still an issue

I was looking at buying one of these lights.  Is it still unusable with Z2MQTT?

psychogun, 2024-05-20:
I have the same issue with this device. 
It pairs just fine and picks up all the entities. In the little control pane in the "Interview Complete" section, if you are fairly quick; you are able to control this light. But when the interview process is "finally" done, it is not responding again; when I am trying to turn off / on, nothing happens. 

This is the error message I am experiencing:
```
Failed to call service light/turn_off. Failed to send request: Failed to deliver packet: <TXStatus.NWK_ROUTE_DISCOVERY_FAILED: 208>
```

Quick googling said that this could be an issue with interference (especially from WiFi 2.5GHz) -- However, everything else which is in the same vicinity is working. ..

I did a little experiment:
I am running Zigbee on Channel 15 which is 2425 MHz.
Zigbee has a bandwidth of 5MHz; therefore as the center frequency is 2425 MHz it is utilizing the band from 2422.5 MHz to 2427.5 MHz.

I am running 2.4GHz networks from several Access points throughout the house, 40MHz bandwidth. The 2.4GHz channels all overlap in to the Zigbee Channel 15 frequency. 

- I disabled alltogether the 2.4GHz network band.

Then I Paired the device once more.
Same applies; it works and is very responsive in the first few seconds in the Interview Complete window, then it "dies" and does not respond anymore > and I get the same error Message as stated above. 

Thus; I do not have any interference issues. 

```
STOFTMOLN ceiling/wall lamp WW37
by IKEA of Sweden
Firmware: 0x00010006
```

Where to look if one want to try to update the firmware of the lamp?

psychogun, 2024-07-27:
Anyone succesfull in using these lights? @knudsvik

knudsvik, 2024-07-27:
Nope, unfortunately not

psychogun, 2024-08-01:
I am not familiar with IKEA products, their smart-hub or whatnot-- but is there any way to maybe update the firmware on the device? I find it pecuilar that it works in the first span of the inclusion, but as you mentioned, strips, it somehow just stop functioning.

psychogun, 2024-11-20:
Hello. Got this working as intented. 

I had the original ConnBee dongle, but switched to Sonoff Zigbee 3.0 USB Dongle Plus and now they all work.

atopheim, 2025-01-11:
Was struggling to find the pairing rhythm of the lamp, but when it finally started flashing it paired easily and worked as intended 👍🏻

closed 2025-03-21 by github-actions[bot]