Koenkk/zigbee2mqtt #17030: Update not working:   error: Your local changes to the following files would be overwritten by merge:  package-lock.json Please commit your changes or stash them before you merge.

Dylantje, 2023-03-13:
### What happened?

When i try to update
I get a error:

error: Your local changes to the following files would be overwritten by merge:
        package-lock.json
Please commit your changes or stash them before you merge.

### What did you expect to happen?

my z2m update 

### Zigbee2MQTT version

    1.29.2 commit: bb3e8f6d

### Adapter firmware version

20221102

### Adapter

    zStack3x0

### Debug log

roidn2:~$ sudo systemctl status zigbee2mqtt
● zigbee2mqtt.service - zigbee2mqtt
     Loaded: loaded (/etc/systemd/system/zigbee2mqtt.service; enabled; vendor preset: enabled)
     Active: active (running) since Mon 2023-03-13 19:37:38 CET; 35s ago
   Main PID: 2193710 (npm start)
      Tasks: 23 (limit: 4067)
     Memory: 150.5M
        CPU: 9.951s
     CGroup: /system.slice/zigbee2mqtt.service
             ├─2193710 "npm start" "" "" "" "" "" "" "" "" "" "" "" "" "" ""
             ├─2193774 sh -c -- "node index.js"
             └─2193775 node index.js

mrt 13 19:37:45 odroidn2 npm[2193775]: Zigbee2MQTT:info  2023-03-13 19:37:45: MQTT publish: topic 'homeassistant/d>
mrt 13 19:37:45 odroidn2 npm[2193775]: Zigbee2MQTT:info  2023-03-13 19:37:45: MQTT publish: topic 'homeassistant/d>
mrt 13 19:37:45 odroidn2 npm[2193775]: Zigbee2MQTT:info  2023-03-13 19:37:45: MQTT publish: topic 'homeassistant/d>
mrt 13 19:37:45 odroidn2 npm[2193775]: Zigbee2MQTT:warn  2023-03-13 19:37:45: Failed to ping 'Lamp Gang' (attempt >
mrt 13 19:37:50 odroidn2 npm[2193775]: Zigbee2MQTT:warn  2023-03-13 19:37:50: Failed to ping 'Lamp Garage' (attemp>
mrt 13 19:37:55 odroidn2 npm[2193775]: Zigbee2MQTT:warn  2023-03-13 19:37:55: Failed to ping 'Lamp Zolder' (attemp>
mrt 13 19:38:00 odroidn2 npm[2193775]: Zigbee2MQTT:warn  2023-03-13 19:38:00: Failed to ping 'Spot Badk Douche rec>
mrt 13 19:38:06 odroidn2 npm[2193775]: Zigbee2MQTT:info  2023-03-13 19:38:06: MQTT publish: topic 'zigbee2mqtt/Spo>
mrt 13 19:38:08 odroidn2 npm[2193775]: Zigbee2MQTT:info  2023-03-13 19:38:08: MQTT publish: topic 'zigbee2mqtt/Spo>
mrt 13 19:38:08 odroidn2 npm[2193775]: Zigbee2MQTT:warn  2023-03-13 19:38:08: Failed to ping 'ST Lidll over' (atte>

pi@odroidn2:~$ sudo systemctl stop zigbee2mqtt
pi@odroidn2:~$ cd /opt/zigbee2mqtt
pi@odroidn2:/opt/zigbee2mqtt$ cp -R data data-backup
pi@odroidn2:/opt/zigbee2mqtt$ git pull
Updating bb3e8f6d..cdf62ea5
error: Your local changes to the following files would be overwritten by merge:
        package-lock.json
Please commit your changes or stash them before you merge.
Aborting
pi@odroidn2:/opt/zigbee2mqtt$ ^C
pi@odroidn2:/opt/zigbee2mqtt$ package-lock.json
-bash: package-lock.json: command not found
pi@odroidn2:/opt/zigbee2mqtt$ ^C
pi@odroidn2:/opt/zigbee2mqtt$ git checkout master
M       data/configuration.yaml
M       package-lock.json
Already on 'master'
Your branch is behind 'origin/master' by 114 commits, and can be fast-forwarded.
  (use "git pull" to update your local branch)
pi@odroidn2:/opt/zigbee2mqtt$ git pull
Updating bb3e8f6d..cdf62ea5
error: Your local changes to the following files would be overwritten by merge:
        package-lock.json
Please commit your changes or stash them before you merge.
Aborting
pi@odroidn2:/opt/zigbee2mqtt$ ^C
pi@odroidn2:/opt/zigbee2mqtt$ npm ci

added 752 packages, and audited 753 packages in 19s

67 packages are looking for funding
  run `npm fund` for details

found 0 vulnerabilities
pi@odroidn2:/opt/zigbee2mqtt$ ^C

junosuarez, 2023-03-14:
Unless you intentionally changed your package-lock.json for some reason and want to save those changes, the best thing to do is `git checkout package-lock.json` to discard your local changes and reset the file to its original state. Then you should be able to upgrade as expected.

closed 2023-04-21 by github-actions[bot]

redaDevCraft, 2024-06-30:
> Unless you intentionally changed your package-lock.json for some reason and want to save those changes, the best thing to do is `git checkout package-lock.json` to discard your local changes and reset the file to its original state. Then you should be able to upgrade as expected.

# yes this worked for me