Koenkk/zigbee2mqtt #17032: Unable to start with Systemd

Mr-Chimp, 2023-03-13:
### What happened?

I can run `npm start` and it works without issues but when I use systemd I get the below odd errors from `sudo journalctl -u zigbee2mqtt.service -f`

`Mar 13 20:45:09 home-automation.server npm[53258]:  7: 0x7f569fbb5a2e v8::Isolate::Initialize(v8::Isolate*, v8::Isolate::CreateParams const&) [/lib64/libnode.so.93]
Mar 13 20:45:09 home-automation.server npm[53258]:  8: 0x7f569f5c7e4f node::NodeMainInstance::NodeMainInstance(node::SnapshotData const*, uv_loop_s*, node::MultiIsolatePlatform*, std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > const&, std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > const&) [/lib64/libnode.so.93]
Mar 13 20:45:09 home-automation.server npm[53258]:  9: 0x7f569f53a681 node::Start(int, char**) [/lib64/libnode.so.93]
Mar 13 20:45:09 home-automation.server npm[53258]: 10: 0x7f569e6a8e50  [/lib64/libc.so.6]
Mar 13 20:45:09 home-automation.server npm[53258]: 11: 0x7f569e6a8efc __libc_start_main [/lib64/libc.so.6]
Mar 13 20:45:09 home-automation.server npm[53258]: 12: 0x5650a0ee10f5 _start [/usr/bin/node]
Mar 13 20:45:09 home-automation.server systemd-coredump[53265]: [🡕] Process 53258 (npm) of user 1001 dumped core.
Mar 13 20:45:09 home-automation.server systemd[1]: zigbee2mqtt.service: Main process exited, code=dumped, status=5/TRAP
Mar 13 20:45:09 home-automation.server systemd[1]: zigbee2mqtt.service: Failed with result 'core-dump'.
`

If it is useful the output of systemctl status zigbee2mqtt:
`[zig@home-automation zigbee2mqtt]$ sudo systemctl status zigbee2mqtt
● zigbee2mqtt.service - zigbee2mqtt
     Loaded: loaded (/etc/systemd/system/zigbee2mqtt.service; enabled; vendor preset: disabled)
     Active: activating (auto-restart) (Result: core-dump) since Mon 2023-03-13 20:44:59 GMT; 441ms ago
    Process: 53242 ExecStart=/usr/bin/npm start (code=dumped, signal=TRAP)
   Main PID: 53242 (code=dumped, signal=TRAP)
        CPU: 24ms
`

For now I am working around this by starting it in a screen but it would be good to find out what is going on. Everything else is working fine and as expected

### How to reproduce it (minimal and precise)

Just run sudo systemctl start zigbee2mqtt

### Zigbee2MQTT version

1.30.2 commit: cdf62ea

### Adapter firmware version

20220302

### Adapter

HamGeek POE Adapter

### Debug log

A log doesn't get created when this error occurs.

tollehenrik, 2023-03-17:
mar 17 22:12:44 iot systemd[1]: zigbee2mqtt.service: Consumed 1.950s CPU time.
mar 17 22:12:55 iot systemd[1]: zigbee2mqtt.service: Scheduled restart job, restart counter is at 141.
mar 17 22:12:55 iot systemd[1]: Stopped zigbee2mqtt.
mar 17 22:12:55 iot systemd[1]: zigbee2mqtt.service: Consumed 1.950s CPU time.
mar 17 22:12:55 iot systemd[1]: Started zigbee2mqtt.
mar 17 22:12:56 iot npm[5145]: > zigbee2mqtt@1.30.2 start
mar 17 22:12:56 iot npm[5145]: > node index.js
mar 17 22:12:56 iot npm[5157]: /opt/zigbee2mqtt/node_modules/rimraf/dist/cjs/src/rimraf-windows.js:41
mar 17 22:12:56 iot npm[5157]:         if (er?.code === 'ENOTEMPTY') {
mar 17 22:12:56 iot npm[5157]:                ^
mar 17 22:12:56 iot npm[5157]: SyntaxError: Unexpected token '.'
mar 17 22:12:56 iot npm[5157]:     at wrapSafe (internal/modules/cjs/loader.js:915:16)
mar 17 22:12:56 iot npm[5157]:     at Module._compile (internal/modules/cjs/loader.js:963:27)
mar 17 22:12:56 iot npm[5157]:     at Object.Module._extensions..js (internal/modules/cjs/loader.js:1027:10)
mar 17 22:12:56 iot npm[5157]:     at Module.load (internal/modules/cjs/loader.js:863:32)
mar 17 22:12:56 iot npm[5157]:     at Function.Module._load (internal/modules/cjs/loader.js:708:14)
mar 17 22:12:56 iot npm[5157]:     at Module.require (internal/modules/cjs/loader.js:887:19)
mar 17 22:12:56 iot npm[5157]:     at require (internal/modules/cjs/helpers.js:74:18)
mar 17 22:12:56 iot npm[5157]:     at Object.<anonymous> (/opt/zigbee2mqtt/node_modules/rimraf/dist/cjs/src/rimraf-manual.js:9:29)
mar 17 22:12:56 iot npm[5157]:     at Module._compile (internal/modules/cjs/loader.js:999:30)
mar 17 22:12:56 iot npm[5157]:     at Object.Module._extensions..js (internal/modules/cjs/loader.js:1027:10)
mar 17 22:12:56 iot systemd[1]: zigbee2mqtt.service: Main process exited, code=exited, status=1/FAILURE
mar 17 22:12:56 iot systemd[1]: zigbee2mqtt.service: Failed with result 'exit-code'.
mar 17 22:12:56 iot systemd[1]: zigbee2mqtt.service: Consumed 1.961s CPU time.

tollehenrik, 2023-03-17:
same issue here

Koenkk, 2023-03-18:
`if (er?.code === 'ENOTEMPTY') {` is caused by using an incompatible node version. Update node.js to fix the issue.

Mr-Chimp, 2023-03-18:
I have node.js version 16.18.1, I don't see the `if (er?.code === 'ENOTEMPTY') {` error

Koenkk, 2023-03-19:
@Mr-Chimp in your case it probably has to do with the system setup. Make sure you run z2m with the same user in the systemd script vs screen.

closed 2023-04-26 by github-actions[bot]