Koenkk/zigbee2mqtt #17033: [New device support]:  Aqara H1M wireles switch WXKG22LM (2 rockers)

norther-pit, 2023-03-13:
### Link

https://www.aqara.com/smart-wall-switch-h1m_overview

### Database entry

{"id":26,"type":"EndDevice","ieeeAddr":"0x54ef4410005942ee","nwkAddr":52872,"manufId":4447,"manufName":"LUMI","powerSource":"Battery","modelId":"lumi.remote.acn009","epList":[1,2,3],"endpoints":{"1":{"profId":260,"epId":1,"devId":0,"inClusterList":[0,1,3,18,64704],"outClusterList":[3,6,25],"clusters":{"genBasic":{"attributes":{"appVersion":7,"modelId":"lumi.remote.acn009","hwVersion":0}}},"binds":[],"configuredReportings":[],"meta":{}},"2":{"profId":260,"epId":2,"devId":0,"inClusterList":[18],"outClusterList":[6],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"3":{"profId":260,"epId":3,"devId":0,"inClusterList":[18],"outClusterList":[6],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":7,"stackVersion":2,"hwVersion":0,"zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1678742697525,"defaultSendRequestWhen":"immediate"}

### Comments

unable to make work new device.
was trying with external converter, using definition from lumi.remote.acn007, but only thing I get was battery % indication.

database entry is after repairing.
External converter code is what I was using.

I'm usign z2m as an addon on my Home Assistant Generic x86-64 installation.

### External converter

```shell
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
        zigbeeModel: ['lumi.remote.acn009'],
        model: 'WXKG22LM',
        vendor: 'Xiaomi',
        description: 'Aqara H1M wireless switch',
        meta: {battery: {voltageToPercentage: '3V_2850_3000'}},
        exposes: [e.battery(), e.battery_voltage(),
            e.action(['single_left', 'single_right', 'single_both', 'double_left', 'double_right', 'hold_left', 'hold_right']),
            // eslint-disable-next-line max-len
            exposes.enum('click_mode', ea.ALL, ['fast', 'multi']).withDescription('Click mode, fast: only supports single click which will be send immediately after clicking, multi: supports more events like double and hold'),
        ],
        fromZigbee: [fz.xiaomi_multistate_action, fz.aqara_opple],
        toZigbee: [tz.xiaomi_switch_click_mode],
        
};

module.exports = definition;
```

ShadowFistOkami, 2023-04-02:
@norther-pit I'm no expert, but I think I see 2 problems
1) `model: 'WXKG22LM'` which isn't found anywhere in fz.xiaomi_multistate_action or tz.xiaomi_switch_click_mode. You'll need to add a local copy of the section of the libs in your converter & add your model there.

2) You're basing your definition on lumi.remote.acn007, which is a single button (WXKG12LM). I'd save myself the headache & base it off the wireless H1 double switch (WXKG15LM) since that supports left, right & both buttons pressed

Search zigbee herdsman for any reference to WXKG15LM in tx & fz - those are the parts you need to use & replace with your model.
Here's the xiaomi.js from devices to get you started:
```    {
        zigbeeModel: ['lumi.remote.b28ac1'],
        model: 'WXKG15LM',
        vendor: 'Xiaomi',
        whiteLabel: [{vendor: 'Xiaomi', model: 'WRS-R02'}],
        description: 'Aqara wireless remote switch H1 (double rocker)',
        fromZigbee: [fz.battery, fz.xiaomi_multistate_action, fz.aqara_opple, fz.command_toggle],
        toZigbee: [tz.xiaomi_switch_click_mode, tz.aqara_opple_operation_mode],
        meta: {battery: {voltageToPercentage: '3V_2850_3000'}, multiEndpoint: true},
        exposes: [
            e.battery(), e.battery_voltage(), e.action([
                'single_left', 'single_right', 'single_both',
                'double_left', 'double_right', 'double_both',
                'triple_left', 'triple_right', 'triple_both',
                'hold_left', 'hold_right', 'hold_both']),
            exposes.enum('click_mode', ea.ALL, ['fast', 'multi'])
                .withDescription('Click mode, fast: only supports single click which will be send immediately after clicking.' +
                    'multi: supports more events like double and hold'),
            exposes.enum('operation_mode', ea.ALL, ['command', 'event'])
                .withDescription('Operation mode, select "command" to enable bindings (wake up the device before changing modes!)'),
        ],
        configure: async (device, coordinatorEndpoint, logger) => {
            const endpoint1 = device.getEndpoint(1);
            const endpoint2 = device.getEndpoint(3);
            // set "event" mode
            await endpoint1.write('aqaraOpple', {'mode': 1}, {manufacturerCode: 0x115f});
            // turn on the "multiple clicks" mode, otherwise the only "single click" events.
            // if value is 1 - there will be single clicks, 2 - multiple.
            await endpoint1.write('aqaraOpple', {0x0125: {value: 0x02, type: 0x20}}, {manufacturerCode: 0x115f});
            await reporting.bind(endpoint1, coordinatorEndpoint, ['genOnOff', 'genPowerCfg']);
            await reporting.bind(endpoint2, coordinatorEndpoint, ['genOnOff']);
            // TODO/BUG:
            // Did not understand how to separate the left and right keys in command mode -
            // the "toggleCommand" always arrives from the first endpoint
        },
    }
```

Simperdy, 2023-05-03:
Hey guys,

I think I have a similar issue.
I bought the Xiaomi H1M (with neutral, tripple rocker) "ZNQBKG26LM" and made my local converter file based on "ZNQBKG31LM" (Aqara E1 3 gang switch (with neutral)".
So far so good, I can already use a lot of the switch, but I only get "action: single" or "action: double" when pressing any rocker on the switch. No information about which rocker (left, center, right).
Do I also need to create local copy of the libs "fz.xiaomi_multistate_action" and "tz.xiaomi_switch_click_mode"? Where do I find them here on gitlab? Where do I need to put them localy?

If you need anymore information, please ask.
It is my first time adding an unsupported device to z2m.

best regards
Simperdy

closed 2023-06-10 by github-actions[bot]

jomud9, 2023-06-22:
> Update from me:
> 
> I think I got it right now. As you explained I added the snipped to my already created converter and used that instead of the fromZigbee.js. Thanks for pointing me into the right direction!

Would like to get some inspiration how to add the related code into external converter? I am working on this and I think I hit the same wall.

https://github.com/Koenkk/zigbee2mqtt/issues/18082

jfcabral, 2023-10-19:
Hi, I've just received my Aqara H1M wireless switch with 2 buttons.

Unfortunately, it seems it is still not supported by the latest available Zigbee2mqtt version ([1.33.1](https://github.com/Koenkk/zigbee2mqtt/releases/tag/1.33.1) commit: [7e63039](https://github.com/Koenkk/zigbee2mqtt/commit/7e63039)) :/ 

This is what I get from Zigbee2mqtt;
[image]

The button is just like this one: https://www.zigbee2mqtt.io/devices/ZNQBKG25LM.html with teh difference that it is wireless.

What can be done to support it?

Thank you

xriser, 2023-11-06:
> As you explained I added the snipped to my already created converter and used that instead of the fromZigbee.js.
> Thanks for pointing me into the right direction!

I have a workaround here: just put the device model: 'WXKG15LM' into your convertor instead of copying pieces of additional code.
It will be discovered as H1 but keys will be working properly.
final converter code:
`const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
        zigbeeModel: ['lumi.remote.acn009'],
        model: 'WXKG15LM',
        vendor: 'Xiaomi',
        whiteLabel: [{vendor: 'Xiaomi', model: 'WRS-R02'}],
        description: 'Aqara wireless remote switch H1 (double rocker)',
        fromZigbee: [fz.battery, fz.xiaomi_multistate_action, fz.aqara_opple, fz.command_toggle],
        toZigbee: [tz.xiaomi_switch_click_mode, tz.aqara_opple_operation_mode],
        meta: {battery: {voltageToPercentage: '3V_2850_3000'}, multiEndpoint: true},
        exposes: [
            e.battery(), e.battery_voltage(), e.action([
                'single_left', 'single_right', 'single_both',
                'double_left', 'double_right', 'double_both',
                'triple_left', 'triple_right', 'triple_both',
                'hold_left', 'hold_right', 'hold_both']),
            e.enum('click_mode', ea.ALL, ['fast', 'multi'])
                .withDescription('Click mode, fast: only supports single click which will be send immediately after clicking.' +
                    'multi: supports more events like double and hold'),
            e.enum('operation_mode', ea.ALL, ['command', 'event'])
                .withDescription('Operation mode, select "command" to enable bindings (wake up the device before changing modes!)'),
        ],
        configure: async (device, coordinatorEndpoint, logger) => {
            const endpoint1 = device.getEndpoint(1);
            const endpoint2 = device.getEndpoint(3);
            // set "event" mode
            await endpoint1.write('aqaraOpple', {'mode': 1}, {manufacturerCode: 0x115f});
            // turn on the "multiple clicks" mode, otherwise the only "single click" events.
            // if value is 1 - there will be single clicks, 2 - multiple.
            await endpoint1.write('aqaraOpple', {0x0125: {value: 0x02, type: 0x20}}, {manufacturerCode: 0x115f});
            await reporting.bind(endpoint1, coordinatorEndpoint, ['genOnOff', 'genPowerCfg']);
            await reporting.bind(endpoint2, coordinatorEndpoint, ['genOnOff']);
            // TODO/BUG:
            // Did not understand how to separate the left and right keys in command mode -
            // the "toggleCommand" always arrives from the first endpoint
        },
};

module.exports = definition;`

Koenkk, 2023-11-06:
@xriser could you  make a PR for this?

xriser, 2023-11-15:
https://github.com/Koenkk/zigbee-herdsman-converters/pull/6481

Manic-Marley, 2024-06-27:
Im a complete noob and will be setting up Z2M this weekend for the first time so i know nothing about everything, ive just bought the H1M 3 switch and i was wondering if someone could do a Dummy guide step by step on what i need to do to get the switch set up so i can use all 3 switches and decouple options. Is that something someone could do?

baradetsky, 2024-12-28:
The device has been added, the wireless H1M, but no action events are taking place. Errors appear when trying to change click mode or operation mode.
[image]
[image]