Koenkk/zigbee2mqtt #17034: TS0505B - Brightness not working

Hexagram6716, 2023-03-13:
### What happened?

So far everything works except the brightness on the device.

for more info the device is this

https://www.aliexpress.com/item/1005005122545123.html?spm=a2g0o.order_list.order_list_main.17.2a991802efXkC5

shows as 

Device type
Router
Zigbee Model
TS0505B
Zigbee Manufacturer
_TZ3210_it1u8ahz
Description
Zigbee RGB+CCT light
Support status
Supported

Firmware version
z.1.0
Manufacturer
[TuYa](https://www.zigbee2mqtt.io/supported-devices/#v=TuYa)
Model
[TS0505B](https://www.zigbee2mqtt.io/devices/TS0505B.html#tuya-ts0505b)
Power
Interview completed
True

I can see brightness up to t254 but moving the slider dose not change anything

Please update soon 

thanks

### Zigbee2MQTT version

1.30.2

### Adapter firmware version

20211115

### Adapter

CC2531

### Debug log

Debug 2023-03-13 21:47:14Received MQTT message on 'homeassistant/sensor/OMEN/OMEN_lastactive/state' with data '2023-03-13 21:47:14Z'
Debug 2023-03-13 21:47:14Received MQTT message on 'zigbee2mqtt/Bedside Lamp Right/set' with data '{"brightness":55}'
Debug 2023-03-13 21:47:14Publishing 'set' 'brightness' to 'Bedside Lamp Right'
Info 2023-03-13 21:47:15MQTT publish: topic 'zigbee2mqtt/Bedside Lamp Right', payload '{"brightness":55,"color":{"x":0.1355,"y":0.0399},"color_mode":"xy","color_power_on_behavior":null,"color_temp":500,"device":{"applicationVersion":112,"friendlyName":"Bedside Lamp Right","hardwareVersion":0,"ieeeAddr":"0xa4c1384aa5d8437f","manufacturerID":4417,"manufacturerName":"_TZ3210_it1u8ahz","model":"TS0505B","networkAddress":32840,"powerSource":"Mains (single phase)","softwareBuildID":"z.1.0","stackVersion":2,"type":"Router","zclVersion":3},"do_not_disturb":null,"linkquality":115,"state":"ON"}'
Debug 2023-03-13 21:47:15Received MQTT message on 'zigbee2mqtt/Bedside Lamp Right/set' with data '{"brightness":125}'
Debug 2023-03-13 21:47:15Publishing 'set' 'brightness' to 'Bedside Lamp Right'
Info 2023-03-13 21:47:15MQTT publish: topic 'zigbee2mqtt/Bedside Lamp Right', payload '{"brightness":125,"color":{"x":0.1355,"y":0.0399},"color_mode":"xy","color_power_on_behavior":null,"color_temp":500,"device":{"applicationVersion":112,"friendlyName":"Bedside Lamp Right","hardwareVersion":0,"ieeeAddr":"0xa4c1384aa5d8437f","manufacturerID":4417,"manufacturerName":"_TZ3210_it1u8ahz","model":"TS0505B","networkAddress":32840,"powerSource":"Mains (single phase)","softwareBuildID":"z.1.0","stackVersion":2,"type":"Router","zclVersion":3},"do_not_disturb":null,"linkquality":115,"state":"ON"}'
Debug 2023-03-13 21:47:15Received MQTT message on 'zigbee2mqtt/Bedside Lamp Right/set' with data '{"brightness":177}'
Debug 2023-03-13 21:47:15Publishing 'set' 'brightness' to 'Bedside Lamp Right'
Info 2023-03-13 21:47:15MQTT publish: topic 'zigbee2mqtt/Bedside Lamp Right', payload '{"brightness":177,"color":{"x":0.1355,"y":0.0399},"color_mode":"xy","color_power_on_behavior":null,"color_temp":500,"device":{"applicationVersion":112,"friendlyName":"Bedside Lamp Right","hardwareVersion":0,"ieeeAddr":"0xa4c1384aa5d8437f","manufacturerID":4417,"manufacturerName":"_TZ3210_it1u8ahz","model":"TS0505B","networkAddress":32840,"powerSource":"Mains (single phase)","softwareBuildID":"z.1.0","stackVersion":2,"type":"Router","zclVersion":3},"do_not_disturb":null,"linkquality":115,"state":"ON"}'
Debug 2023-03-13 21:47:16Received MQTT message on 'homeassistant/sensor/DESKTOP-UH1KREP-satellite/availability' with data 'online'
Debug 2023-03-13 21:47:16Received MQTT message on 'zigbee2mqtt/Bedside Lamp Right/set' with data '{"brightness":220}'
Debug 2023-03-13 21:47:16Publishing 'set' 'brightness' to 'Bedside Lamp Right'
Info 2023-03-13 21:47:16MQTT publish: topic 'zigbee2mqtt/Bedside Lamp Right', payload '{"brightness":220,"color":{"x":0.1355,"y":0.0399},"color_mode":"xy","color_power_on_behavior":null,"color_temp":500,"device":{"applicationVersion":112,"friendlyName":"Bedside Lamp Right","hardwareVersion":0,"ieeeAddr":"0xa4c1384aa5d8437f","manufacturerID":4417,"manufacturerName":"_TZ3210_it1u8ahz","model":"TS0505B","networkAddress":32840,"powerSource":"Mains (single phase)","softwareBuildID":"z.1.0","stackVersion":2,"type":"Router","zclVersion":3},"do_not_disturb":null,"linkquality":115,"state":"ON"}'
Debug 2023-03-13 21:47:16Received MQTT message on 'homeassistant/sensor/OMEN/OMEN_lastactive/state' with data '2023-03-13 21:47:16Z'
Debug 2023-03-13 21:47:17Received MQTT message on 'zigbee2mqtt/Bedside Lamp Right/set' with data '{"brightness":10}'
Debug 2023-03-13 21:47:17Publishing 'set' 'brightness' to 'Bedside Lamp Right'
Info 2023-03-13 21:47:17MQTT publish: topic 'zigbee2mqtt/Bedside Lamp Right', payload '{"brightness":10,"color":{"x":0.1355,"y":0.0399},"color_mode":"xy","color_power_on_behavior":null,"color_temp":500,"device":{"applicationVersion":112,"friendlyName":"Bedside Lamp Right","hardwareVersion":0,"ieeeAddr":"0xa4c1384aa5d8437f","manufacturerID":4417,"manufacturerName":"_TZ3210_it1u8ahz","model":"TS0505B","networkAddress":32840,"powerSource":"Mains (single phase)","softwareBuildID":"z.1.0","stackVersion":2,"type":"Router","zclVersion":3},"do_not_disturb":null,"linkquality":115,"state":"ON"}'

Dimmers81, 2023-03-17:
My TS0505B lights also stopped being able to be dimmed in RGB but colour temp is dimmable for some reason.

Hexagram6716, 2023-03-18:
Hopefully they fix it soon

closed 2023-03-18 by Hexagram6716

bjoe10, 2023-08-01:
same problem with RGB

Dimmers81, 2023-08-01:
It's weird because while RGB has this issue, colour temp works fine (for me, anyway). Also just updated to the latest version of MQTT this morning and still not working :(