Koenkk/zigbee2mqtt #17036: [New device support]: IKUU Mercator Dimmer Mech SSWM-DIMZ

cbriddell, 2023-03-14:
### Link

https://www.ikuu.com.au/product/switch-mechanism-3/

### Database entry

{"id":7,"type":"Router","ieeeAddr":"0xcc86ecfffe9a1278","nwkAddr":61522,"manufId":4098,"manufName":"_TZE200_9cxuhakf","powerSource":"Mains (single phase)","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[0,4,5,61184],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"modelId":"TS0601","manufacturerName":"_TZE200_9cxuhakf","powerSource":1,"zclVersion":3,"appVersion":64,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":64,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1678770305098,"defaultSendRequestWhen":"immediate"}

### Comments

I created external converter and can't understand what I should do next.

This is a Mercator Ikuü Dimmer (in wall mechanism) SSWM-DIMZ 

### External converter

```shell
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const definition = {
    // Since a lot of TuYa devices use the same modelID, but use different datapoints
    // it's necessary to provide a fingerprint instead of a zigbeeModel
    fingerprint: [
        {
            // The model ID from: Device with modelID 'TS0601' is not supported
            // You may need to add \u0000 at the end of the name in some cases
            modelID: 'TS0601',
            // The manufacturer name from: Device with modelID 'TS0601' is not supported.
            manufacturerName: '_TZE200_9cxuhakf',
        },
    ],
    model: 'TS0601_new',
    vendor: 'TuYa',
    description: 'Zigbee Smart Dimmer',
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [tuya.tz.datapoints],
    onEvent: tuya.onEventSetTime, // Add this if you are getting no converter for 'commandMcuSyncTime'
    configure: tuya.configureMagicPacket,
    exposes: [
        // Here you should put all functionality that your device exposes
    ],
    meta: {
        // All datapoints go in here
        tuyaDatapoints: [
        ],
    },
};

module.exports = definition;
```

cbriddell, 2023-03-14:
I tried this https://github.com/Koenkk/zigbee2mqtt/issues/13174 and am now getting on, off and brightness working. The power on behaviour is giving an error saying 'no converter available'.  So is Light Type and Countdown...?

Have extracted the following from logs if this helps... Thanks so much :)

debug 2023-03-14 16:42:11: Received Zigbee message from '0xcc86ecfffe9a1278', type 'read', cluster 'genTime', data '["localTime"]' from endpoint 1 with groupID 0
debug 2023-03-14 16:42:12: Received MQTT message on 'zigbee2mqtt/bridge/request/device/rename' with data '{"from":"0xcc86ecfffe9a1278","homeassistant_rename":true,"to":"Landing Ceiling","transaction":"2xzk2-2"}'
info  2023-03-14 16:42:13: MQTT publish: topic 'zigbee2mqtt/0xcc86ecfffe9a1278', payload ''
info  2023-03-14 16:42:13: MQTT publish: topic 'zigbee2mqtt/0xcc86ecfffe9a1278/availability', payload 'null'
info  2023-03-14 16:42:13: MQTT publish: topic 'zigbee2mqtt/Landing Ceiling/availability', payload '{"state":"online"}'
debug 2023-03-14 16:42:13: Refreshing Home Assistant discovery topic for 'Landing Ceiling'
info  2023-03-14 16:42:13: MQTT publish: topic 'homeassistant/light/0xcc86ecfffe9a1278/light/config', payload 'null'
info  2023-03-14 16:42:13: MQTT publish: topic 'homeassistant/sensor/0xcc86ecfffe9a1278/power_on_behavior/config', payload 'null'
info  2023-03-14 16:42:13: MQTT publish: topic 'homeassistant/select/0xcc86ecfffe9a1278/power_on_behavior/config', payload 'null'
info  2023-03-14 16:42:13: MQTT publish: topic 'homeassistant/sensor/0xcc86ecfffe9a1278/countdown/config', payload 'null'
info  2023-03-14 16:42:13: MQTT publish: topic 'homeassistant/number/0xcc86ecfffe9a1278/countdown/config', payload 'null'
info  2023-03-14 16:42:13: MQTT publish: topic 'homeassistant/sensor/0xcc86ecfffe9a1278/light_type/config', payload 'null'
info  2023-03-14 16:42:13: MQTT publish: topic 'homeassistant/select/0xcc86ecfffe9a1278/light_type/config', payload 'null'
info  2023-03-14 16:42:13: MQTT publish: topic 'homeassistant/sensor/0xcc86ecfffe9a1278/linkquality/config', payload 'null'
info  2023-03-14 16:42:13: MQTT publish: topic 'zigbee2mqtt/Landing Ceiling', payload '{"brightness":3,"countdown":null,"level":39,"light_type":null,"linkquality":109,"power_on_behavior":null,"state":"OFF"}'
info  2023-03-14 16:42:13: MQTT publish: topic 'zigbee2mqtt/Landing Ceiling/type', payload '{"light":null}'
info  2023-03-14 16:42:13: MQTT publish: topic 'zigbee2mqtt/bridge/response/device/rename', payload '{"data":{"from":"0xcc86ecfffe9a1278","homeassistant_rename":true,"to":"Landing Ceiling"},"status":"ok","transaction":"2xzk2-2"}'
debug 2023-03-14 16:42:13: Received MQTT message on 'homeassistant/light/0xcc86ecfffe9a1278/light/config' with data ''
debug 2023-03-14 16:42:13: Received MQTT message on 'homeassistant/sensor/0xcc86ecfffe9a1278/power_on_behavior/config' with data ''
debug 2023-03-14 16:42:13: Received MQTT message on 'homeassistant/select/0xcc86ecfffe9a1278/power_on_behavior/config' with data ''
debug 2023-03-14 16:42:13: Received MQTT message on 'homeassistant/sensor/0xcc86ecfffe9a1278/countdown/config' with data ''
debug 2023-03-14 16:42:13: Received MQTT message on 'homeassistant/number/0xcc86ecfffe9a1278/countdown/config' with data ''
debug 2023-03-14 16:42:13: Received MQTT message on 'homeassistant/sensor/0xcc86ecfffe9a1278/light_type/config' with data ''
debug 2023-03-14 16:42:13: Received MQTT message on 'homeassistant/select/0xcc86ecfffe9a1278/light_type/config' with data ''
debug 2023-03-14 16:42:13: Received MQTT message on 'homeassistant/sensor/0xcc86ecfffe9a1278/linkquality/config' with data ''
info  2023-03-14 16:42:15: MQTT publish: topic 'homeassistant/light/0xcc86ecfffe9a1278/light/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"},{"topic":"zigbee2mqtt/Landing Ceiling/availability","value_template":"{{ value_json.state }}"}],"availability_mode":"all","brightness":true,"brightness_scale":254,"command_topic":"zigbee2mqtt/Landing Ceiling/set","device":{"identifiers":["zigbee2mqtt_0xcc86ecfffe9a1278"],"manufacturer":"TuYa","model":"Zigbee smart dimmer (TS0601_dimmer)","name":"Landing Ceiling"},"name":"Landing Ceiling","schema":"json","state_topic":"zigbee2mqtt/Landing Ceiling","unique_id":"0xcc86ecfffe9a1278_light_zigbee2mqtt"}'
info  2023-03-14 16:42:15: MQTT publish: topic 'homeassistant/sensor/0xcc86ecfffe9a1278/power_on_behavior/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"},{"topic":"zigbee2mqtt/Landing Ceiling/availability","value_template":"{{ value_json.state }}"}],"availability_mode":"all","device":{"identifiers":["zigbee2mqtt_0xcc86ecfffe9a1278"],"manufacturer":"TuYa","model":"Zigbee smart dimmer (TS0601_dimmer)","name":"Landing Ceiling"},"enabled_by_default":false,"entity_category":"config","icon":"mdi:power-settings","name":"Landing Ceiling power on behavior","state_topic":"zigbee2mqtt/Landing Ceiling","unique_id":"0xcc86ecfffe9a1278_power_on_behavior_zigbee2mqtt","value_template":"{{ value_json.power_on_behavior }}"}'
info  2023-03-14 16:42:15: MQTT publish: topic 'homeassistant/select/0xcc86ecfffe9a1278/power_on_behavior/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"},{"topic":"zigbee2mqtt/Landing Ceiling/availability","value_template":"{{ value_json.state }}"}],"availability_mode":"all","command_topic":"zigbee2mqtt/Lan [... cut ...] device":{"identifiers":["zigbee2mqtt_0xcc86ecfffe9a1278"],"manufacturer":"TuYa","model":"Zigbee smart dimmer (TS0601_dimmer)","name":"Landing Ceiling"},"enabled_by_default":false,"entity_category":"diagnostic","icon":"mdi:signal","name":"Landing Ceiling linkquality","state_class":"measurement","state_topic":"zigbee2mqtt/Landing Ceiling","unique_id":"0xcc86ecfffe9a1278_linkquality_zigbee2mqtt","unit_of_measurement":"lqi","value_template":"{{ value_json.linkquality }}"}'
debug 2023-03-14 16:42:15:

closed 2023-04-21 by github-actions[bot]

loopback007, 2023-06-05:
I have 2 of these switches installed and successfully made an external converter to handle on/off toggles and brightness. 
However, I suspect with the recent update to Z2M, external converters don't work and I'd imagine the "manufName":"_TZE200_9cxuhakf" would need to be added to the existing database for TS0601_dimmer. 

Below is the converter I mashed up for a different TS0601_dimmer which works. You would need to add an external adapter line to the Z2M configuration file so that Z2M knows which external adapter files to refer to. 

**TS0601_dimmer.js**

`const fz = {...require('zigbee-herdsman-converters/converters/fromZigbee'), legacy: require('zigbee-herdsman-converters/lib/legacy').fromZigbee};
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const utils = require('zigbee-herdsman-converters/lib/utils');

const tzLocal = {
    tuya_dimmer_level: {
        key: ['brightness_min', 'brightness', 'brightness_percent', 'level'],
        convertSet: async (entity, key, value, meta) => {
            // upscale to 1000
            let newValue;
            let dp = tuya.dataPoints.dimmerLevel;
            if (meta.device.manufacturerName === '_TZE200_3p5ydos3') {
                dp = tuya.dataPoints.eardaDimmerLevel;
            }
            if (key === 'brightness_min') {
                if (value >= 0 && value <= 100) {
                    newValue = utils.mapNumberRange(value, 0, 100, 0, 1000);
                    dp = tuya.dataPoints.dimmerLevel;
                } else {
                    throw new Error('Dimmer brightness_min is out of range 0..100');
                }
            } else if (key === 'level') {
                if (value >= 0 && value <= 1000) {
                    newValue = Math.round(Number(value));
                } else {
                    throw new Error('Dimmer level is out of range 0..1000');
                }
            } else if (key === 'brightness_percent') {
                if (value >= 0 && value <= 100) {
                    newValue = utils.mapNumberRange(value, 0, 100, 0, 1000);
                } else {
                    throw new Error('Dimmer brightness_percent is out of range 0..100');
                }
            } else {
                if (value >= 0 && value <= 255) {
                    newValue = utils.mapNumberRange(value, 0, 255, 0, 1000);
                } else {
                    throw new Error('Dimmer brightness is out of range 0..255');
                }
            }
            // Always use same transid as tuya_dimmer_state (https://github.com/Koenkk/zigbee2mqtt/issues/6366)
            await tuya.sendDataPointValue(entity, dp, newValue, 'dataRequest', 1);
        },
    },
};

const definition = {
    fingerprint: [
        {modelID: 'TS0601', manufacturerName: '_TZE200_9cxuhakf'},
    ],
    model: 'TS0601_dimmer',
    vendor: 'TuYa',
    description: 'Zigbee smart dimmer',
    fromZigbee: [fz.tuya_dimmer, fz.ignore_basic_report],
    toZigbee: [tz.tuya_dimmer_state, tzLocal.tuya_dimmer_level],
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(1);
        await reporting.bind(endpoint, coordinatorEndpoint, ['genOnOff', 'genLevelCtrl']);
    },
    exposes: [e.light_brightness().setAccess('state', ea.STATE_SET).setAccess('brightness', ea.STATE_SET)],
    whiteLabel: [
        {vendor: 'Larkkey', model: 'ZSTY-SM-1DMZG-EU'},
        {vendor: 'Earda', model: 'EDM-1ZAA-EU'},
        {vendor: 'Earda', model: 'EDM-1ZAB-EU'},
        {vendor: 'Earda', model: 'EDM-1ZBA-EU'},
        {vendor: 'Mercator Ikuü', model: 'SSWM-DIMZ'},
    ],
};

module.exports = definition;`

**configuration.yaml file**
`external_converters:
 - TS0601_new.js`