Koenkk/zigbee2mqtt #17039: zigbee2mqtt fails to start as docker container with "Failed to connect to the adapter"

elronzo, 2023-03-14:
### What happened?

I do always fear that I have to restart the zigbee2mqtt docker container as it refuses to start in many attempts.

My adapter is an Olizig (ZigStar POE) which I configured in config.yaml by using its FQDN and port. 

### What did you expect to happen?

that zigbee2mqtt starts properly

### How to reproduce it (minimal and precise)

Try to start zigbee2mqtt with an adapter connected via LAN (an Olizig POE Adapter in my case)

### Zigbee2MQTT version

1.30.2

### Adapter firmware version

CC1352P2_CC2652P_launchpad_coordinator_20221226

### Adapter

ZigStar PoE Coordinator (OliZig)

### Debug log

```
Zigbee2MQTT:error 2023-03-14 12:59:12: Error: Failed to connect to the adapter (Error: SRSP - SYS - ping after 6000ms)
    at ZStackAdapter.start (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:103:27)
    at Controller.start (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:132:29)
    at Zigbee.start (/app/lib/zigbee.ts:58:27)
    at Controller.start (/app/lib/controller.ts:101:27)
    at start (/app/index.js:107:5)
```

Of course I can connect to the adapter - with telnet for example:
```
telnet zigstargw.lan 6638
Trying 192.168.73.190...
Connected to zigstargw.lan.
Escape character is '^]'.
```

I can see zigbee2mqtt connecting in the ZigStar WebInterface but it immediately disconnects...

frfstu, 2023-03-14:
> ### What happened?
> 
> I do always fear that I have to restart the zigbee2mqtt docker container as it refuses to start in many attempts.
> 
> My adapter is an Olizig (ZigStar POE) which I configured in config.yaml by using its FQDN and port.
> ### What did you expect to happen?
> 
> that zigbee2mqtt starts properly
> ### How to reproduce it (minimal and precise)
> 
> Try to start zigbee2mqtt with an adapter connected via LAN (an Olizig POE Adapter in my case)
> ### Zigbee2MQTT version
> 
> 1.30.2
> ### Adapter firmware version
> 
> CC1352P2_CC2652P_launchpad_coordinator_20221226
> ### Adapter
> 
> ZigStar PoE Coordinator (OliZig)
> ### Debug log
> 
> ```
> Zigbee2MQTT:error 2023-03-14 12:59:12: Error: Failed to connect to the adapter (Error: SRSP - SYS - ping after 6000ms)
>     at ZStackAdapter.start (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:103:27)
>     at Controller.start (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:132:29)
>     at Zigbee.start (/app/lib/zigbee.ts:58:27)
>     at Controller.start (/app/lib/controller.ts:101:27)
>     at start (/app/index.js:107:5)
> ```
> 
> Of course I can connect to the adapter - with telnet for example:
> 
> ```
> telnet zigstargw.lan 6638
> Trying 192.168.73.190...
> Connected to zigstargw.lan.
> Escape character is '^]'.
> ```
> 
> I can see zigbee2mqtt connecting in the ZigStar WebInterface but it disconnects after a few seconds...
> 
> NOTE: I managed to start zigbee2mqtt. I unplugged my ZigStar POE device and waited a minute. Then it worked again. Does this indicate a faulty device?

Try the configuration like mine. I did not like the work of the December firmware (I had several devices that previously worked fine with it, stopped working correctly and I returned to the February one). I also have doubts about z2m 1.30.2, but I can’t return to 1.29.2 (I can only do this on home assistant (menu item reinstall from ...). One thing I can say for sure is that in such a configuration (bundle ) the manual on/off mode works for me and there are no errors in the log.It is quite likely (since the author of the firmware of our coordinators is the same) returning to the firmware of the coordinator version 20220219 will partially correct the situation!
[image]

frfstu, 2023-03-14:
In general, my system sees my coordinator (it is determined and you can select it), and at the same time, when setting up zigbee, it tells me that it cannot determine what kind of coordinator it is and connect it! Although it is working with zigbee2MQTT at this moment, it manages the transmitted data between devices properly (without obvious errors), and in my browser on the zigbee2MQTT page I can manually turn off and on functions in the devices of my network (nothing more, automation does not work at all) !

elronzo, 2023-04-11:
@Koenkk How could a novice like me help to further debug this particular problem?

closed 2023-05-19 by github-actions[bot]