Koenkk/zigbee2mqtt #17043: [Feature request]: Develco KEYZB-110 expose RFID reader

luisetex84, 2023-03-14:
### Is your feature request related to a problem? Please describe

Develco KEYZB-110 don't show any information when a RFID tag is near its RFID reader

### Describe the solution you'd like

I would like to operate the keypad with RFID tags

### Describe alternatives you've considered

I've considered trying with ZHA if this function is supported but I don't have a ZHA dongle

### Additional context

Other functionality like presence sensor is not supported

luisetex84, 2023-03-16:
I was using wrong RFID tags. If you use the right ones the info appears in action_code

closed 2023-03-16 by luisetex84

davidsmoker, 2025-05-26:
Hello, which RFID tags did you use?
I'm still trying to get this working :-(