Koenkk/zigbee2mqtt #17044: [New device support]:

erickmhonorio, 2023-03-14:
### Link

https://pt.aliexpress.com/item/1005002932398088.html?spm=a2g0o.detail.1000060.1.5bfc1cd2Cwxe9B&gps-id=pcDetailBottomMoreThisSeller&scm=1007.13339.291025.0&scm_id=1007.13339.291025.0&scm-url=1007.13339.291025.0&pvid=00246ade-8a1d-4c07-9404-ffdc82ece9d7&_t=gps-id%3ApcDetailBottomMoreThisSeller%2Cscm-url%3A1007.13339.291025.0%2Cpvid%3A00246ade-8a1d-4c07-9404-ffdc82ece9d7%2Ctpp_buckets%3A668%232846%238112%231997&isseo=y&pdp_ext_f=%7B%22sku_id%22%3A%2212000022859290965%22%2C%22sceneId%22%3A%223339%22%7D&pdp_npi=3%40dis%21BRL%21129.92%2164.96%21%21%21%21%21%402101d1b616788264940876793e4354%2112000022859290965%21rec%21BR%21722355980&gatewayAdapt=glo2bra

### Database entry

,

### Comments

Could you help me to add this unsupported device

junosuarez, 2023-03-15:
Please edit the issue title to include the name of the new device

erickmhonorio, 2023-03-17:
Done

closed 2023-04-24 by github-actions[bot]