Koenkk/zigbee2mqtt #17048: Yale doorlock: NULL attribute due to double Z2M published payloads

farhanito, 2023-03-15:
### What happened?

Yale YRD210-HA-605 paired and interviewed successfully. But some attributes always have NULL value, i believe this was caused by Z2M published 2 payloads (in quick succession) with the same attributes following every messages from the device. 

First payloads contain correct values, the second contains NULL or empty string.
"lockState', 'battery', and 'linkquality' reported correctly.
'action', 'source_name', and 'user' always has NULL values.

### What did you expect to happen?

'action', 'source_name', and 'user' reported correctly.

### How to reproduce it (minimal and precise)

Problem happened with every action on the device.

### Zigbee2MQTT version

1.30.2 and 1.30.2-dev commit: 4272d9a

### Adapter firmware version

20221226

### Adapter

USB CC2652P zStack3x0

### Debug log

`Debug 2023-03-15 07:18:29Received Zigbee message from '0x000d6f00024c541e', type 'commandOperationEventNotification', cluster 'closuresDoorLock', data '{"data":0,"opereventcode":14,"opereventsrc":2,"pin":{"data":[],"type":"Buffer"},"userid":65535,"zigbeelocaltime":732179911}' from endpoint 1 with groupID 0
Info 2023-03-15 07:18:29MQTT publish: topic 'zigbee2mqtt/0x000d6f00024c541e', payload '{"action":"manual_unlock","action_source":2,"action_source_name":"manual","action_user":65535,"auto_relock_time":0,"battery":80,"battery_low":false,"linkquality":123,"lock_state":"locked","pin_code":null,"sound_volume":null,"state":"LOCK"}'
Info 2023-03-15 07:18:29MQTT publish: topic 'zigbee2mqtt/0x000d6f00024c541e', payload '{"action":"","action_source_name":null,"action_user":null,"auto_relock_time":0,"battery":80,"battery_low":false,"linkquality":123,"lock_state":"locked","pin_code":null,"sound_volume":null,"state":"LOCK"}'
Info 2023-03-15 07:18:29MQTT publish: topic 'zigbee2mqtt/0x000d6f00024c541e/action', payload 'manual_unlock'
Debug 2023-03-15 07:18:31Received Zigbee message from '0x000d6f00024c541e', type 'attributeReport', cluster 'closuresDoorLock', data '{"lockState":2}' from endpoint 1 with groupID 0
Info 2023-03-15 07:18:31MQTT publish: topic 'zigbee2mqtt/0x000d6f00024c541e', payload '{"action":null,"action_source_name":null,"action_user":null,"auto_relock_time":0,"battery":80,"battery_low":false,"linkquality":116,"lock_state":"unlocked","pin_code":null,"sound_volume":null,"state":"UNLOCK"}'
Debug 2023-03-15 07:19:05Received Zigbee message from '0x000d6f00024c541e', type 'commandOperationEventNotification', cluster 'closuresDoorLock', data '{"data":0,"opereventcode":13,"opereventsrc":2,"pin":{"data":[],"type":"Buffer"},"userid":65535,"zigbeelocaltime":732179947}' from endpoint 1 with groupID 0
Info 2023-03-15 07:19:05MQTT publish: topic 'zigbee2mqtt/0x000d6f00024c541e', payload '{"action":"manual_lock","action_source":2,"action_source_name":"manual","action_user":65535,"auto_relock_time":0,"battery":80,"battery_low":false,"linkquality":123,"lock_state":"unlocked","pin_code":null,"sound_volume":null,"state":"UNLOCK"}'
Info 2023-03-15 07:19:05MQTT publish: topic 'zigbee2mqtt/0x000d6f00024c541e', payload '{"action":"","action_source_name":null,"action_user":null,"auto_relock_time":0,"battery":80,"battery_low":false,"linkquality":123,"lock_state":"unlocked","pin_code":null,"sound_volume":null,"state":"UNLOCK"}'
Info 2023-03-15 07:19:05MQTT publish: topic 'zigbee2mqtt/0x000d6f00024c541e/action', payload 'manual_lock'
Debug 2023-03-15 07:19:07Received Zigbee message from '0x000d6f00024c541e', type 'attributeReport', cluster 'closuresDoorLock', data '{"lockState":1}' from endpoint 1 with groupID 0
Info 2023-03-15 07:19:07MQTT publish: topic 'zigbee2mqtt/0x000d6f00024c541e', payload '{"action":null,"action_source_name":null,"action_user":null,"auto_relock_time":0,"battery":80,"battery_low":false,"linkquality":142,"lock_state":"locked","pin_code":null,"sound_volume":null,"state":"LOCK"}'
Debug 2023-03-15 07:19:48Received Zigbee message from '0x000d6f00024c541e', type 'commandOperationEventNotification', cluster 'closuresDoorLock', data '{"data":0,"opereventcode":2,"opereventsrc":0,"pin":{"data":[],"type":"Buffer"},"userid":3,"zigbeelocaltime":732179990}' from endpoint 1 with groupID 0
Info 2023-03-15 07:19:48MQTT publish: topic 'zigbee2mqtt/0x000d6f00024c541e', payload '{"action":"unlock","action_source":0,"action_source_name":"keypad","action_user":3,"auto_relock_time":0,"battery":80,"battery_low":false,"linkquality":145,"lock_state":"locked","pin_code":null,"sound_volume":null,"state":"LOCK"}'
Info 2023-03-15 07:19:48MQTT publish: topic 'zigbee2mqtt/0x000d6f00024c541e', payload '{"action":"","action_source_name":null,"action_user":null,"auto_relock_time":0,"battery":80,"battery_low":false,"linkquality":145,"lock_state":"locked","pin_code":null,"sound_volume":null,"state":"LOCK"}'
Info 2023-03-15 07:19:48MQTT publish: topic 'zigbee2mqtt/0x000d6f00024c541e/action', payload 'unlock'
Debug 2023-03-15 07:19:50Received Zigbee message from '0x000d6f00024c541e', type 'attributeReport', cluster 'closuresDoorLock', data '{"lockState":2}' from endpoint 1 with groupID 0
Info 2023-03-15 07:19:50MQTT publish: topic 'zigbee2mqtt/0x000d6f00024c541e', payload '{"action":null,"action_source_name":null,"action_user":null,"auto_relock_time":0,"battery":80,"battery_low":false,"linkquality":138,"lock_state":"unlocked","pin_code":null,"sound_volume":null,"state":"UNLOCK"}'
`

closed 2023-04-23 by github-actions[bot]

Montreal666, 2024-01-13:
Same issue here. Did you figure out a solution? Thanks

Montreal666, 2024-01-14:
Future reference for HA users:

https://community.home-assistant.io/t/mqtt-sensor-filter-states-to-ignore-null-none-unknown/671945

DarkPark, 2024-10-30:
the same problem - duplicate events with empty action

```
info 2024-10-30 22:31:08z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0x00158d0003456b48', payload '{"action":"single_right","battery":100,"device_temperature":27,"last_seen":"2024-10-30T22:31:08+02:00","linkquality":48,"power_outage_count":213,"voltage":3015}'
info 2024-10-30 22:31:08z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0x00158d0003456b48', payload '{"action":"","battery":100,"device_temperature":27,"last_seen":"2024-10-30T22:31:08+02:00","linkquality":48,"power_outage_count":213,"voltage":3015}'
info 2024-10-30 22:31:08z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0x00158d0003456b48/action', payload 'single_right'
```