Koenkk/zigbee2mqtt #17057: Window sensor failed to interview

gokuu, 2023-03-16:
### What happened?

I have a window sensor that joins the network immediately when entering pairing mode. It starts the interview process, but after about 1 or 2 minutes I get an error message saying that the interview failed. The device then remains in the list, but marked as unsupported.

I created a new Zigbee2mqtt instance just to pair this device to get a clean log, and after failing the interview, going through the log, I see a lot of [seemingly] repeated blocks:
```
...
2023-03-15T18:57:13.781Z zigbee-herdsman:controller:log Device '0xa4c138e3a4e5da78' joined
2023-03-15T18:57:13.781Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2023-03-15T18:57:13.781Z zigbee-herdsman:controller:log Device '0xa4c138e3a4e5da78' accepted by handler
2023-03-15T18:57:13.781Z zigbee-herdsman:controller:log Device '0xa4c138e3a4e5da78' is already in database with different networkAddress, updating networkAddress
2023-03-15T18:57:13.781Z zigbee-herdsman:controller:database:log Writing database to '/app/data/database.db'
2023-03-15T18:57:13.875Z zigbee-herdsman:controller:log Not interviewing '0xa4c138e3a4e5da78', completed 'false', in progress 'true'
2023-03-15T18:57:18.022Z zigbee-herdsman:adapter:zStack:adapter Node descriptor request for '23166' failed (Error: AREQ - ZDO - nodeDescRsp after 10000ms), retry
2023-03-15T18:57:18.022Z zigbee-herdsman:adapter:zStack:adapter Discovering route to 23166
2023-03-15T18:57:18.023Z zigbee-herdsman:adapter:zStack:znp:SREQ --> ZDO - extRouteDisc - {"dstAddr":23166,"options":0,"radius":30}
2023-03-15T18:57:18.023Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,4,37,69,126,90,0,30,94]
2023-03-15T18:57:18.026Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,101,69,0,33]
2023-03-15T18:57:18.026Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,101,69,0,33]
2023-03-15T18:57:18.026Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 5 - 69 - [0] - 33
2023-03-15T18:57:18.026Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- ZDO - extRouteDisc - {"status":0}
2023-03-15T18:57:18.026Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2023-03-15T18:57:19.744Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,35,68,129,0,0,0,0,113,168,1,1,0,126,0,134,148,44,0,0,15,8,164,10,1,0,32,77,226,255,32,54,228,255,32,0,113,168,29,78]
2023-03-15T18:57:19.744Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,35,68,129,0,0,0,0,113,168,1,1,0,126,0,134,148,44,0,0,15,8,164,10,1,0,32,77,226,255,32,54,228,255,32,0,113,168,29,78]
2023-03-15T18:57:19.744Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 35 - 2 - 4 - 129 - [0,0,0,0,113,168,1,1,0,126,0,134,148,44,0,0,15,8,164,10,1,0,32,77,226,255,32,54,228,255,32,0,113,168,29] - 78
2023-03-15T18:57:19.744Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - incomingMsg - {"groupid":0,"clusterid":0,"srcaddr":43121,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":126,"securityuse":0,"timestamp":2921606,"transseqnumber":0,"len":15,"data":{"type":"Buffer","data":[8,164,10,1,0,32,77,226,255,32,54,228,255,32,0]}}
2023-03-15T18:57:19.744Z zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":0,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":false,"reservedBits":0},"transactionSequenceNumber":164,"manufacturerCode":null,"commandIdentifier":10},"Payload":[{"attrId":1,"dataType":32,"attrData":77},{"attrId":65506,"dataType":32,"attrData":54},{"attrId":65508,"dataType":32,"attrData":0}],"Command":{"ID":10,"name":"report","parameters":[{"name":"attrId","type":33},{"name":"dataType","type":32},{"name":"attrData","type":1000}]}},"address":43121,"endpoint":1,"linkquality":126,"groupID":0,"wasBroadcast":false,"destinationEndpoint":1}'
2023-03-15T18:57:19.744Z zigbee-herdsman:controller:log 'zcl' data is from unknown device with address '43121', skipping...
2023-03-15T18:57:19.744Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2023-03-15T18:57:19.950Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,35,68,129,0,0,0,0,113,168,1,1,0,126,0,123,197,44,0,0,15,8,164,10,1,0,32,77,226,255,32,54,228,255,32,0,113,168,29,226]
2023-03-15T18:57:19.950Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,35,68,129,0,0,0,0,113,168,1,1,0,126,0,123,197,44,0,0,15,8,164,10,1,0,32,77,226,255,32,54,228,255,32,0,113,168,29,226]
2023-03-15T18:57:19.950Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 35 - 2 - 4 - 129 - [0,0,0,0,113,168,1,1,0,126,0,123,197,44,0,0,15,8,164,10,1,0,32,77,226,255,32,54,228,255,32,0,113,168,29] - 226
2023-03-15T18:57:19.950Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - incomingMsg - {"groupid":0,"clusterid":0,"srcaddr":43121,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":126,"securityuse":0,"timestamp":2934139,"transseqnumber":0,"len":15,"data":{"type":"Buffer","data":[8,164,10,1,0,32,77,226,255,32,54,228,255,32,0]}}
2023-03-15T18:57:19.951Z zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":0,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":false,"reservedBits":0},"transactionSequenceNumber":164,"manufacturerCode":null,"commandIdentifier":10},"Payload":[{"attrId":1,"dataType":32,"attrData":77},{"attrId":65506,"dataType":32,"attrData":54},{"attrId":65508,"dataType":32,"attrData":0}],"Command":{"ID":10,"name":"report","parameters":[{"name":"attrId","type":33},{"name":"dataType","type":32},{"name":"attrData","type":1000}]}},"address":43121,"endpoint":1,"linkquality":126,"groupID":0,"wasBroadcast":false,"destinationEndpoint":1}'
2023-03-15T18:57:19.951Z zigbee-herdsman:controller:log 'zcl' data is from unknown device with address '43121', skipping...
2023-03-15T18:57:19.951Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2023-03-15T18:57:20.145Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,35,68,129,0,0,0,0,113,168,1,1,0,123,0,109,246,44,0,0,15,8,164,10,1,0,32,77,226,255,32,54,228,255,32,0,113,168,29,194]
2023-03-15T18:57:20.145Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,35,68,129,0,0,0,0,113,168,1,1,0,123,0,109,246,44,0,0,15,8,164,10,1,0,32,77,226,255,32,54,228,255,32,0,113,168,29,194]
2023-03-15T18:57:20.145Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 35 - 2 - 4 - 129 - [0,0,0,0,113,168,1,1,0,123,0,109,246,44,0,0,15,8,164,10,1,0,32,77,226,255,32,54,228,255,32,0,113,168,29] - 194
2023-03-15T18:57:20.145Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - incomingMsg - {"groupid":0,"clusterid":0,"srcaddr":43121,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":123,"securityuse":0,"timestamp":2946669,"transseqnumber":0,"len":15,"data":{"type":"Buffer","data":[8,164,10,1,0,32,77,226,255,32,54,228,255,32,0]}}
2023-03-15T18:57:20.145Z zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":0,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":false,"reservedBits":0},"transactionSequenceNumber":164,"manufacturerCode":null,"commandIdentifier":10},"Payload":[{"attrId":1,"dataType":32,"attrData":77},{"attrId":65506,"dataType":32,"attrData":54},{"attrId":65508,"dataType":32,"attrData":0}],"Command":{"ID":10,"name":"report","parameters":[{"name":"attrId","type":33},{"name":"dataType","type":32},{"name":"attrData","type":1000}]}},"address":43121,"endpoint":1,"linkquality":123,"groupID":0,"wasBroadcast":false,"destinationEndpoint":1}'
2023-03-15T18:57:20.145Z zigbee-herdsman:controller:log 'zcl' data is from unknown device with address '43121', skipping...
2023-03-15T18:57:20.145Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2023-03-15T18:57:20.239Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,12,69,202,78,243,120,218,229,164,227,56,193,164,0,0,99]
2023-03-15T18:57:20.239Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,12,69,202,78,243,120,218,229,164,227,56,193,164,0,0,99]
2023-03-15T18:57:20.239Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 12 - 2 - 5 - 202 - [78,243,120,218,229,164,227,56,193,164,0,0] - 99
2023-03-15T18:57:20.239Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - tcDeviceInd - {"nwkaddr":62286,"extaddr":"0xa4c138e3a4e5da78","parentaddr":0}
...
```

The full log is attached at the end.

Any help, or idea on how to resolve this?

### What did you expect to happen?

The device should be interview successfully and identified as [TuYa TM001-ZA/TM081 ](https://www.zigbee2mqtt.io/devices/TM001-ZA_TM081.html#tuya-tm001-za%252Ftm081).

### Zigbee2MQTT version

1.30.2 commit: cdf62ea

### Adapter firmware version

20210708

### Adapter

Slaesh's CC2652RB Stick

### Debug log

[zigbee.log](https://github.com/Koenkk/zigbee2mqtt/files/10994112/zigbee.log)

gokuu, 2023-03-24:
Pinging this to see if anyone has any idea? 😕

TheLordVader, 2023-04-30:
Would like to know if there is a solution to this..?

gokuu, 2023-04-30:
My solution was to buy a similar one from a different brand :-(

closed 2023-06-11 by github-actions[bot]