Koenkk/zigbee2mqtt #17060: TS011F_plug_3 _TZ3000_cphmq0q7 power don't work

jocamane, 2023-03-16:
### What happened?

After update core to 2023.3.3 and Zigbee2mqtt 1.30.2 the power stopped working. when I plug in the power is at 0. 
if I restart the z2m it will work for a while. after some time it stops working again.

In 2023.3.4 core don't work also.

In Core 2023.2.5 and Zigbee2MQTT 1.30.1-1 work ok.

Tomada 5 ZG

### What did you expect to happen?

The power work ok

### Zigbee2MQTT version

1.30.2 commit: unknown

### Adapter firmware version

0x26780700

### Adapter

conbee ii

### Debug log

[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/10996525/log.txt)

jocamane, 2023-03-16:
[image]
[image]
[image]

jocamane, 2023-03-16:
With 2023.3.5 core don't work also.

Important information: when I restart the HA it doesn't work. It only works if after that I restart the Z2M, and only for a while (I don't know how long).
If instead of restarting HA I do reboot system, it also works but for a short time too

Baconwolf, 2023-03-17:
Same problem for me for the TS0121 (_TZ3000_g5xawfcq). After some time the "power" status stops updating.
What I can do is read vluster _haElectricalMeasurement_ and attribute _activePower_ in the dev console manually. This returns an updated value and also updates the status.
So maybe it's just the polling not working correctly?

pwlgrzs, 2023-03-17:
Can confirm what @Baconwolf stated, 

I have 2x TS0121 _TZ3000_okaz9tjs with FW version 500+TZSKT31BS101 and 1x with FW 20B+TZSKT11BS108.

20B+TZSKT11BS108 works OK
500+TZSKT31BS101 not returning polling results.

jocamane, 2023-04-15:
https://github.com/Koenkk/zigbee2mqtt/issues/17352#issue-1669528358

join to club

closed 2023-05-23 by github-actions[bot]