Koenkk/zigbee2mqtt #17066: ROB_200-060-0 - Dimmable led light strip

zboersen, 2023-03-17:
### Link

https://www.robbshop.nl/robb-smarrt-zigbee-ledstrip-kleur-2-meter

### Database entry

{"id":15,"type":"Router","ieeeAddr":"0x04cd15fffeb4c9c5","nwkAddr":56446,"manufId":4644,"manufName":"ROBB smarrt","powerSource":"Mains (single phase)","modelId":"ROB_200-060-0","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":269,"inClusterList":[0,3,4,5,6,8,768,2821,4096],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"modelId":"ROB_200-060-0","manufacturerName":"ROBB smarrt","powerSource":1,"zclVersion":3,"appVersion":0,"stackVersion":0,"hwVersion":1,"dateCode":"NULL","swBuildId":"2.9.2_r63"}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":102,"inClusterList":[33],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":0,"stackVersion":0,"hwVersion":1,"dateCode":"NULL","swBuildId":"2.9.2_r63","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1679072239219,"defaultSendRequestWhen":"immediate"}

### Comments

Tried to follow the guid. However, I am new at this so got stuck.
I added 
"
advanced:
  log_level: debug
external_converters:
  - ROB2000600.js
"
To my zigbeemqtt addon configuration.
Also added the following to the ROB2000600.js:
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['lumi.sens'], // The model ID from: Device with modelID 'lumi.sens' is not supported.
    model: 'WSDCGQ01LM', // Vendor model number, look on the device for a model number
    vendor: 'Xiaomi', // Vendor of the device (only used for documentation and startup logging)
    description: 'MiJia temperature & humidity sensor', // Description of the device, copy from vendor site. (only used for documentation and startup logging)
    fromZigbee: [], // We will add this later
    toZigbee: [], // Should be empty, unless device can be controlled (e.g. lights, switches).
    exposes: [e.battery(), e.temperature(), e.humidity()], // Defines what this device exposes, used for e.g. Home Assistant discovery and in the frontend
};

module.exports = definition;

Got stuck at the step: Zigbee2MQTT: Once finished, restart Zigbee2MQTT and trigger some actions on the device. You will see messages like: debug  2019-11-09T12:24:22: No converter available for 'WSDCGQ01LM' with cluster 'msTemperatureMeasurement' and type 'attributeReport' and data '{"measuredValue":2512}'

Any help?

Koenkk, 2023-03-18:
Please add the color temperature range as described on https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html#_3-adding-converter-s-for-your-device

Could you check if the issue is fixed with the following external converter: 

```js
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const utils = require('zigbee-herdsman-converters/lib/utils');
const globalStore = require('zigbee-herdsman-converters/lib/store');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['ROB_200-060-0'],
    model: 'ROB_200-060-0',
    vendor: 'ROBB',
    description: 'Zigbee LED driver',
    extend: extend.light_onoff_brightness_colortemp_color(),
};

module.exports = definition;
```

- save this as file next to `configuration.yaml` as `ext_converter.js`
- add it to `configuration.yaml`:
```yaml
external_converters:
  - ext_converter.js
```
- start z2m, check if issue is fixed

zboersen, 2023-03-18:
Thank you for your quick response!! I did as you said, but I did run into an error when trying to restart. I added some screenshots so you can see if i messed up somewhere (probably)

1. I have added the above to the ext_converter.js which is "next to the configuration.yaml":
[image]

2. Added the following to my configuration.yaml :
external_converters:
  - ext_converter.js
[image]

3. Run into error when trying to restart. 
[image]

Do you know what i am doing wrong here?
And do i need to change anything to the zigbeemqtt add-on? 

Thanks so much for helping!

Greetings Zoë

Koenkk, 2023-03-18:
Remove the 2 spaces before `external_converters:` and `- ext_converter.js`

zboersen, 2023-03-19:
Thanks. I got it right in the zigbeemqtt add-on.

However, still get the following error: Device '0x04cd15fffeb4c9c5' cannot be configured

Any ideas how to fix this?

zboersen, 2023-03-19:
Yes! Got it working! Thanks you for the help and very quick response :)

Just got to do step 4 and add native support. However, what is ment by : "Clone [zigbee2mqtt.io](https://github.com/Koenkk/zigbee2mqtt.io)"?

Koenkk, 2023-03-20:
I've added the device, you can keep using the external converter until the next release.

closed 2023-03-20 by Koenkk