Koenkk/zigbee2mqtt #17069: Legrand 067773, 002a firmware only Toggle action

mkapuscinski, 2023-03-18:
### What happened?

I have two Legrand witches with firmware 002a

after pressing a button [no difference if ON or OFF button is being pressed], I am getting only TOGGLE action
After pressing it twice, a am getting only OFF action.

This issue is visible only with Legrand 067773 with firmware 002a 

all other Legrand 067773 with older firmware [001e] works OK

### What did you expect to happen?

pressing ON -> ON action
pressing OFF -> OFF action

### How to reproduce it (minimal and precise)

Press the buttons on 067773 with 002a firmware

### Zigbee2MQTT version

1.30.2

### Adapter firmware version

26720700

### Adapter

deconz / conbee II

### Debug log

Info 2023-03-18 14:53:23MQTT publish: topic 'zigbee2mqtt/valena_balkon_1', payload '{"action":null,"battery":100,"last_seen":"2023-03-18T13:53:23.866Z","linkquality":255,"voltage":3000}'
Info 2023-03-18 14:53:24MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"valena_balkon_1","ieee_address":"0x0004740000c44e73"},"type":"device_announce"}'
Info 2023-03-18 14:53:24MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"valena_balkon_1"},"type":"device_announced"}'
Info 2023-03-18 14:53:24MQTT publish: topic 'zigbee2mqtt/valena_balkon_1', payload '{"action":null,"battery":100,"last_seen":"2023-03-18T13:53:24.058Z","linkquality":255,"voltage":3000}'
Info 2023-03-18 14:53:24MQTT publish: topic 'zigbee2mqtt/valena_balkon_1', payload '{"action":"toggle","action_group":6,"battery":100,"last_seen":"2023-03-18T13:53:24.082Z","linkquality":255,"voltage":3000}'
Info 2023-03-18 14:53:24MQTT publish: topic 'zigbee2mqtt/valena_balkon_1', payload '{"action":"","battery":100,"last_seen":"2023-03-18T13:53:24.082Z","linkquality":255,"voltage":3000}'
Info 2023-03-18 14:53:24MQTT publish: topic 'zigbee2mqtt/valena_balkon_1/action', payload 'toggle'
Info 2023-03-18 14:53:24MQTT publish: topic 'zigbee2mqtt/valena_balkon_1', payload '{"action":null,"battery":100,"last_seen":"2023-03-18T13:53:24.910Z","linkquality":255,"voltage":3000}'
Info 2023-03-18 14:53:24MQTT publish: topic 'zigbee2mqtt/valena_balkon_1', payload '{"action":null,"battery":100,"last_seen":"2023-03-18T13:53:24.929Z","linkquality":255,"voltage":3000}'
Info 2023-03-18 14:53:33MQTT publish: topic 'zigbee2mqtt/valena_balkon_1', payload '{"action":"toggle","action_group":6,"battery":100,"last_seen":"2023-03-18T13:53:34.022Z","linkquality":255,"voltage":3000}'
Info 2023-03-18 14:53:33MQTT publish: topic 'zigbee2mqtt/valena_balkon_1', payload '{"action":"","battery":100,"last_seen":"2023-03-18T13:53:34.022Z","linkquality":255,"voltage":3000}'
Info 2023-03-18 14:53:33MQTT publish: topic 'zigbee2mqtt/valena_balkon_1/action', payload 'toggle'

jiquem, 2023-03-18:
Upgrade your firmware. We are at firmware 63 now as indicated here https://community.jeedom.com/t/legrand-with-netatmo-firmware-des-equipements/75218

mkapuscinski, 2023-03-19:
Hey @jiquem thanks!

I've managed to force the update, a now working on 63 [in the Legrand App] previously it was 40
nevertheless, when I'm connecting it back to Zigbee2mqtt [via home assistant], the problem is still the same.

toggle instead of ON, OFF action. And even when switch is paired to Legrand gateway, behaves the same [toggle instead of ON, OFF]

P.S. zigbee2mqtt still shows firmware at 002a

jiquem, 2023-03-19:
You're right, I don't know exactly when but the "on / off" feature was changed to "toggle" in one of the firmware revision.
Regarding zigbee2mqtt, the version is never updated

Loic691, 2023-04-03:
Yes Legrand remove the on/off to toggle

closed 2023-05-11 by github-actions[bot]

paulhenric, 2025-01-18:
I have exactly the same problem here.
@mkapuscinski : did you manage to solve your problem?

mkapuscinski, 2025-01-18:
@paulhenric, unfortunately not. 
I did even pair the switch with official Legrand Hub, and had the same result [both buttons had toggle action]

Loic691, 2026-04-02:
For my side Z2M last version, these 067773 devices have been updated few month ago on 
```
ID du firmware
004d

20171123

Version du firmware
Inconnu
```
If you have information to switch the device in ON/OFF mode rather than TOGGLE I take your information because TOGGLE mode is not for me an acceptable behaviour when managing HA automation

oliviertassinari, 2026-09-18:
A solution that seems to work reliably: https://github.com/Koenkk/zigbee2mqtt/issues/32608#issuecomment-5737458158 with the latest firmware.