Koenkk/zigbee2mqtt #17070: Some Aqara's WSDCGQ11LM drops from the network (two in the same time)

hnykda, 2023-03-18:
### What happened?

I have about 5 [Xiaomi WSDCGQ11LM](https://www.zigbee2mqtt.io/devices/WSDCGQ11LM.html). Some of them sometimes drop from the network, but one specifically do it almost every week and it's driving me crazy. This time, however, two of these dropped from the network in almost exactly same time, so I suspect there is something _else_ causing this. Here
[image]

they dropped out of the network after ~1:25 AM. The logs don't show anything, they just stopped reporting 🤷 . Look for "THSensorBeta" and "THSensorGamma": https://gist.github.com/hnykda/54eb8b72fb24c710447ac7edaffb0952

It's probably related to https://github.com/Koenkk/zigbee2mqtt/issues/2363 , but not sure.

I usually have to repair the device, either by (force) removing them and pairing them again. 

I can provide sniffed logs.

### What did you expect to happen?

Devices stay on the network.

### Zigbee2MQTT version

1.30.2

### Adapter firmware version

20220219

### Adapter

Slaesh

### Debug log

https://gist.github.com/hnykda/54eb8b72fb24c710447ac7edaffb0952

satter, 2023-03-19:
I made an simultaneous upgrade of z2m (1.30.1->1.30.2) and Sonoff USB stick firmware (20211217 -> 20221226)  
Now both of my AB3257001NJ Osram plugs drops off the network at the same time

Nothing unusual appears in the logs before disconnect, but I forgot to enable herdsman debug last time  
Now waiting for the issue to reproduce itself

satter, 2023-03-30:
My plugs did drop off the network again
```
Mar 30 12:57:31 homie1 npm[1362636]: 2023-03-30T12:57:31.184Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,13,69,201,104,146,18,104,10,10,170,62,176,124,0,0,1,88,254,13,69,201,125,14,241,95,174,0,170,62,176,124,0,0,1,171]
Mar 30 12:57:31 homie1 npm[1362636]: 2023-03-30T12:57:31.184Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,13,69,201,104,146,18,104,10,10,170,62,176,124,0,0,1,88,254,13,69,201,125,14,241,95,174,0,170,62,176,124,0,0,1,171]
Mar 30 12:57:31 homie1 npm[1362636]: 2023-03-30T12:57:31.184Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 13 - 2 - 5 - 201 - [104,146,18,104,10,10,170,62,176,124,0,0,1] - 88
Mar 30 12:57:31 homie1 npm[1362636]: 2023-03-30T12:57:31.184Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - leaveInd - {"srcaddr":37480,"extaddr":"0x7cb03eaa0a0a6812","request":0,"removechildren":0,"rejoin":1}
Mar 30 12:57:31 homie1 npm[1362636]: 2023-03-30T12:57:31.185Z zigbee-herdsman:controller:log Device leave '0x7cb03eaa0a0a6812'
Mar 30 12:57:31 homie1 npm[1362636]: 2023-03-30T12:57:31.185Z zigbee-herdsman:controller:log Removing device from database '0x7cb03eaa0a0a6812'
Mar 30 12:57:31 homie1 npm[1362636]: 2023-03-30T12:57:31.185Z zigbee-herdsman:controller:database:log Writing database to '/opt/zigbee2mqtt/data/database.db'
Mar 30 12:57:31 homie1 npm[1362636]: Zigbee2MQTT:warn  2023-03-30 12:57:31: Device 'kitchen/osram_plug2' left the network
```
[Full logs](https://github.com/Koenkk/zigbee2mqtt/files/11114591/z2m_logs_disconnects.txt)

satter, 2023-04-02:
I've updated z2m to 1.30.3 and my plugs are still leaving the network but the difference they are not returning back  
Looks like it's time to fiddle with the stick FW version

ItsEcholot, 2023-04-03:
Same issue here, running [1.30.3](https://github.com/Koenkk/zigbee2mqtt/releases/tag/1.30.3) and coordinator v20221226.
Worked fine for 1-2 weeks connected directly to the coordinator, would drop of the network even faster (~1h) if connected to a router.

hnykda, 2023-04-06:
I also updated, but the result is the same. 2/5 (always the same ones) drop off from the network after some time. They worked more reasonably in the past (although not perfect), I think they are simply "done"...

andree182, 2023-04-08:
I think my input won't help, but since I last replaced battery on one of my sensors, it started behaving quite erratically - disconnects almost daily, and the temperature measurements have quite a jitter. Once I set permit_join=true and press the sensor button, it connects normally...

I swapped batteries between two of the sensors, and it doesn't seem to be the cause... Quite strange stuff, no idea if I can even debug it somehow, probably I'll connect it to lab power source to check it behaves better.

[image]

hnykda, 2023-05-09:
still an issue

dyarfaradj, 2023-06-02:
I have the same exact issue, it won't reconnect to zigbee2mqtt anymore

closed 2023-07-11 by github-actions[bot]

hnykda, 2024-04-21:
Just that I found that this seems to be a known issue, and it's now described in https://www.zigbee2mqtt.io/devices/WSDCGQ11LM.html and also https://community.hubitat.com/t/xiaomi-aqara-devices-pairing-keeping-them-connected/623 and here https://www.reddit.com/r/homeassistant/comments/kexdru/zigbee2mqtt_aqara_wsdcgq11lm_not_updating/