Koenkk/zigbee2mqtt #17074: Lidl 14153905L, light temperature too cold

fs79, 2023-03-18:
### What happened?

Hi, 
please change the light color from 133 to 500 instead of 333.
Light is capable of having color temperature from 3000K to 6500K.

existing device:
    {
        fingerprint: [{modelID: 'TS0502A', manufacturerName: '_TZ3000_8uaoilu9'}],
        model: '14153905L',
        vendor: 'Lidl',
        description: 'Livarno Home LED floor lamp',
        extend: tuya.extend.light_onoff_brightness_colortemp({colorTempRange: [153, 333], noConfigure: true}),
        configure: async (device, coordinatorEndpoint, logger) => {
            device.getEndpoint(1).saveClusterAttributeKeyValue('lightingColorCtrl', {colorCapabilities: 16});
        },

change to:
    {
        fingerprint: [{modelID: 'TS0502A', manufacturerName: '_TZ3000_8uaoilu9'}],
        model: '14153905L',
        vendor: 'Lidl',
        description: 'Livarno Home LED floor lamp',
        extend: tuya.extend.light_onoff_brightness_colortemp({colorTempRange: [153, **### 500**], noConfigure: true}),
        configure: async (device, coordinatorEndpoint, logger) => {
            device.getEndpoint(1).saveClusterAttributeKeyValue('lightingColorCtrl', {colorCapabilities: 16});
        },

### What did you expect to happen?

Light color should be more than 333 in zigbee2mqtt web interface.

### Zigbee2MQTT version

1.30.2

### Adapter firmware version

20221226

### Adapter

Sonoff USB Stick

Koenkk, 2023-03-18:
Please add the color temperature range as described on https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html#_3-adding-converter-s-for-your-device

Koenkk, 2023-03-20:
Thanks, integrated the fix!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2023-03-20 by Koenkk

paoloantinori, 2023-09-30:
Hi, I was about to report that the color values are off on my lamp: warmest actually outputs a cold white.

Until I've realised that I have a slightly different model:

`14154002L`

I'd be happy to contribute to the correct config.
I've checked this page:
https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html

but the operation for the physical values fails:

```Publish 'set' 'read' to 'Lampada_Divano' failed: 'Error: Read 0xcc86ecfffe0fb0ff/1 lightingColorCtrl(["colorTempPhysicalMin"], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')'```

any hint on how to proceed?