Koenkk/zigbee2mqtt #17080: The model 'TS0601_din_1' is wrong for manufacturerName '_TZE200_eaac7dkw'

zerofull56, 2023-03-19:
### What happened?

My DAC2161C energy meter was mistakenly connected as 'TS0601_din_1' and is either TS0601_din_3 or TS0601_din_4 due to the erroneously added ManufacturerName '_TZE200_eaac7dkw' in the fingerprint for the model 'TS0601_din_1'
Link to commit: [https://github.com/Mephistofeles/zigbee-herdsman-converters/commit/d718ddbdcc0ebd695fbb45f31f5ff7e85c390ea3](url)

### Zigbee2MQTT version

1.30.2 as HA add-on

### Adapter firmware version

20200211

### Adapter

sls gateway as zigbee bridge

Koenkk, 2023-03-20:
You mention TS0601_din_4 but no such device is supported, should it be detected as TS0601_din_3?

zerofull56, 2023-03-22:
I think yes, but i don't know it's will be work correct... Need test it.

Koenkk, 2023-03-22:
Could you check if the issue is fixed with the following external converter: 

```js
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const utils = require('zigbee-herdsman-converters/lib/utils');
const globalStore = require('zigbee-herdsman-converters/lib/store');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    fingerprint: tuya.fingerprint('TS0601', ['_TZE200_eaac7dkw']),
    model: 'TS0601_din_3',
    vendor: 'TuYa',
    description: 'Zigbee DIN energy meter',
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [tuya.tz.datapoints],
    configure: tuya.configureMagicPacket,
    whiteLabel: [{vendor: 'XOCA', model: 'DAC2161C'}],
    exposes: [tuya.exposes.switch(), e.energy(), e.produced_energy(), e.power(), e.voltage(), e.current(),
        exposes.enum('fault', ea.STATE, ['clear', 'over_current_threshold', 'over_power_threshold',
            'over_voltage threshold', 'wrong_frequency_threshold']).withDescription('Fault status of the device (clear = nothing)'),
        exposes.enum('threshold_1', ea.STATE, ['not_set', 'over_current_threshold', 'over_voltage_threshold'])
            .withDescription('State of threshold_1'),
        exposes.binary('threshold_1_protection', ea.STATE, 'ON', 'OFF')
            .withDescription('OFF - alarm only, ON - relay will be off when threshold reached'),
        exposes.numeric('threshold_1_value', ea.STATE)
            .withDescription('Can be in Volt or Ampere depending on threshold setting. Setup the value on the device'),
        exposes.enum('threshold_2', ea.STATE, ['not_set', 'over_current_threshold', 'over_voltage_threshold'])
            .withDescription('State of threshold_2'),
        exposes.binary('threshold_2_protection', ea.STATE, 'ON', 'OFF')
            .withDescription('OFF - alarm only, ON - relay will be off when threshold reached'),
        exposes.numeric('threshold_2_value', ea.STATE)
            .withDescription('Setup value on the device'),
        exposes.binary('clear_fault', ea.STATE_SET, 'ON', 'OFF')
            .withDescription('Turn ON to clear last the fault'),
        exposes.text('meter_id', ea.STATE).withDescription('Meter ID (ID of device)'),
    ],
    meta: {
        tuyaDatapoints: [
            [1, 'energy', tuya.valueConverter.divideBy100],
            [2, 'produced_energy', tuya.valueConverter.divideBy100],
            [3, null, null], // Monthly, but sends data only after request
            [4, null, null], // Dayly, but sends data only after request
            [6, null, tuya.valueConverter.phaseVariant2], // voltage and current
            [10, 'fault', tuya.valueConverterBasic.lookup({'clear': 0, 'over_current_threshold': 1,
                'over_power_threshold': 2, 'over_voltage_threshold': 4, 'wrong_frequency_threshold': 8})],
            [11, null, null], // Frozen - strange function, in native app - nothing is clear
            [16, 'state', tuya.valueConverter.onOff],
            [17, null, tuya.valueConverter.threshold], // It's settable, but can't write converter
            [18, 'meter_id', tuya.valueConverter.raw],
            [20, 'clear_fault', tuya.valueConverter.onOff], // Clear fault
            [21, null, null], // Forward Energy T1 - don't know what this
            [22, null, null], // Forward Energy T2 - don't know what this
            [23, null, null], // Forward Energy T3 - don't know what this
            [24, null, null], // Forward Energy T4 - don't know what this
        ],
    },
};

module.exports = definition;
```

- save this as file next to `configuration.yaml` as `ext_converter.js`
- add it to `configuration.yaml`:
```yaml
external_converters:
  - ext_converter.js
```
- start z2m, check if issue is fixed

zerofull56, 2023-03-22:
Thank you for help, Koenkk! I'm already do it now, but I think, I will need up to date the firmware for this... But I'm not have tuya gateway. Right now i can see voltage, power etc and can't see some another like threshold_1, fault, threshold_1_protection, etc. I used TS0601_din_2 and TS0601_din_3 converters.

zerofull56, 2023-03-31:
Для всех, кто столкнулся с похожей проблемой - вторичные данные не появились, пока я не подключил устройство к шлюзу Tuya. Новых прошивок на момент написания этого поста не было...

closed 2023-03-31 by zerofull56

fred-c1, 2023-06-21:
About
"
            [3, null, null], // Monthly, but sends data only after request
            [4, null, null], // Dayly, but sends data only after request
"

Does someone know how to support these datapoints that need a request?

DP3 is to request "monthly energy".
DP4 is to request "daily energy".

The format should be:

*********** DP3
1，APP ask the Energy from someday to someday(Months).  need APP give the  command first.
2，Big-endian,HEX
3，0.01KWH,  X100
4，Format:
1)Request：Start_year（1 byte）+Start_Month（1byte）+End_year（1byte）+End_Month（1byte）
eg：14 01 14 02 means request energy from Year: 2020, Month: 01 to Year: 2020, Month:02 , 2 Months
2)ACK from MCU：
Start_year（1 byte）+Start_Month（1byte）+End_year（1byte）+End_Month（1byte）+energy(4bytes)
eg：14 01 14 02 00 00 4E 2F，means energy data is 200.15KWH from y from Year: 2020, Month: 01 to Year: 2020, Month:02 

*********** DP4
1，APP ask the Energy from someday to someday(daily).  need APP give the  command first.
2，Big-endian,HEX
3，0.01KWH,  X100
4，Format:
1)Request：Start_Month（1 byte）+Start_day（1byte）+end_month（1byte）+end_day（1byte）
eg：08 0A 08 0F means request energy from August, 10th to August 15th , 6days
2)ACK from MCU：
Start_Month（1 byte）+Start_day（1byte）+Month（1byte）+end_day（1byte）+energy(4bytes)
eg：08 0A 08 0F 14 44，means energy data is 51.88KWH from August, 10th to August 15th