Koenkk/zigbee2mqtt #17088: Error after pairing of water valve GiEX QT06

terierul, 2023-03-20:
### What happened?

I paired this a water valve, this one: https://www.zigbee2mqtt.io/devices/QT06.html.
The interview was ok (see the first picture). But after the interview an error comes (second picture) and I am seeing the sensor as offline.
The nice part is that also the valve appears as offline in the exposes I can change the state and is working. Basically the valve is functional, but the state is offline.

### Zigbee2MQTT version

1.30.2-1

### Adapter firmware version

0x26780700

### Adapter

ConBee2/RaspBee2

closed 2023-04-27 by github-actions[bot]

machv, 2023-05-07:
hi,
I just paired mine and I've also got this error. For a while I could control the valve from Z2M, and manually from the valve (with state properly propagated back to Z2M), but after few hours the valve is not responding to zigbee network at all. Could this connectivity issue be related to that error after pairing?

lucode, 2024-04-12:
I run two GX-02 valves (2024 version) and I can confirm that are connection issues. I needed to place a Zigbee Router quite near to the two valves to have stable connections. I also discovered that a battery level below 70% reduces the connectivity significant. I run it with ZHA.

xnextx, 2024-05-12:
I have the same problem :(

lucode, 2024-05-13:
> I have the same problem :(

According to the GiEX Support, either a fixed firmware or the possibility of getting a new one is under way.