Koenkk/zigbee2mqtt #17097: [New device support]:  TS0603 _TZE600_wxq8dpha (dual 0-10V dimmer)

svde, 2023-03-21:
### Link

https://www.aliexpress.com/item/1005004069503282.html

### Database entry

{"id":3,"type":"Router","ieeeAddr":"0xa4c13859262289d2","nwkAddr":54467,"manufId":4417,"manufName":"_TZE600_wxq8dpha\u0000","powerSource":"Mains (single phase)","modelId":"TS0603","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":257,"inClusterList":[4,5,6,8,768,61184,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"\u0016 *\u0017\u0018 *\u0013","65506":37,"65508":0,"65534":0,"modelId":"TS0603","manufacturerName":"_TZE600_wxq8dpha\u0000","powerSource":1,"zclVersion":3,"appVersion":65,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":65,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":-708457359},"lastSeen":1679406455604,"defaultSendRequestWhen":"immediate"}

### Comments

I tried setting up a TuYa converter for this item but haven't gotten anywhere.

svde, 2023-04-05:
Trying with:

```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const utils = require('zigbee-herdsman-converters/lib/utils');
const globalStore = require('zigbee-herdsman-converters/lib/store');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
//  zigbeeModel: 'TS0603',
    fingerprint: [
        {
            modelID: 'TS0603',
            manufacturerName: '_TZE600_wxq8dpha\u0000',
        },
    ],
    model: 'VM-Zigbee-S02-0-10V',
    vendor: 'Lonsonho',
    description: 'Zigbee dimmer module 2 channel',
    extend: tuya.extend.light_onoff_brightness({minBrightness: true}),
};

module.exports = definition;
```

More action in the logs:
debug 2023-04-05 22:13:53Received Zigbee message from '0xa4c13859262289d2', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[1],"type":"Buffer"},"datatype":1,"dp":1}],"seq":256}' from endpoint 1 with groupID 0
debug 2023-04-05 22:14:02Received Zigbee message from '0xa4c13859262289d2', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[1],"type":"Buffer"},"datatype":1,"dp":1}],"seq":256}' from endpoint 1 with groupID 0
error 2023-04-05 22:14:03Publish 'set' 'state' to '0xa4c13859262289d2' failed: 'Error: Command 0xa4c13859262289d2/1 genOnOff.off({}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 54467 - 1 - 5 - 6 - 11 after 10000ms)'
debug 2023-04-05 22:14:03Error: Command 0xa4c13859262289d2/1 genOnOff.off({}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 54467 - 1 - 5 - 6 - 11 after 10000ms) at Timeout._onTimeout (/app/node_modules/zigbee-herdsman/src/utils/waitress.ts:64:35) at listOnTimeout (node:internal/timers:569:17) at processTimers (node:internal/timers:512:7)
info 2023-04-05 22:14:03MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'state' to '0xa4c13859262289d2' failed: 'Error: Command 0xa4c13859262289d2/1 genOnOff.off({}, {\"sendWhen\":\"immediate\",\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":false,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null,\"writeUndiv\":false}) failed (Timeout - 54467 - 1 - 5 - 6 - 11 after 10000ms)'","meta":{"friendly_name":"0xa4c13859262289d2"},"type":"zigbee_publish_error"}'
debug 2023-04-05 22:14:03Received Zigbee message from '0xa4c13859262289d2', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[1],"type":"Buffer"},"datatype":1,"dp":1}],"seq":256}' from endpoint 1 with groupID 0
warning 2023-04-05 22:14:10Device '0xa4c13859262289d2' left the network
info 2023-04-05 22:14:10MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xa4c13859262289d2","ieee_address":"0xa4c13859262289d2"},"type":"device_leave"}'
info 2023-04-05 22:14:10MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"left_network","meta":{"friendly_name":"0xa4c13859262289d2"},"type":"device_removed"}'
info 2023-04-05 22:14:14Device '0xa4c13859262289d2' joined
info 2023-04-05 22:14:14MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xa4c13859262289d2","ieee_address":"0xa4c13859262289d2"},"type":"device_joined"}'
info 2023-04-05 22:14:14MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":{"friendly_name":"0xa4c13859262289d2"},"type":"device_connected"}'
info 2023-04-05 22:14:14Starting interview of '0xa4c13859262289d2'
info 2023-04-05 22:14:14MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xa4c13859262289d2","ieee_address":"0xa4c13859262289d2","status":"started"},"type":"device_interview"}'
info 2023-04-05 22:14:14MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_started","meta":{"friendly_name":"0xa4c13859262289d2"},"type":"pairing"}'
debug 2023-04-05 22:14:14Received Zigbee message from '0xa4c13859262289d2', type 'attributeReport', cluster 'genBasic', data '{"65506":37,"65508":1,"appVersion":65}' from endpoint 1 with groupID 0
debug 2023-04-05 22:14:18Received Zigbee message from '0xa4c13859262289d2', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[1],"type":"Buffer"},"datatype":1,"dp":1}],"seq":256}' from endpoint 1 with groupID 0
debug 2023-04-05 22:14:18Received Zigbee message from '0xa4c13859262289d2', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[1],"type":"Buffer"},"datatype":1,"dp":1}],"seq":256}' from endpoint 1 with groupID 0
[... cut ...]
debug 2023-04-05 22:14:45Received Zigbee message from '0xa4c13859262289d2', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,3,232],"type":"Buffer"},"datatype":2,"dp":11}],"seq":3328}' from endpoint 1 with groupID 0
debug 2023-04-05 22:14:45Received Zigbee message from '0xa4c13859262289d2', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[2],"type":"Buffer"},"datatype":4,"dp":14}],"seq":3584}' from endpoint 1 with groupID 0
debug 2023-04-05 22:14:57Received Zigbee message from '0xa4c13859262289d2', type 'read', cluster 'genTime', data '["localTime"]' from endpoint 1 with groupID 0

svde, 2023-05-06:
still not fixed

svde, 2023-07-08:
Please keep issue open.

LiZhenhuan1019, 2023-09-18:
Please try this one, which works for most functions of this device. I didn't figure out the meaning of datapoint 14 though.
```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const utils = require('zigbee-herdsman-converters/lib/utils')

const tzLocal = {
    switchType: {
        key: ['switch_type'],
        convertSet: async (entity, key, value, meta) => {
             // A set converter is only called once; therefore we need to loop
             const state = {};
             const datapoints = meta.mapped.meta?.tuyaDatapoints;
             if (!datapoints) throw new Error('No datapoints map defined');
             for (const [attr, value] of Object.entries(meta.message)) {
                 const convertedKey = meta.mapped.meta.multiEndpoint && meta.endpoint_name && !attr.startsWith(`${key}_`) ?
                     `${attr}_${meta.endpoint_name}` : attr;
                 const dpEntry = datapoints.find((d) => d[1] === convertedKey);
                 if (!dpEntry?.[1] || !dpEntry?.[2].to) {
                     throw new Error(`No datapoint defined for '${attr}'`);
                 }
                 if (dpEntry[3] && dpEntry[3].skip && dpEntry[3].skip(meta)) continue;
                 const dpId = dpEntry[0];
                 const convertedValue = await dpEntry[2].to(value, meta);
                 const sendCommand = utils.getMetaValue(entity, meta.mapped, 'tuyaSendCommand', undefined, 'dataRequest');
                 if (convertedValue === undefined) {
                     // conversion done inside converter, ignore.
                 } else if (typeof convertedValue === 'boolean') {
                     await tuya.sendDataPointBool(entity, dpId, convertedValue, sendCommand, 1);
                 } else if (typeof convertedValue === 'number') {
                     await tuya.sendDataPointValue(entity, dpId, convertedValue, sendCommand, 1);
                 } else if (typeof convertedValue === 'string') {
                     await tuya.sendDataPointStringBuffer(entity, dpId, convertedValue, sendCommand, 1);
                 } else if (Array.isArray(convertedValue)) {
                     await tuya.sendDataPointRaw(entity, dpId, convertedValue, sendCommand, 1);
                 } else if (convertedValue instanceof tuya.Enum) {
                     await tuya.sendDataPointEnum(entity, dpId, convertedValue.valueOf(), sendCommand, 1);
                 } else if (convertedValue instanceof tuya.Bitmap) {
                     await tuya.sendDataPointBitmap(entity, dpId, convertedValue.valueOf(), sendCommand, 1);
                 } else {
                     throw new Error(`Don't know how to send type '${typeof convertedValue}'`);
                 }
 
                 if (dpEntry[3] && dpEntry[3].optimistic === false) continue;
 
                 state[key] = value;
             }
             return {state};
        }
    }
}
const definition = {
    // Since a lot of TuYa devices use the same modelID, but use different datapoints
    // it's necessary to provide a fingerprint instead of a zigbeeModel
    fingerprint: [
        {
            // The model ID from: Device with modelID 'TS0601' is not supported
            // You may need to add \u0000 at the end of the name in some cases
            modelID: 'TS0603',
            // The manufacturer name from: Device with modelID 'TS0601' is not supported.
            manufacturerName: '_TZE600_wxq8dpha\u0000',
        },
    ],
    model: 'VM-Zigbee-S02-0-10V',
[... 18 lines cut ...]
        return {'l1': 1, 'l2': 1, 'l3': 1};
    },
    meta: {
        multiEndpoint: true,
        // All datapoints go in here
        tuyaDatapoints: [
            [1, 'state_l1', tuya.valueConverter.onOff],
            [2, 'brightness_l1', tuya.valueConverter.scale0_254to0_1000],
            [3, 'min_brightness_l1', tuya.valueConverter.scale0_254to0_1000], // * [3], // {"data":{"data":[0,0,0,11],"type":"Buffer"},"datatype":2,"dp":3}
            [4, 'switch_type_l1', tuya.valueConverterBasic.lookup({'momentary': new tuya.Enum(0), 'toggle': new tuya.Enum(1), 'state': new tuya.Enum(2)})], // [4], //{"data":{"data":[0],"type":"Buffer"},"datatype":4,"dp":4}
            [5, 'max_brightness_l1', tuya.valueConverter.scale0_254to0_1000], // * [5], //{"data":{"data":[0,0,3,232],"type":"Buffer"},"datatype":2,"dp":5}
            [6, 'countdown_l1', tuya.valueConverter.raw], // [6], // {"data":{"data":[0,0,0,0],"type":"Buffer"},"datatype":2,"dp":6}
            [7, 'state_l2', tuya.valueConverter.onOff], // [7], //{"data":{"data":[1],"type":"Buffer"},"datatype":1,"dp":7}
            [8, 'brightness_l2',tuya.valueConverter.scale0_254to0_1000], //{"data":{"data":[0,0,3,232],"type":"Buffer"},"datatype":2,"dp":8}
            [9, 'min_brightness_l2', tuya.valueConverter.scale0_254to0_1000], // * [9], //{"data":{"data":[0,0,0,11],"type":"Buffer"},"datatype":2,"dp":9}
            [10, 'switch_type_l2', tuya.valueConverterBasic.lookup({'momentary': new tuya.Enum(0), 'toggle': new tuya.Enum(1), 'state': new tuya.Enum(2)})], // [10], //{"data":{"data":[0],"type":"Buffer"},"datatype":4,"dp":10}
            [11, 'max_brightness_l2', tuya.valueConverter.scale0_254to0_1000], // * [11], //{"data":{"data":[0,0,3,232],"type":"Buffer"},"datatype":2,"dp":11}
            [12, 'countdown_l2', tuya.valueConverter.raw], //[12], //{"data":{"data":[0,0,0,0],"type":"Buffer"},"datatype":2,"dp":12}
            // * [14], //{"data":{"data":[2],"type":"Buffer"},"datatype":4,"dp":14}
        ],
    },
};

module.exports = definition;
```

svde, 2023-09-21:
YES! This works. Thank you!

@Koenkk please add it so the device is supported without external converter.

Koenkk, 2023-09-23:
@LiZhenhuan1019 already added it, you can remove the ext converter after upgrading to the next release.

closed 2023-09-23 by Koenkk