Koenkk/zigbee2mqtt #17099: [New device support]: Tuya TS0202

Pawel22103, 2023-03-21:
### Link

https://www.zigbee2mqtt.Io/devices/TS0202.html

### Database entry

Warning 2023-03-21 16:14:12Received message from unsupported device with Zigbee model 'TS0202' and manufacturer name '_TZ3000_wrgn6xrz' Warning 2023-03-21 16:14:12Please see: https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html Debug 2023-03-21 16:14:12Received Zigbee message from 'Czujnik obecności łazienka', type 'attributeReport', cluster 'ssIasZone', data '{"61441":1}' from endpoint 1 with groupID 0

### Comments

I triem to ADHD na extremalnie converter but it doesn’t work. Device added to zigbee2mqtt does not expose anything. Please support.

Madaraxa, 2023-03-21:
I think i got the same Tuya Sensor (if its the "new" vibration sensor). 
Its this sensor:
[https://drive.google.com/file/d/1B_LcelDfzl2EREIwTPjkyXV57pCq9M5m/view]
When trying to pair i get the following messages (i dont know if it helps thought):
`
Starting interview of '0xa4c138f3f877674f'
States user redis pmessage zigbee.0.*/zigbee.0.info.pairingMessage:{"val":"Interview started: 0xa4c138f3f877674f","ack":true,"ts":1679427234437,"q":0,"from":"system.adapter.zigbee.0","user":"system.user.admin","lc":1679427234437}
Failed to execute routing table for '0xa4c138f3f877674f'. "Error: AREQ - ZDO - mgmtRtgRsp after 10000ms\n at Timeout._onTimeout (/opt/iobroker/node_modules/zigbee-herdsman/src/utils/waitress.ts:64:35)\n at listOnTimeout (node:internal/timers:569:17)\n at processTimers (node:internal/timers:512:7)"
Routing for '0xa4c138f3f877674f': undefined
Routing table succeeded for '0xa4c138f3f877674f'
`
I use zigbee on iobroker and not zigbee2mqtt directly. Anyway i hope this helps. I cant add any more information as it seems the pairing itself fails

Koenkk, 2023-03-21:
Added!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2023-03-21 by Koenkk

Pawel22103, 2023-03-22:
I must use external converter?

Koenkk, 2023-03-22:
No, dev branch or wait until the next release on 1 April