Koenkk/zigbee2mqtt #17100: [New device support]: Tongou TO-Q-SY2-JZT advanced din rail circuit breaker

henkiejan1, 2023-03-21:
### Link

https://elcb.net/product/zigbee-smart-circuit-breaker-over-current-under-voltage-protection-to-q-sy2-jzt/

### Database entry

https://elcb.net/product/zigbee-smart-circuit-breaker-over-current-under-voltage-protection-to-q-sy2-jzt/

### Comments

This is a new smart dinrail switch with more options. Not only is it possible to measure the current and electricity. But also you can set some advanced settings what not is recognized by Zigbee2mqtt:
- Turn off current (0-63A)
- Turn off under voltage
- Turn off over voltage
- Turn off by temperature limit
- Countdown timer

It now detected as a ordinary Tuya TS011_F switch with only power metering. Should be nice to support al functions

henkiejan1, 2023-04-01:
Does anyone have an idea how to support/add the advanced functions of this device? I read something about external converters but looks really difficult...

henkiejan1, 2023-04-11:
What i see looks like this device very similar to the Tongou one? Maybe it could be related? https://github.com/Koenkk/zigbee2mqtt/issues/16972

matlab22, 2023-04-20:
I have the same and mine got detected as "_TZ3000_cayepv1a" with limited functionality.

- Power
- Current
- Voltage
- Energy -> not sure if this is working. I tested it with a 40W bulb and the unit is kWh. Time will tell :D

There is also a Child Lock, when I set it the button on the device is not working as expected, but I can not turn it on remotely ether.

Would be nice to enable the other features as well:

- Turn off current (0-63A)
- Turn off under voltage
- Turn off over voltage
- Turn off by temperature limit
- Countdown timer

Does anybody know how to do this?

henkiejan1, 2023-05-01:
It's just added with the latest release @matlab22 ! Update to the latest Zigbee2mqtt 1.30.4 and delete and add the switch again. I will close this section.

closed 2023-05-01 by henkiejan1

eekamouze, 2023-05-24:
Ok, forget about my question, sorry. I found the answer there  #17456

NorDroN, 2024-01-07:
Hi!
What about OTA updates? 
[image]
[image]

Z2M 1.35.1

fernandeusto, 2024-01-24:
same OTA problem here

Elkiff71, 2024-01-30:
I have the same problem

lolvince, 2024-01-31:
I have the same problem

Trebor87, 2024-02-08:
I have the same message occurring as well ;(

sarzyk, 2025-03-23:
Is possible to add it z2m and Tuya in the same time? (Zígbee version)

Rookie805, 2025-11-15:
Advanced functions are still not working, but present in device page. Just getting errors, when trying to set.

What did you expect to happen?

I expect that mentioned values are set in device.

How to reproduce it (minimal and precise)

Just try to set values in device page.

Zigbee2MQTT version

2.6.2
Adapter firmware version

EmberZNet 8.0.2 [GA]
Adapter

Sonoff ZBDongle-E
Setup

haos
Debug log

[15.11.2025, 19:50:59] z2m: Publish 'set' 'temperature_threshold' to 'Pump' failed: 'Error: Expected string or number, got: undefined'

[15.11.2025, 19:51:00] z2m: Publish 'set' 'temperature_breaker' to 'Pump' failed: 'TypeError: Cannot read properties of undefined (reading 'toString')'

[15.11.2025, 19:51:02] z2m: Publish 'set' 'power_threshold' to 'Pump' failed: 'Error: Expected string or number, got: undefined'

[15.11.2025, 19:51:02] z2m: Publish 'set' 'power_breaker' to 'Pump' failed: 'TypeError: Cannot read properties of undefined (reading 'toString')'

[15.11.2025, 19:51:04] z2m: Publish 'set' 'over_current_threshold' to 'Pump' failed: 'Error: Expected string or number, got: undefined'

[15.11.2025, 19:51:06] z2m: Publish 'set' 'over_current_breaker' to 'Pump' failed: 'TypeError: Cannot read properties of undefined (reading 'toString')'

[15.11.2025, 19:51:08] z2m: Publish 'set' 'over_voltage_threshold' to 'Pump' failed: 'Error: Expected string or number, got: undefined'

[15.11.2025, 19:51:09] z2m: Publish 'set' 'over_voltage_breaker' to 'Pump' failed: 'TypeError: Cannot read properties of undefined (reading 'toString')'

[15.11.2025, 19:51:12] z2m: Publish 'set' 'under_voltage_threshold' to 'Pump' failed: 'Error: Expected string or number, got: undefined'

[15.11.2025, 19:51:13] z2m: Publish 'set' 'under_voltage_breaker' to 'Pump' failed: 'TypeError: Cannot read properties of undefined (reading 'toString')'

JordiPF, 2025-12-04:
Same error for me. any solution?