Koenkk/zigbee2mqtt #17103: [New device support]:

l154, 2023-03-21:
### Link

https://www.bosch-smarthome.com/uk/en/products/devices/water-alarm/

### Database entry

{"id":4,"type":"EndDevice","ieeeAddr":"0x804b50fffecedf12","nwkAddr":41651,"manufId":4617,"manufName":"Bosch","powerSource":"Battery","modelId":"RBSH-WS-ZB-EU","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":1026,"inClusterList":[0,1,3,32,1280,2821,64684],"outClusterList":[25],"clusters":{"64684":{"attributes":{"0":0}},"ssIasZone":{"attributes":{"iasCieAddr":"0x00124b0025e15a13","zoneState":1}},"genPollCtrl":{"attributes":{"checkinInterval":6480}},"genPowerCfg":{"attributes":{"batteryPercentageRemaining":200,"batteryVoltage":30}}},"binds":[{"cluster":32,"type":"endpoint","deviceIeeeAddress":"0x00124b0025e15a13","endpointID":1},{"cluster":1,"type":"endpoint","deviceIeeeAddress":"0x00124b0025e15a13","endpointID":1}],"configuredReportings":[],"meta":{}}},"appVersion":2,"stackVersion":6,"hwVersion":1,"dateCode":"20210205-18","swBuildId":"2.02","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1679429731565,"defaultSendRequestWhen":"active","pollCheckingInterval":6480}

### Comments

Everything work like it should but there is an error:

```
error 2023-03-21 21:18:08: Failed to configure '0x804b50fffecedf12', attempt 3 (Error: Cluster with name '64684' does not exist
    at Object.getCluster (/app/node_modules/zigbee-herdsman/src/zcl/utils.ts:61:19)
    at Endpoint.bind (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:579:35)
    at Object.bind (/app/node_modules/zigbee-herdsman-converters/lib/reporting.js:40:24)
    at Object.configure (/app/data/extension/externally-loaded.js:73:9)
    at Configure.configure (/app/lib/extension/configure.ts:117:13))
```

### External converter

```shell
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const utils = require('zigbee-herdsman-converters/lib/utils');
const constants = require('zigbee-herdsman-converters/lib/constants');
const ota = require('zigbee-herdsman-converters/lib/ota');
const e = exposes.presets;
const ea = exposes.access;

const boschManufacturer = {manufacturerCode: 0x1209};

const stateOffOn = {
    'OFF': 0,
    'ON': 1,
};

const tzLocal = {
    bwa1: {
        key: ['alarm_on_motion', 'test'],
        convertSet: async (entity, key, value, meta) => {
            if (key === 'alarm_on_motion') {
                value = value.toUpperCase();
                const index = stateOffOn[value];
                await entity.write(0xFCAC, {0x0003: {value: index, type: 0x10}}, boschManufacturer);
                return {state: {alarm_on_motion: value}};
            }
        },
        convertGet: async (entity, key, meta) => {
            switch (key) {
            case 'alarm_on_motion':
                await entity.read(0xFCAC, [0x0003], boschManufacturer);
                break;
            default: // Unknown key
                throw new Error(`Unhandled key toZigbee.bosch_twinguard.convertGet ${key}`);
            }
        },
    },
};

const fzLocal = {
    bwa1_alarm_on_motion: {
        cluster: '64684',
        type: ['attributeReport', 'readResponse'],
        options: [],
        convert: (model, msg, publish, options, meta) => {
            const result = {};
            const data = msg.data;
            if (data.hasOwnProperty(0x0003)) {
                result.alarm_on_motion = (Object.keys(stateOffOn)[msg.data[0x0003]]);
            }
            return result;
        },
    },
};

const definition = {
    zigbeeModel: ['RBSH-WS-ZB-EU'],
    model: 'BWA-1',
    vendor: 'Bosch',
    description: 'Zigbee smart water leak detector',
    fromZigbee: [fz.ias_water_leak_alarm_1, fz.battery, fzLocal.bwa1_alarm_on_motion],
    exposes: [e.water_leak(), e.battery(), e.tamper(),
    exposes.binary('alarm_on_motion', ea.ALL, 'ON', 'OFF')
                .withDescription('Enable/Disable sound alarm on motion'),
    ],
    toZigbee: [tzLocal.bwa1],
    meta: {battery: {voltageToPercentage: '3V_2500'}},
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(1);
        await reporting.bind(endpoint, coordinatorEndpoint, ['genPowerCfg', '64684']);
        await reporting.batteryPercentageRemaining(endpoint);
        await reporting.batteryVoltage(endpoint);
        await endpoint.configureReporting(0xFCAC, [{
            attribute: {ID: 0x0003, type: herdsman.Zcl.DataType.bool},
            minimumReportInterval: 0,
            maximumReportInterval: constants.repInterval.HOUR,
            reportableChange: 1,
        }], boschManufacturer);
    },
};

module.exports = definition;
```

Koenkk, 2023-03-22:
Removing the following from the converter should fix the error:

```
        await endpoint.configureReporting(0xFCAC, [{
            attribute: {ID: 0x0003, type: herdsman.Zcl.DataType.bool},
            minimumReportInterval: 0,
            maximumReportInterval: constants.repInterval.HOUR,
            reportableChange: 1,
        }], boschManufacturer);
```

junosuarez, 2023-03-22:
Please edit the issue title to include the name of the new device, like `[New Device] Bosch Water Alarm RBSH-WS-ZB-EU`

l154, 2023-03-22:
hi Koenkk,
the problem is in this  line:
`await reporting.bind(endpoint, coordinatorEndpoint, ['genPowerCfg', '64684']);`
when I remove the cluster name from here, the problem is fixed but then I can't configure "alarm_on_motion
Enable/Disable sound alarm on motion".
so probably the cluster name should be added to zigmee2mqtt.

junosuarez
the name was changed

```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const utils = require('zigbee-herdsman-converters/lib/utils');
const constants = require('zigbee-herdsman-converters/lib/constants');
const ota = require('zigbee-herdsman-converters/lib/ota');
const e = exposes.presets;
const ea = exposes.access;

const boschManufacturer = {manufacturerCode: 0x1209};

const stateOffOn = {
    'OFF': 0,
    'ON': 1,
};

const tzLocal = {
    bwa1: {
        key: ['alarm_on_motion', 'test'],
        convertSet: async (entity, key, value, meta) => {
            if (key === 'alarm_on_motion') {
                value = value.toUpperCase();
                const index = stateOffOn[value];
                await entity.write(0xFCAC, {0x0003: {value: index, type: 0x10}}, boschManufacturer);
                return {state: {alarm_on_motion: value}};
            }
        },
        convertGet: async (entity, key, meta) => {
            switch (key) {
            case 'alarm_on_motion':
                await entity.read(0xFCAC, [0x0003], boschManufacturer);
                break;
            default: // Unknown key
                throw new Error(`Unhandled key toZigbee.bosch_twinguard.convertGet ${key}`);
            }
        },
    },
};

const fzLocal = {
    bwa1_alarm_on_motion: {
        cluster: '64684',
        type: ['attributeReport', 'readResponse'],
        options: [],
        convert: (model, msg, publish, options, meta) => {
            const result = {};
            const data = msg.data;
            if (data.hasOwnProperty(0x0003)) {
                result.alarm_on_motion = (Object.keys(stateOffOn)[msg.data[0x0003]]);
            }
            return result;
        },
    },
};

const definition = {
    zigbeeModel: ['RBSH-WS-ZB-EU'],
    model: 'BWA-1',
    vendor: 'Bosch',
    description: 'Bosch Water Alarm RBSH-WS-ZB-EU',
    fromZigbee: [fz.ias_water_leak_alarm_1, fz.battery, fzLocal.bwa1_alarm_on_motion],
    exposes: [e.water_leak(), e.battery(), e.tamper(),
    exposes.binary('alarm_on_motion', ea.ALL, 'ON', 'OFF')
                .withDescription('Enable/Disable sound alarm on motion'),
    ],
    toZigbee: [tzLocal.bwa1],
    meta: {battery: {voltageToPercentage: '3V_2500'}},
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(1);
        await reporting.bind(endpoint, coordinatorEndpoint, ['genPowerCfg', '64684']);
        await reporting.batteryPercentageRemaining(endpoint);
        await reporting.batteryVoltage(endpoint);
    },
};

module.exports = definition;
```

l154, 2023-03-22:
hi Koenkk,
I found the almost the same error that was fixed but for other cluster name

https://github.com/Koenkk/zigbee2mqtt/discussions/10870

l154, 2023-03-23:
> Try removing the quotes ` '64684'` -> ` 64684`
Changed. No errors from now. Issue solved.
```
Correct configuration below:

const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const utils = require('zigbee-herdsman-converters/lib/utils');
const constants = require('zigbee-herdsman-converters/lib/constants');
const ota = require('zigbee-herdsman-converters/lib/ota');
const e = exposes.presets;
const ea = exposes.access;

const boschManufacturer = {manufacturerCode: 0x1209};

const stateOffOn = {
    'OFF': 0,
    'ON': 1,
};

const tzLocal = {
    bwa1: {
        key: ['alarm_on_motion', 'test'],
        convertSet: async (entity, key, value, meta) => {
            if (key === 'alarm_on_motion') {
                value = value.toUpperCase();
                const index = stateOffOn[value];
                await entity.write(0xFCAC, {0x0003: {value: index, type: 0x10}}, boschManufacturer);
                return {state: {alarm_on_motion: value}};
            }
        },
        convertGet: async (entity, key, meta) => {
            switch (key) {
            case 'alarm_on_motion':
                await entity.read(0xFCAC, [0x0003], boschManufacturer);
                break;
            default: // Unknown key
                throw new Error(`Unhandled key toZigbee.bosch_twinguard.convertGet ${key}`);
            }
        },
    },
};

const fzLocal = {
    bwa1_alarm_on_motion: {
        cluster: '64684',
        type: ['attributeReport', 'readResponse'],
        options: [],
        convert: (model, msg, publish, options, meta) => {
            const result = {};
            const data = msg.data;
            if (data.hasOwnProperty(0x0003)) {
                result.alarm_on_motion = (Object.keys(stateOffOn)[msg.data[0x0003]]);
            }
            return result;
        },
    },
};

const definition = {
    zigbeeModel: ['RBSH-WS-ZB-EU'],
    model: 'BWA-1',
    vendor: 'Bosch',
    description: 'Bosch Water Alarm RBSH-WS-ZB-EU',
    fromZigbee: [fz.ias_water_leak_alarm_1, fz.battery, fzLocal.bwa1_alarm_on_motion],
    exposes: [e.water_leak(), e.battery(), e.tamper(),
    exposes.binary('alarm_on_motion', ea.ALL, 'ON', 'OFF')
                .withDescription('Enable/Disable sound alarm on motion'),
    ],
    toZigbee: [tzLocal.bwa1],
    meta: {battery: {voltageToPercentage: '3V_2500'}},
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(1);
        await reporting.bind(endpoint, coordinatorEndpoint, ['genPowerCfg', 64684]);
        await reporting.batteryPercentageRemaining(endpoint);
        await reporting.batteryVoltage(endpoint);
    },
};

module.exports = definition;

```

closed 2023-05-10 by Koenkk