Koenkk/zigbee2mqtt #17104: [EnOcean PTM 216Z] Cannot read properties of undefined (reading 'raw')

Thib1404, 2023-03-21:
### What happened?

I have a Zigbee Green Power switch, a EnOcean PTM 216Z.
The device is supported.
I followed the pairing intructions for channel 25 as mentioned on this two links ;
https://www.zigbee2mqtt.io/devices/PTM_216Z.html
https://www.enocean.com/wp-content/uploads/downloads-produkte/en/products/enocean_modules_24ghz/ptm-216z/user-manual-pdf/PTM-216Z-User-Manual-1.pdf
-> Section 5.2

The device pairs and appears in the list of devices.
I have a philips Hue on the network. A Gu10 white and color. I also tried with a e25 w&c and a hue play bar.

When pressing any button on the switch I have this error

Zigbee2MQTT:error 2023-03-21 20:47:57: Exception while calling fromZigbee converter: Cannot read properties of undefined (reading 'raw')}
Zigbee2MQTT:debug 2023-03-21 20:47:57: TypeError: Cannot read properties of undefined (reading 'raw')
    at Object.convert (/app/node_modules/zigbee-herdsman-converters/converters/fromZigbee.js:4886:62)
    at Receive.onDeviceMessage (/app/lib/extension/receive.ts:143:51)
    at EventEmitter.emit (node:events:525:35)
    at EventBus.emitDeviceMessage (/app/lib/eventBus.ts:102:22)
    at Controller.<anonymous> (/app/lib/zigbee.ts:106:27)
    at Controller.emit (node:events:513:28)
    at Controller.selfAndDeviceEmit (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:515:14)
    at Controller.onZclOrRawData (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:726:18)

[image]

### What did you expect to happen?

Not an error

[image]

### Zigbee2MQTT version

1.30.2

### Adapter firmware version

6.10.3.0 build 297

### Adapter

SONOFF ZBDongle-E

### Debug log

[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/11033510/log.txt)

closed 2023-04-29 by github-actions[bot]

GhisHD, 2023-10-01:
Hello @Thib1404 
I was face to the same issue .
  
Did you solve the issue finally ?

sebastienserre, 2024-02-12:
Same issue here

danielpetrovic, 2024-02-13:
No problems here, but I am using the Niko FriendsOfHue which are based on the [PTM 215Z ](https://www.zigbee2mqtt.io/devices/PTM_215Z.html).

Keep in mind that for Zigbee Green Power to work, that your button needs to be in the region of the Hue lamp that you want to use as 'translator' for the Zigbee Green Power messages. Simply having the Hue lamps inside your Zigbee network is not enough. You don't need to control that specific light with it, it is just the translator.

When pairing it is advisable to select a specific Hue light to 'pair' it with instead of through 'all' . This way you are only sending unicast Zigbee signals and this will enhance your performance greatly, especially on bigger Zigbee networks.

We have only Hue lights, 47 all across the house and outside, so I can place the Friends of Hue buttons anywhere and have currently 12 of them working with 0 issues. We are using them to control our Somfy RTS and IO motors through HA.