Koenkk/zigbee2mqtt #17105: [New device support]: WZ5 RGB CCT (_TZB210_3zfp8mki)

jonwilliams84, 2023-03-21:
### Link

https://a.aliexpress.com/_m0SgDGO

### Database entry

{"id":37,"type":"Router","ieeeAddr":"0xa4c1389c0eadc306","nwkAddr":49374,"manufId":4417,"manufName":"_TZB210_3zfp8mki","powerSource":"Mains (single phase)","modelId":"TS0505B","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":269,"inClusterList":[3,4,5,6,4096,8,768,61184,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"modelId":"TS0505B","manufacturerName":"_TZB210_3zfp8mki","powerSource":1,"zclVersion":3,"appVersion":64,"stackVersion":0,"hwVersion":1,"dateCode":""}},"lightingColorCtrl":{"attributes":{"colorCapabilities":27,"colorTempPhysicalMin":153,"colorTempPhysicalMax":500,"colorMode":2,"currentX":45494,"currentY":19418,"currentHue":16,"currentSaturation":16,"colorTemperature":153}},"genOnOff":{"attributes":{"onOff":0}},"genLevelCtrl":{"attributes":{"currentLevel":254}}},"binds":[],"configuredReportings":[],"meta":{"onLevelSupported":false}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":64,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":88764544},"lastSeen":1679354028239,"defaultSendRequestWhen":"immediate"}

### Comments

My device is actually like this (or a derivative of this): https://www.zigbee2mqtt.io/devices/WZ5_rgbcct.html

The device is detected in Z2M, however the colours are way off (red is orange, blue is not true blue and green is close but not green). White and cool white appear to be mostly working but the secondary issue below is evident across any state of the settings.

The secondary issue is the status of the device in z2m is not representative of the real state when controlled via the external rf remote, even forcefully refreshing the state of the colour selection does not feed back the actual colour.

I have tried a modified external converter from an earlier issue thread, but this does not allow control of the device at all, won't even toggle on and off.

I believe the problem may be with the colour modes supported on the device is per the skydance wz5 rgbcct product and not the color_xy as per the TS0505B device that Z2M believes it to be.

I believe this maybe the case as I also have the WiFi version of this product and can see there are 3 colour modes. 

Can I override this somehow and get Z2M to detect the device as the Skydance wz5 rgbcct?

### External converter

```shell
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
  fingerprint: [
      {modelID: 'TS0601', manufacturerName: '_TZE200_mde0utnv'},
      {modelID: 'TS0601', manufacturerName: '_TZE200_aa9awrng'},
      {modelID: 'TS0501B', manufacturerName: '_TZB210_rkgngb5o'},
      {modelID: 'TS0505B', manufacturerName: '_TZB210_3zfp8mki'},
  ],
  model: 'WZ5_rgbcct',
  vendor: 'Skydance',
  description: 'Zigbee & RF 5 in 1 LED controller (RGB+CCT mode)',
  fromZigbee: [fz.tuya_light_wz5],
  toZigbee: [tz.tuya_dimmer_state, tz.tuya_light_wz5],
  exposes: [
      exposes.light().withBrightness().setAccess('state', ea.STATE_SET).setAccess('brightness',
          ea.STATE_SET).withColor(['hs']).withColorTemp([250, 454]).setAccess('color_temp',
          ea.STATE_SET),
      exposes.numeric('white_brightness', ea.STATE_SET).withValueMin(0).withValueMax(254).withDescription(
          'White brightness of this light'),
  ],
  meta: {separateWhite: true},
};

module.exports = definition;
```

jonwilliams84, 2023-03-21:
I have continued to tinker this evening and now have colour_hs exposed instead of colour_xy...but it still appears that the colours are way off. It looks like the white is staying on when RGB is in play, instead of keeping CCT and RGB separate and only activating one at any time.

This is as far as I got with the converter:

`
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
  fingerprint: [{modelID: 'TS0505B', manufacturerName: '_TZB210_3zfp8mki'}],
  model: 'TS0505B',
  vendor: 'TuYa',
  description: 'Zigbee RGB+CCT light',
  exposes: [e.light_brightness_colortemp_colorhs([153, 500]).removeFeature('color_temp_startup')],
  extend: tuya.extend.light_onoff_brightness_colortemp_color({colorTempRange: [153, 500], supportsHS: true, preferHS: true, disableEffect: true}),
  meta: {separateWhite: true, applyRedFix: true, enhancedHue: false},
};

module.exports = definition;
`

jonwilliams84, 2023-04-11:
> I have continued to tinker this evening and now have colour_hs exposed instead of colour_xy...but it still appears that the colours are way off. It looks like the white is staying on when RGB is in [...]

Still no further with this, but today realised that with the above convertor the colours are actually working correctly when changed in home assistant! However, they don't work correctly from the zigbee2mqtt UI!

As it stands, I am actually fine with it working in HA only. 

However, I would like to know how to get the status to update if controlled via the remote if possible?

closed 2023-05-20 by github-actions[bot]

LaurentiuGrama, 2024-01-28:
Any update on this device?
I have the same issue :(