Koenkk/zigbee2mqtt #17108: Device doesn't support reporting

brywithawhy, 2023-03-22:
### What happened?

I have been using an external converter as provided in [#13190](https://github.com/Koenkk/zigbee2mqtt/issues/13190) for the same socket I have in Australia.

The on/off functions generally work however I believe my device doesn't support reporting. I get an error message on startup and during configuration

`
Failed to configure 'Bathroom GPO', attempt 2 (Error: ConfigureReporting 0x000d6ffffe9d363c/1 haElectricalMeasurement([{"attribute":"rmsVoltage","minimumReportInterval":5,"maximumReportInterval":3600,"reportableChange":5}], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 5635 - 1 - 6 - 2820 - 7 after 10000ms) at Timeout._onTimeout (/app/node_modules/zigbee-herdsman/src/utils/waitress.ts:64:35) at listOnTimeout (node:internal/timers:559:17) at processTimers (node:internal/timers:502:7))
`

I can manually read the datapoints however I am not sure how I can get them to update regularly automatically.

### What did you expect to happen?

Energy data to be reported however shows N/A or the last value I request manually in the dev console.

### Zigbee2MQTT version

1.30.2-1

### Adapter firmware version

20220219

### Adapter

Sonoff Zigbee 3.0 USB Dongle Plus

brywithawhy, 2023-04-22:
Not Stale

brywithawhy, 2023-05-15:
Bump - can @Koenkk or anyone help? Happy to sponsor or throw some $$ your way to get this working, as I want to install these power sockets throughout the house

brywithawhy, 2023-05-15:
Update:
I've hacked together parts from other converters I've found and have it very close to working.

Problems:
* Current divisor can't be set? Currently reporting 4000 amps when 4 amps drawn.
* Missing Power Restore options - adding this seems to crash zigbee2mqtt
* Can't figure out how to reset power usage meter

Here is the line from database.db:
`{"id":40,"type":"Router","ieeeAddr":"0x000d6ffffe9d363c","nwkAddr":8924,"manufId":4098,"manufName":"_TZ3000_raviyuvk","powerSource":"Mains (single phase)","modelId":"TS011F","epList":[1,2],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[0,4,5,6,1794,2820],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"\u0000\u0000\u0000\u0000\u0005","65506":31,"65534":0,"modelId":"TS011F","manufacturerName":"_TZ3000_raviyuvk","powerSource":1,"zclVersion":3,"appVersion":68,"stackVersion":0,"hwVersion":1,"dateCode":""}},"genOnOff":{"attributes":{"onOff":0,"onTime":0,"offWaitTime":0,"moesStartUpOnOff":1}},"seMetering":{"attributes":{"currentSummDelivered":[0,2],"multiplier":1}},"haElectricalMeasurement":{"attributes":{"rmsVoltage":244,"activePower":4,"rmsCurrent":32}},"genPowerCfg":{"attributes":{}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b0024c0522c","endpointID":1},{"cluster":2820,"type":"endpoint","deviceIeeeAddress":"0x00124b0024c0522c","endpointID":1},{"cluster":1794,"type":"endpoint","deviceIeeeAddress":"0x00124b0024c0522c","endpointID":1}],"configuredReportings":[],"meta":{}},"2":{"profId":260,"epId":2,"devId":81,"inClusterList":[4,5,6,1794,2820],"outClusterList":[],"clusters":{"genOnOff":{"attributes":{"onOff":1,"onTime":0,"offWaitTime":0}},"haElectricalMeasurement":{"attributes":{"rmsVoltage":244,"rmsCurrent":32,"activePower":4}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b0024c0522c","endpointID":1}],"configuredReportings":[],"meta":{}}},"appVersion":68,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":948850772},"lastSeen":1684145480174,"defaultSendRequestWhen":"immediate"}
`

And here is the external converter I'm working on:

```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const utils = require('zigbee-herdsman-converters/lib/utils');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const e = exposes.presets;
const ea = exposes.access;
const globalStore = require('zigbee-herdsman-converters/lib/store');

const fzLocal = {
    // MG-AUZG01 requires multiEndpoint only for on_off
    // https://github.com/Koenkk/zigbee2mqtt/issues/13190
    MGAUZG01_on_off: {
        cluster: 'genOnOff',
        type: ['attributeReport', 'readResponse'],
        convert: (model, msg, publish, options, meta) => {
            if (msg.data.hasOwnProperty('onOff')) {
                const endpointName = utils.getKey(model.endpoint(meta.device), msg.endpoint.ID);
                return {[`state_${endpointName}`]: msg.data['onOff'] === 1 ? 'ON' : 'OFF'};
            }
        },
    },
};

const definition = {
    fingerprint: [{modelID: 'TS011F', manufacturerName: '_TZ3000_raviyuvk'}],
    model: 'MG-AUZG01',
    vendor: 'Makegood',
    description: 'Double Zigbee power point',
    fromZigbee: [
        fzLocal.MGAUZG01_on_off,
        fz.electrical_measurement,
        fz.metering,
        fz.ignore_basic_report],
    toZigbee: [tz.on_off],
    exposes: [
        e.switch().withEndpoint('l1'),
        e.switch().withEndpoint('l2'),
        e.power(),
        e.current(),
        e.voltage().withAccess(ea.STATE),
        e.energy()],

    endpoint: (device) => {
        return {'l1': 1, 'l2': 2};
    },
    
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(1);
        await endpoint.read('genBasic', ['manufacturerName', 'zclVersion', 'appVersion', 'modelId', 'powerSource', 0xfffe]);
        await reporting.bind(endpoint, coordinatorEndpoint, ['genOnOff', 'haElectricalMeasurement', 'seMetering']);
        await reporting.bind(device.getEndpoint(2), coordinatorEndpoint, ['genOnOff']);
        endpoint.saveClusterAttributeKeyValue('haElectricalMeasurement', {acCurrentDivisor: 1000, acCurrentMultiplier: 1});
        endpoint.saveClusterAttributeKeyValue('seMetering', {divisor: 100, multiplier: 1});
        device.save();
    },
    options: [exposes.options.measurement_poll_interval()],
    onEvent: onEventMeasurementPoll,
     
      
};

async function onEventMeasurementPoll(type, data, device, options) {
    const endpoint = device.getEndpoint(1);
    const endpoint2 = device.getEndpoint(2);
    if (type === 'stop') {
      clearInterval(globalStore.getValue(device, 'interval'));
      globalStore.clearValue(device, 'interval');
    } else if (!globalStore.hasValue(device, 'interval')) {
      const seconds =
        options && options.measurement_poll_interval
          ? options.measurement_poll_interval
          : 60;
      if (seconds === -1) return;
      const interval = setInterval(async () => {
        try {
          await endpoint.read('haElectricalMeasurement', [
            'rmsVoltage',
            'rmsCurrent',
            'activePower',
          ]);
          await endpoint2.read('haElectricalMeasurement', [
            'rmsVoltage',
            'rmsCurrent',
            'activePower',
          ]);
        } catch (error) {
          /* Do nothing*/
        }
      }, seconds * 1000);
      globalStore.putValue(device, 'interval', interval);
    }
  }

module.exports = definition;

```

Hoping someone can help tidy up these options!

Koenkk, 2023-05-15:
- You can manually set the divisor by adding the following to `configure`: `            endpoint.saveClusterAttributeKeyValue('haElectricalMeasurement', {acCurrentDivisor: 1000, acCurrentMultiplier: 1});`
- Can you provide the log of this? 
- Probably by publishign to `zigbee2mqtt/DEVICE_ID/set` payload `{"factory_reset":""}`

brywithawhy, 2023-05-15:
> * You can manually set the divisor by adding the following to `configure`

This is already in the config above but doesn't work. If I try and set this manually in the dev console an error is returned saying "Unsupported attribute"

> * Can you provide the log of this?

It actually stops zigbee2mqtt from starting. The line I'm adding to fromZigbee and toZigbee is `fz.moes_power_on_behavior,`
```
[08:15:53] INFO: Preparing to start...
[08:15:54] INFO: Socat not enabled
[08:15:55] INFO: Starting Zigbee2MQTT...
/app/node_modules/zigbee-herdsman-converters/index.js:91
        if (converter.options) {
                      ^
TypeError: Cannot read properties of undefined (reading 'options')
    at Object.addDefinition [as addDeviceDefinition] (/app/node_modules/zigbee-herdsman-converters/index.js:91:23)
    at new ExternalConverters (/app/lib/extension/externalConverters.ts:15:17)
    at new Controller (/app/lib/controller.ts:84:58)
    at start (/app/index.js:106:18)
```

> * Probably by publishign to `zigbee2mqtt/DEVICE_ID/set` payload `{"factory_reset":""}`

This worked, thanks.

Koenkk, 2023-05-16:
@brywithawhy 

> This is already in the config above but doesn't work. If I try and set this manually in the dev console an error is returned saying "Unsupported attribute"

Can you specify what doesn't work? This will set the divisor to be used in `fz.electrical_measurements`

> It actually stops zigbee2mqtt from starting. The line I'm adding to fromZigbee and toZigbee is fz.moes_power_on_behavior,

Yesterday I added support for a similar device, please try with that config: https://github.com/Koenkk/zigbee-herdsman-converters/blob/639bce4346328546d64fa2be6dd219a143276384/devices/tuya.js#L5241

brywithawhy, 2023-05-17:
> Can you specify what doesn't work?

`endpoint.saveClusterAttributeKeyValue('haElectricalMeasurement', {acCurrentDivisor: 1000, acCurrentMultiplier: 1});` is in my config but it does not change the reported value. I.e. it is still using 76 amps to charge my toothbrush. Even if I change this to 1000000 it produces the same result

> Yesterday I added support for a similar device, please try with that config:

This removes my energy monitoring entries, but the power restore options are back. I'll use those parts to get that option back. Thank you

Koenkk, 2023-05-20:
Maybe it needs to be setup for ep2:

```js
device.getEndpoint(2).saveClusterAttributeKeyValue('haElectricalMeasurement', {acCurrentDivisor: 1000, acCurrentMultiplier: 1});
```

brywithawhy, 2023-06-21:
Not stale
Need to start over as recent update broke my external converter.
Will post back when I get time

brywithawhy, 2023-09-15:
FWIW, energy monitoring works out of the box for this device with ZHA, and I adjusted the divisor with a custom quirk so I'm now using zha instead

Here is the post that helped me correct it in ZHA, in case anyone is looking for this in the future and can write a Z2M converter
https://github.com/zigpy/zha-device-handlers/issues/1784#issuecomment-1257261521

flybrys, 2024-02-02:
@Koenkk wondering if we can re-open this one and give it another crack?
Was running it on ZHA on a separate dongle however had drop out issues so I've moved it back to Z2M.

The device pairs with the out-of-box converter and the on/off switch works. Child Lock and Backlight controls report invalid attribute (don't want them anyway).

But would REALLY like to get energy monitoring working again as my whole house uses these GPO's. The custom converter I wrote ages ago no longer works and Z2m won't start with it configured.

Thanks in advance

chukaonline, 2024-04-20:
@flybrys did you eventually get this to work? I have same device and lost power monitoring with the latest update

flybrys, 2024-04-21:
> @flybrys did you eventually get this to work? I have same device and lost power monitoring with the latest update

No I didn't unfortunately. I've spent several hours on it but I feel the documentation is lacking and it's just too technical for me to understand. I'm running a separate dongle with ZHA just for these power points with the hope of someone else with the same model might come along and be able to fix it.

closed 2024-11-18 by github-actions[bot]