Koenkk/zigbee2mqtt #17111: IKEA controller E2001/E2002 binding to LED1924G9 doesn't work

hnykda, 2023-03-22:
### What happened?

When binding this controller to this RGBW bulb, only ON/OFF works, but I cannot cycle through colors (including White). It used to work on older firmware/versions of z2m (but I don't know when it broke exactly). Also, once I bind the controller to the light bulb, I no longer see any messages in z2m topic from this remote, I cannot unbind, nothing - it seems it's "invisible" from z2m from that point. I have to force remove the controller and pair it again.

### What did you expect to happen?

1. Binding the controller to the light would control the colors. 
2. Binding wouldn't make the controller invisible from z2m POV

### Zigbee2MQTT version

1.30.2

### Adapter firmware version

20221226

### Adapter

Slaesh

mamrai1, 2023-03-30:
i have the same problem with same light/controller... 
the arrows cycle the color only when controller is not paired.

rospe, 2023-04-02:
Same here.
On/off and brightness works, but left/right arrows do not change colors. Adapter ist sonoff e here, though.

hnykda, 2023-05-03:
This is still an issue

bronkoo, 2023-05-16:
Same here.
On/off and brightness works, but left/right arrows do not change colors. Adapter is Conbee II.

hnykda, 2023-06-16:
Don't do it github-actions

bronkoo, 2023-07-17:
still alive...

Tonio16, 2023-08-10:
Hello

The same for me with an hue iris bulb.

Antoine

closed 2023-09-18 by github-actions[bot]

MattWestb, 2023-09-18:
@Tonio16 This function is only working with IKEA lights and need adding IKEAs default scenes in the device or doing one firmware upgrade of it and its creating them until the device is being reseed.

MattWestb, 2023-09-18:
You can do wot you like with other devices then IKEA is sending the commands as manufacture tagged and its only IKEA devices that shall reacting on them if they is Zigbee certified..
Do normal automation for others but they is not working if the host system is not online as it so with correct setup IKEA scenes.

Read more in earlier post of the magic IKEA scene if you like getting it working OK with IKEA scene controllers.