Koenkk/zigbee2mqtt #17112: NWK_TABLE_FULL

guillaume042, 2023-03-22:
### What happened?

After added 2 news devices, my network suddenly became unstable.
Devices are getting offline. 
Is see a lot of : 
``
Failed to read state of 'salon-salleamanger-inter-plafonnier' after reconnect (Read 0x5c0272fffe16b298/1 genOnOff(["onOff"], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (SREQ '--> ZDO - extRouteDisc - {"dstAddr":12780,"options":0,"radius":30}' failed with status '(0xc7: NWK_TABLE_FULL)' (expected '(0x00: SUCCESS)')))
``
On several devices.

### What did you expect to happen?

Stable network

### Zigbee2MQTT version

1.30.2 commit: cdf62ea

### Adapter firmware version

20221226

### Adapter

SONOFF Zigbee 3.0 USB Dongle Plus ZBDongle-P

### Debug log

2023-03-22 21:00:21Failed to read state of 'cuisine_inter_spot_plandetravail' after reconnect (Read 0x60a423fffeaae67c/1 genOnOff(["onOff"], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'No network route' (205)))
error 2023-03-22 21:00:23Failed to read state of 'sdb_bas_inter' after reconnect (Read 0x60a423fffee2a712/1 genOnOff(["onOff"], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'MAC channel access failure' (225)))
error 2023-03-22 21:01:42Failed to read state of 'sdb_haut_inter' after reconnect (Read 0x847127fffea48c30/1 genOnOff(["onOff"], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'No network route' (205)))
error 2023-03-22 21:03:52Publish 'set' 'state' to ' nathan_plafonnier' failed: 'Error: Command 0xa4c13853239193d7/1 genOnOff.on({}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (SREQ '--> ZDO - extRouteDisc - {"dstAddr":52725,"options":0,"radius":30}' failed with status '(0xc7: NWK_TABLE_FULL)' (expected '(0x00: SUCCESS)'))'
error 2023-03-22 21:04:16Publish 'set' 'state' to ' nathan_plafonnier' failed: 'Error: Command 0xa4c13853239193d7/1 genOnOff.on({}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 52725 - 1 - 166 - 6 - 11 after 10000ms)'
error 2023-03-22 21:04:39Publish 'set' 'state' to ' nathan_plafonnier' failed: 'Error: Command 0xa4c13853239193d7/1 genOnOff.off({}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 52725 - 1 - 167 - 6 - 11 after 10000ms)'
error 2023-03-22 21:05:02Publish 'set' 'state' to ' nathan_plafonnier' failed: 'Error: Command 0xa4c13853239193d7/1 genOnOff.off({}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 52725 - 1 - 169 - 6 - 11 after 10000ms)'
error 2023-03-22 21:06:30Publish 'set' 'state' to 'sdb_bas_inter' failed: 'Error: Command 0x60a423fffee2a712/1 genOnOff.on({}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (SREQ '--> ZDO - extRouteDisc - {"dstAddr":39445,"options":0,"radius":30}' failed with status '(0xc7: NWK_TABLE_FULL)' (expected '(0x00: SUCCESS)'))'
error 2023-03-22 21:10:10Failed to read state of ' nathan_plafonnier' after reconnect (Read 0xa4c13853239193d7/1 lightingColorCtrl(["colorMode","colorTemperature"], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 52725 - 1 - 205 - 768 - 1 after 10000ms))

CamSoper, 2023-03-22:
I think you're running into [this](https://github.com/Koenkk/Z-Stack-firmware/issues/418). I get the same thing whenever I upgrade my firmware past 20220219.

guillaume042, 2023-03-23:
Txs for the clue i've rollback to check if it works