Koenkk/zigbee2mqtt #17115: Light group is shown as working but nothing happens

maxime1992, 2023-03-22:
### What happened?

When I try to toggle a group of lights, everything seems to work as it should be (frontend, logs and in HA), but the lights are not turning on or off in real life. I can toggle the lights one by one with no issue, but not in a group.

### What did you expect to happen?

I want my lights to turn on and off in real life when I toggle the group

### How to reproduce it (minimal and precise)

See https://github.com/Koenkk/zigbee2mqtt/issues/11744

### Zigbee2MQTT version

1.30.2-dev commit: 2e593e3

### Adapter firmware version

20220219

### Adapter

SONOFF ZigBee 3.0 USB Dongle Plus, TI CC2652P

### Debug log

See https://github.com/Koenkk/zigbee2mqtt/issues/11744

jafd, 2023-03-26:
I have an Electrollama zzh and some Trådfri bulbs in one room which I hoped to group together and use as one. I distinctly remember that some previous version of z2m had them working just fine. Right now, though, the symptoms are exactly as described in the issue: you recall a scene, you can see in the web UI that the state has changed as programmed, you can see the corresponding messages being published in the log, except that nothing is happening in the actual physical world.

Which is also frustrating because it leads to HomeAssistant thinking the bulbs are in a different state than they actually are.

I have a followup on this. What helped was deleting the group and re-creating it again. But! I cannot bind a remote to the group. That is, I bind it just fine (supposedly), but when I press buttons, nothing happens. Toggling the state of the group in HA, or recalling a scene in z2m works.

I'm not sure where groups actually live, but it looks like it's partly on the coordinator to be aware of the groups, and maybe it had lost some state information it was supposed to hold without z2m being aware of it.

jafd, 2023-03-26:
Off-topic, but stalebot is next worst thing to pure evil when it comes to this.

maxime1992, 2023-04-26:
Still happening

maxime1992, 2023-05-28:
Up. This is preventing me to manipulate all my lights at once which is quite painful

closed 2023-07-07 by github-actions[bot]