Koenkk/zigbee2mqtt #17116: Pet feeder ZNCWWSQ01LM: Schedule issue

Stilnight, 2023-03-22:
### What happened?

Hi,

Here is my feedback with the last update of zigbee2MQTT and this device ZNCWWSQ01LM Pet feeder.
I updated the last firmware with OTA through zigbee2MQTT (btw versions are tagged are "null" but it's well detected and updated).

1. Start Feeding works perfectly it's giving the set up portion 
(BTW would be nice to have a switch button for this one, I only get on Homekit the witch button of the led indicator)

2. Led indicator Switch is reversed: When On the leds are turned off. When Off they are turned On

3. my biggest issur is on the Scheduling. I did not succeed to make it works.
Tried UTC time/ UTC+1 with 24hours format. That does not trigger anything on the device.
Also when trying to add more than 4 time slots with the scheduler I get the following error:

Info 2023-03-22 23:07:09MQTT publish: topic 'zigbee2mqtt/Croquettes', payload '{"child_lock":"UNLOCK","error":false,"feed":"","feeding_size":2,"feeding_source":"manual","led_indicator":"ON","linkquality":255,"mode":"schedule","portion_weight":20,"portions_per_day":16,"schedule":[{"days":"everyday","hour":21,"minute":38,"size":2},{"days":"everyday","hour":21,"minute":39,"size":2},{"days":"everyday","hour":21,"minute":40,"size":2},{"days":"everyday","hour":22,"minute":40,"size":2},{"days":"everyday","hour":23,"minute":41,"size":2}],"serving_size":2,"update":{"installed_version":9761,"latest_version":9761,"state":"idle"},"update_available":false,"weight_per_day":240}'
**Warning 2023-03-22 23:07:10zigbee-herdsman-converters:aqara_feeder: Unknown attribute   = �67F1526
Error 2023-03-22 23:07:10Exception while calling fromZigbee converter: Attempt to access memory outside buffer bounds}**

4. Finally number of portion is clearly having an incidence on the size of food given, but the other factor weight of ration is not changing anything on the given portion, they are exactly the same size.

Please do not hesitate to ask me anyquestion with guidance if it's a bit technical to get :)

Thanks in advance!

### What did you expect to happen?

Led indicator reverse behaviour, when On led On, when Off led OFF.

Scheduling triggering the device at the given hours.

Increasing weight of portion giving more foods.

### How to reproduce it (minimal and precise)

use frontend from MQTT on a C1 pet feeder from Aqara.

### Zigbee2MQTT version

 1.30.2 commit: cdf62ea

### Adapter firmware version

0x26720700

### Adapter

ConBee2/RaspBee2

Stilnight, 2023-03-24:
Hey,

After some tests the scheduler is working but the hour is not at all UTC or UTC+1. 
It's 00:00 is corresponding to 21:00 (UTC+1). So basically it is set up currently in UTC-2.

I do not know if it requires a correction or not or if we could have a set up to choose the time zone. 
Could be simply the Wiki on the device updated to give this info.

Then There still two issues to be corrected to my mind:

- Led indicator inverted
- Cannot set up more than 4 time slots (device limit? or issue in the adpater?)

Potentially also a new feature of the device which would be really appreciated for integration to assistant and potential automation:
- Function Start feeding to be set up as a timer switch (On then back to Off after info sent tot he device)

Thanks!

eikowagenknecht, 2023-03-25:
Interesting finding regarding the time zone. I have the same issue and will see when the stats reset to find out when "0:00" in device-time is.

I also noticed that the firmware version is not shown in the OTA view.

The LED indicater setting does not work for me at all, the LED is always on.

And another thing:
- Manually feeding "1" portion rotates the blade one turn - just fine. I can do this as often as I want without problems
- Manually feeding "2" portions rotates the blade 1.8 times (not everything from the second chamber is dispensed)
- Manually feeding "3" portions or more rotates the blade 1.8 times, then a bit backwards and then the LED keeps blinking red and the error flag is set.

Are you experiencing this as well?

Stilnight, 2023-03-25:
Hi, No actually currently giving 4 portions and no errors. Did you updated the OTA with check version? May be a reset pairing necessary?

eikowagenknecht, 2023-03-25:
I updated the firmware and then repaired. While pairing, there were errors shown with the following logs:

```
Debug 2023-03-25 18:44:14Received Zigbee message from '0x54ef44100070e06b', type 'attributeReport', cluster 'aqaraOpple', data '{"65521":{"data":[0,5,9,8,0,7,209,1,100],"type":"Buffer"}}' from endpoint 1 with groupID 0
Warning 2023-03-25 18:44:14zigbee-herdsman-converters:aqara_feeder: Unhandled attribute � = d
Debug 2023-03-25 18:44:14Received Zigbee message from '0x54ef44100070e06b', type 'attributeReport', cluster 'aqaraOpple', data '{"65521":{"data":[0,5,10,13,9,0,85,1,0],"type":"Buffer"}}' from endpoint 1 with groupID 0
Warning 2023-03-25 18:44:14zigbee-herdsman-converters:aqara_feeder: Unhandled attribute U = 
```

I then tried it again after the repairing:
- 1 portion: Works fine
- 2 portions: Works fine
- 3 portions: Rotates 2 chambers in the right direction, then 1 backwards, then 1.3 or so forwards again. This also happens when I press the button on the device when 3 or more portions are set. Maybe it's faulty hardware? I can't check with the Aqara app unforunately because I don't own a hub from them.

closed 2023-05-02 by github-actions[bot]

inomdzhon, 2024-12-19:
Hi, @Stilnight and @eikowagenknecht 

@Stilnight I see you have `portion_weight` is 20g. Does is it work?

In my case, device use always default `portion_weight` (is ~6g) instead 20g :( 

All other features works (like schedule, manual control, severing size)