Koenkk/zigbee2mqtt #17119: Tuya IH012-RT01 Motion Sensor only works directly on Coordinator, not via Router

SuperSnackman, 2023-03-23:
### What happened?

For testing purposes I set up a very small Zigbee network:
Coordinator (Sonoff ZBDongle-P) Downstairs.
Hue Bulb Upstairs directly connected tot the Coordinator
Tuya Motion Sensor (IH012-RT01) connected using the Hue Bulb as a router. This also shows up in my Network Map.

My Motion Sensor pairs correctly, but I do not receive messages after pairing. I can changed Sensitivity and KeepTime, but I will never get a message out of the Sensor.

When pairing the Motion sensor to the Coordinator (I Have to move it closer) everyting works as expected.
When Using ZHA instead of Z2M everyting works as expected.

What could be causing this issue?

[image]

### What did you expect to happen?

The Motion sensor also working using a Router

### How to reproduce it (minimal and precise)

Set up a fresh Zigbee network on Channel 11. New key, new panid, net ext-pan-id.
Pair the Hue Bulb
Permit Join Only via the Hue Bulb
Pair the Motion Sensor

### Zigbee2MQTT version

1.30.2-dev commit (Also tried Latest)

### Adapter firmware version

20221226

### Adapter

Sonoff ZBDongle-P

d4rthpaul, 2023-04-03:
I have a similar issue with this motion sensor, paired fine to my coordinator and works when in a room near it. But I bought it to use in my garage which has its own router as its too far from the coordinator. But although it flashes blue to acknowledge motion it never connects to the router and passes the message. I have a zigbee door sensor on the garage that works fine.

WizBangCrash, 2023-04-06:
I have a similar issue with my Hue Still Ceiling light.
The hue light will pair to the coordinator and work fine, but I cannot pair any devices to the light. I have an Aqara LUMI weather that will pair fine to the coordinator or any other router in the house, but I cannot get it to pair to the Hue light.
The hue light is running the latest firmware (1.101.2) and I have had no issues with the LUMI weather while it was paired to other routers.

closed 2023-05-14 by github-actions[bot]

anthonym21, 2023-09-09:
Did anyone find a solution to this? Still having the issue

WizBangCrash, 2023-09-10:
I never found a solution. I put in another non-Hue device that can act as a router to extend my network. I found no way to get the Hue light to accept a pairing from another device. Thinking about it though. I did not try another Hue device to see if that paired okay.