Koenkk/zigbee2mqtt #17124: TS0601_cover_1 Stops responding & Goes Offline Randomly

miannelli516, 2023-03-23:
### What happened?

i recently purchased 15 Tuya battery operated shade motors for my blinds. i installed them about 2 months ago and randomly the devices will sometimes go completely offline and require me to repair the device or the device it self will not respond at all via zigbee2mqtt. The interesting thing is, if i try an use the manufactures remote that came with the motors, it works no problem and half of the time after using the remote, zigbee2mqtt is able to communicate with the device again. In addition, on 2 occasions, ive also seen the motor completely stop responding when paired with zigbee2mqtt even using the remote. Ive swapped out the motors with new ones thinking it was the motor and the same issue happens and i even tried 3 different zigbee adapters. Any help you can provide is greatly appreciated. 

### Zigbee2MQTT version

1.30.2

### Adapter firmware version

20220302

### Adapter

ZigStar LAN Coordinator

giejay, 2023-03-27:
Just curious, do you see state changes in z2m when manually adjusting the shades? I'm facing something similar with a tuya dimmer which is not controllable after a while but I am seeing state changes when clicking the button myself.

jdbrookes, 2023-04-04:
I have the _TZE200_xuzcvlku variant of this cover, 3 devices, and the same thing happens randomly with all of them. One or more will randomly not respond to zigbee commands but if I press a button on a device, or control them with the manufacturer remote (which works on RF?) then they work fine and the state will update in zigbee2mqtt. Most of the time if I leave it a few hours they will start responding to zigbee commands again, but if I notice the issue I am just in the habit of setting zigbee2mqtt to pairing mode, setting the cover motor to pairing mode, and then they re-pair and work again just fine a few moments later.

I haven't been able to identify any particular reason for this - network is otherwise strong and reliable with all devices. Sometimes there is no issue for several weeks, then it might happen 5 times over a week.

closed 2023-05-13 by github-actions[bot]