Koenkk/zigbee2mqtt #17132: [New device support]: Vibration Sensor TS0210 _TYZB01_geigpsy4 not supported

darkness0808, 2023-03-24:
### Link

https://ifeel.systems/catalog/datchiki/datchik_vibratsii_ifeel_vibro_ifs_sv001/

### Database entry

{"id":4,"type":"EndDevice","ieeeAddr":"0x9035eafffe613e7b","nwkAddr":29193,"manufId":4098,"manufName":"_TYZB01_geigpsy4","powerSource":"Battery","modelId":"TS0210","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":1026,"inClusterList":[0,10,1,1280],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"65503":"9\u0001�+\u0013","modelId":"TS0210","manufacturerName":"_TYZB01_geigpsy4","stackVersion":0,"dateCode":""}},"ssIasZone":{"attributes":{"iasCieAddr":"0xdc8e95fffe08ac91","zoneState":1,"currentZoneSensitivityLevel":2}},"genPowerCfg":{"attributes":{"batteryPercentageRemaining":160,"batteryVoltage":29}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":70,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1679643588264,"defaultSendRequestWhen":"immediate"}

### Comments

I think that this is a TUYA vibration sensor only in a different case.
[image]

I'm not bad at this and I can't write a converter, please help me.

darkness0808, 2023-03-24:
Debug Received Zigbee message from '0x9035eafffe613e7b', type 'attributeReport', cluster 'ssIasZone', data '{"currentZoneSensitivityLevel":2}' from endpoint 1 with groupID 0

Warning Device '0x9035eafffe613e7b' with Zigbee model 'TS0210' and manufacturer name '_TYZB01_geigpsy4' is NOT supported, please follow https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html

Info 2023-03-24 14:29:49MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"definition":null,"friendly_name":"0x9035eafffe613e7b","ieee_address":"0x9035eafffe613e7b","status":"successful","supported":false},"type":"device_interview"}'

darkness0808, 2023-03-24:
complite.

const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const definition = {
        fingerprint: [{modelID: 'TS0210', manufacturerName: '_TYZB01_geigpsy4'},
        model: 'TS0210',
        vendor: 'TuYa',
        description: 'Vibration sensor',
        fromZigbee: [fz.battery, fz.ias_vibration_alarm_1_with_timeout],
        toZigbee: [tz.TS0210_sensitivity],
        exposes: [e.battery(), e.battery_voltage(), e.vibration(), exposes.enum('sensitivity', ea.STATE_SET, ['low', 'medium', 'high'])],
};

module.exports = definition;

Koenkk, 2023-03-26:
Added, thanks!

closed 2023-03-26 by Koenkk

alfem, 2024-04-02:
Hi!

I received another variation of this Tuya vibration sensor. 

I added `modelID: 'TS0210', manufacturerName: '_TZ3000_lqpt3mvr'` to `node_modules/zigbee-herdsman-converters/devices/tuya.js` and it worked properly with z2m (and Home Assistant).

Zbeed, 2024-04-25:
Hi, is there way to add this sensor (TZ3000_lqpt3mvr) to ZHA? I bought it and paired with zha, but sensor not change state. Still "clear" and not change when sensor move. Thank you for advice...

nigels0, 2025-05-24:
> Hi, is there way to add this sensor (TZ3000_lqpt3mvr) to ZHA? I bought it and paired with zha, but sensor not change state. Still "clear" and not change when sensor move. Thank you for advice...

It works with this quirk:
`import logging
from zigpy.profiles import zha
from zigpy.quirks import CustomDevice
from zigpy.zcl.clusters.general import Basic, PowerConfiguration, Identify
from zigpy.zcl.clusters.security import IasZone
from zhaquirks.tuya import TuyaManufCluster
import zigpy.types as t

_LOGGER = logging.getLogger(__name__)

class TuyaAxisCluster(TuyaManufCluster):
    """Custom cluster to decode accelerometer axis data from TS0601."""

    cluster_id = 0xEF00

    attributes = {
        0x0101: ("accel_x", t.int32s, True),
        0x0102: ("accel_y", t.int32s, True),
        0x0103: ("accel_z", t.int32s, True),
    }

    def handle_cluster_request(self, hdr, args, dst_addressing=None):
        """Handle incoming Tuya cluster requests."""
        _LOGGER.debug("handle_cluster_request: hdr=%s args=%s", hdr, args)

        if not args or not hasattr(args[0], "command_id"):
            _LOGGER.warning("Unexpected args: %s", args)
            return

        cmd_id = args[0].command_id
        raw_val = int.from_bytes(args[0].data[-1:], byteorder="big", signed=True)

        if cmd_id == 0x0265:
            self._update_attribute(0x0101, raw_val)
            _LOGGER.debug("Updated accel_x: %d", raw_val)
        elif cmd_id == 0x0266:
            self._update_attribute(0x0102, raw_val)
            _LOGGER.debug("Updated accel_y: %d", raw_val)
        elif cmd_id == 0x0267:
            self._update_attribute(0x0103, raw_val)
            _LOGGER.debug("Updated accel_z: %d", raw_val)
        else:
            _LOGGER.debug("Unhandled command_id: 0x%04x", cmd_id)

class TuyaVibrationSensor(CustomDevice):
    """Custom device representing Tuya vibration sensor _TZ3000_lqpt3mvr."""

    signature = {
        "model": "TS0601",
        "manufacturer": "_TZ3000_lqpt3mvr",
        "endpoints": {
            1: {
                "profile_id": zha.PROFILE_ID,
                "device_type": 0x0402,
                "input_clusters": [
                    0x0000,  # Basic
                    0x0001,  # PowerConfiguration
                    0x0003,  # Identify
                    0x0500,  # IasZone
                ],
                "output_clusters": [],
            },
        },
    }

    replacement = {
        "endpoints": {
            1: {
                "profile_id": zha.PROFILE_ID,
                "device_type": 0x0000,
                "input_clusters": [
                    Basic.cluster_id,
                    Identify.cluster_id,
                    IasZone.cluster_id,
                    PowerConfiguration.cluster_id,
                    TuyaAxisCluster,
                ],
                "output_clusters": [],
            },
        },
    }`