Koenkk/zigbee2mqtt #17134: [Feature request]: Add option to not send duplicate mqtt messages

vandalon, 2023-03-24:
### Is your feature request related to a problem? Please describe

The "problem" this would solve is to prevent flooding of mqtt messages. 

### Describe the solution you'd like

When a mqtt message for a device is the same is the previously sent message, don't sent it out. 

### Describe alternatives you've considered

I've considered using debouncing but this brings the disadvantage that messages are delayed.

### Additional context

When using the "Filtered publish attributes" option on my tuya mmWave sensors z2m still sends out a message when one of the filtered attributes is changed. The above solution would prevent sending out "unnecessary" mqtt messages

danieledwardgeorgehitchcock, 2023-03-28:
As far as I am aware, the debounce logic should not add any additional time to the processing of the message.

Z2M uses the `hasAlreadyProcessedMessage` function to debounce - this effectively lets through the first message as it comes into the stack, but no subsequent messages within the given debounce period.

junosuarez, 2023-03-28:
Can you provide an example from the logs of repeated published messages?

vandalon, 2023-03-29:
Sure:

The first column is the unix timestamp with nanoseconds

```
1680076451.739552450 zigbee2mqtt/Eetkamer mmWave {"illuminance_lux":290,"presence":false}
1680076451.792046669 zigbee2mqtt/Eetkamer mmWave {"illuminance_lux":290,"presence":false}
1680076451.851917373 zigbee2mqtt/Eetkamer mmWave {"illuminance_lux":290,"presence":false}
1680076451.903376464 zigbee2mqtt/Eetkamer mmWave {"illuminance_lux":290,"presence":false}
1680076451.955591047 zigbee2mqtt/Eetkamer mmWave {"illuminance_lux":290,"presence":false}
1680076452.009332653 zigbee2mqtt/Eetkamer mmWave {"illuminance_lux":290,"presence":false}
1680076452.062816790 zigbee2mqtt/Eetkamer mmWave {"illuminance_lux":290,"presence":false}
1680076452.113896740 zigbee2mqtt/Eetkamer mmWave {"illuminance_lux":290,"presence":false}
1680076452.166795268 zigbee2mqtt/Eetkamer mmWave {"illuminance_lux":290,"presence":false}
```

with the following config:
```
  '0xe0798dfffeb1fab4':
    friendly_name: Eetkamer mmWave
    filtered_attributes:
      - target_distance
      - last_seen
      - linkquality
      - elapsed
      - self_test
      - radar_sensitivity
      - maximum_range
      - minimum_range
      - fading_time
      - detection_delay
      - cli
    legacy: false
    optimistic: true
```

closed 2023-05-06 by github-actions[bot]

kevincaradant, 2024-04-29:
Hi everyone,

I'm experiencing the exact same behavior.

image

I've filtered some fields to alleviate Z2M/the coordinator and also HA (DB). However, I'm still receiving identical messages.

Have you found anything to avoid this?

Thank you.

jamesjharper, 2024-06-19:
@kevincaradant as per junosuarez comment, adding the the following extension will prevent identical messages (after removing `filtered_attributes`) from publishing to MQTT. 

Add `filtered_attributes` for the fields you want to be ignored for uniqueness.

_Note: this will only reduce the event volume on MQTT and not the zigbee network._ 

```
// `extension/detectDuplicateEvents.js`
class DetectedDuplicateEvents {
    constructor(zigbee, mqtt, state, publishEntityState, eventBus, settings, logger) {
        this.logger = logger;
        this.previousStates = {};
    }

    adjustMessageBeforePublish(entity, message) {
        // Only apply for devices, change as needed 
        if (!entity.isDevice()) {
            return;
        }

        // `adjustMessageBeforePublish` is called before `filterProperties(...)`,
        // which means the message will still contains filtered_attributes.
        
        // To omit the `filtered_attributes` from determining if the message is 
        // unique we need to call `filterProperties` earlier. Not ideal, as this could 
        // be broken by future updates, and will result in filterProperties be called twice.

        this.filterProperties(entity.options.filtered_attributes, message);

        const currentState = JSON.stringify(message);

        // Index cache with the entities ieeeAddr
        const ieeeAddr = entity.zh.ieeeAddr;

        // Treat as unique message on cache miss
        if (!(ieeeAddr in this.previousStates)) {
            this.previousStates[ieeeAddr] = currentState;
            return;
        }
        
        if (this.previousStates[ieeeAddr] === currentState) {
            // If the message is a duplicate, remove all fields 
            // later in the code path, (here https://github.com/Koenkk/zigbee2mqtt/blob/f18473017ebfcc65e9db1de3002b7eb21079805a/lib/controller.ts#L293)
            // the message will be omitted if no fields are present 
            for (const key in message) {
                if (message.hasOwnProperty(key)) {
                    delete message[key];
                }
            }
        } else {
            // Update the previous state with the current state
            this.previousStates[ieeeAddr] = currentState;
        }
    }

    filterProperties(filter, data) {
        // Copy-pasted from https://github.com/Koenkk/zigbee2mqtt/blob/f18473017ebfcc65e9db1de3002b7eb21079805a/lib/util/utils.ts#L327
        // as unable to import from outside a module. 
        // Very seldom work with js so there's likely a better way to do this. 

        if (filter) {
            for (const property of Object.keys(data)) {
                if (filter.find((p) => property.match(`^${p}$`))) {
                    delete data[property];
                }
            }
        }
    }
  
    async stop() {
        this.eventBus.removeListeners(this.constructor.name);
    }
}

module.exports = DetectedDuplicateEvents;
```