Koenkk/zigbee2mqtt #17135: Upgrading from 1.28.2-1 to 1.30.2-1 breaks Z2MQTT

SHxKM, 2023-03-24:
### What happened?

- I upgraded to `1.30.2-1`
- Zigbee2MQTT fails to start
- The logs show:

> /app/node_modules/zigbee-herdsman-converters/index.js:91
        if (converter.options) {
                      ^
TypeError: Cannot read properties of undefined (reading 'options')
    at Object.addDefinition [as addDeviceDefinition] (/app/node_modules/zigbee-herdsman-converters/index.js:91:23)
    at new ExternalConverters (/app/lib/extension/externalConverters.ts:15:17)
    at new Controller (/app/lib/controller.ts:84:58)
    at start (/app/index.js:106:18)

### What did you expect to happen?

The upgrade to not break Z2M

### How to reproduce it (minimal and precise)

Upgrade from 1.28.2-1 to 1.30.2-1

### Zigbee2MQTT version

1.30.2-1

### Adapter firmware version

0x26580700

### Adapter

ConBee2

### Debug log

/app/node_modules/zigbee-herdsman-converters/index.js:91
        if (converter.options) {
                      ^
TypeError: Cannot read properties of undefined (reading 'options')
    at Object.addDefinition [as addDeviceDefinition] (/app/node_modules/zigbee-herdsman-converters/index.js:91:23)
    at new ExternalConverters (/app/lib/extension/externalConverters.ts:15:17)
    at new Controller (/app/lib/controller.ts:84:58)
    at start (/app/index.js:106:18)

SHxKM, 2023-03-24:
The issue was that I'd registered an external converter which was already mapped in herdsman. It has been mapped for over two years so I don't know why Z2M broke just now.

sanyafifa, 2023-05-14:
```
May 14 20:32:32 RP4 npm[30017]:         if (converter.options) {
May 14 20:32:32 RP4 npm[30017]:                       ^
May 14 20:32:32 RP4 npm[30017]: TypeError: Cannot read properties of undefined (reading 'options')
May 14 20:32:32 RP4 npm[30017]:     at Object.addDefinition [as addDeviceDefinition] (/opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/index.js:
May 14 20:32:32 RP4 npm[30017]:     at new ExternalConverters (/opt/zigbee2mqtt/lib/extension/externalConverters.ts:15:17)
May 14 20:32:32 RP4 npm[30017]:     at new Controller (/opt/zigbee2mqtt/lib/controller.ts:84:58)
May 14 20:32:32 RP4 npm[30017]:     at start (/opt/zigbee2mqtt/index.js:106:18)
May 14 20:32:32 RP4 npm[30017]:     at processTicksAndRejections (node:internal/process/task_queues:96:5)
May 14 20:32:32 RP4 systemd[1]: zigbee2mqtt.service: Succeeded.
May 14 20:32:32 RP4 systemd[1]: Stopped zigbee2mqtt.

```

I have the same problem. help :)