Koenkk/zigbee2mqtt #17137: No converter available for 'action' // Button: BTP01

K44nk1231, 2023-03-24:
### What happened?

I cannot set the action status of my button to hold.

Error in Log: No converter available for 'action' ("hold")

### What did you expect to happen?

MQTT publish: topic 'zigbee2mqtt/Button1', payload '{"action":"hold","battery":82,"linkquality":132,"temperature":23.14,"voltage":2700}'

### How to reproduce it (minimal and precise)

Sending: {"action":"hold"}
To the commandTopic: "zigbee2mqtt/Button1/set"

### Zigbee2MQTT version

1.30.2

### Adapter firmware version

6.10.3.0 build 297

### Adapter

Sonoff Zigbee 3.0 USB Dongle Plus E

Koenkk, 2023-03-24:
An action is something that comes from the device (e.g. when you click it), not something you send to the device.

K44nk1231, 2023-03-24:
Yeah after I thought about it a bit more, it would make sense to be able to write the status of the device, however due to it being a Zigbee device there is no constant communication and it cant write it as such.

I guess I got confused because the documentation here: [https://www.zigbee2mqtt.io/devices/IM6001-BTP01.html](url)
suggested that I can write the action. Maybe that should be amended:

`Exposes
Action (enum)

Triggered action (e.g. a button click). Value can be found in the published state on the action property. It's not possible to read (/get) or write (/set) this value. The possible values are: off, single, double, hold.`

Thanks for the quick reply anyway!

closed 2023-03-26 by Koenkk